openhwfoundation/core-v-verif #1874: ISS: mseccfg reset value override not functional

silabs-hfegran, 2023-05-05:
## ISS: mseccfg reset value override not functional

### Type
* Functionally incorrect behavior

### Steps to Reproduce
Approach 1:
1. cv32e40s/dev - any recent hash
2. Add --override cpu/mseccfg=0x00000004 to `core-v-verif/cv32e40s/tests/cfg/pmp.yaml` in the ovpsim section
3. `make test TEST=hello-world CFG=pmp`
4. Result: No mismatch (**wrong** - RTL mseccfg should have reset value 0, ISS should have value 4 on first instruction retirement)

Approach 2:
1. cv32e40s/dev - any recent hash
2. Add --override cpu/mseccfg=0x00000004 to `core-v-verif/cv32e40s/tests/cfg/pmp.yaml` in the ovpsim section
3. Change: `parameter cv32e40s_pkg::mseccfg_t CORE_PARAM_PMP_MSECCFG_RV = cv32e40s_pkg::MSECCFG_DEFAULT;` to `parameter cv32e40s_pkg::mseccfg_t CORE_PARAM_PMP_MSECCFG_RV = 4;` in `core-v-verif/cv32e40s/tb/uvmt/uvmt_cv32e40s_constants.sv`
4. `make test TEST=hello-world CFG=pmp`
5. Result: Mismatch (**wrong** - RTL and ISS should match with mseccfg = 4)

### Additional context
This has been tested with multiple configurations, PMP, no PMP, clic, no clic, etc, with the setting first and last in the list of overrides; 
 in no observed cases has it been possible to get the ISS to actually override this value. Its override dump claims that the override is set, but the value set is not the value the ISS uses for comparison during execution of the first instruction.

ISS showoverrides output from approach 2, appears that the setting is set correctly
```diff
# refRoot/cpu/CSR_Defaults
    --override refRoot/cpu/marchid=21 (Uns64) (default=21) (default) Override marchid register
    --override refRoot/cpu/mconfigptr=0 (Uns64) (default=0) (default) Override mconfigptr register
    --override refRoot/cpu/mhartid=0 (Uns64) (default=0) (override) Override mhartid register (or first mhartid of an incrementing sequence if this is an SMP variant)
    --override refRoot/cpu/mimpid=0 (Uns64) (default=0) (override) Override mimpid register
+    --override refRoot/cpu/mseccfg=0x00000004 (Uns64) (default=0) (override) Override mseccfg register
    --override refRoot/cpu/mtvec=0x00000001 (Uns64) (default=1) (override) Override mtvec register
    --override refRoot/cpu/mvendorid=0x602 (Uns64) (default=0x602) (default) Override mvendorid register

(...)
# refRoot/cpu/PMP Configuration
    --override refRoot/cpu/PMP_decompose=F (Boolean) (default=F) (default) Whether unaligned PMP accesses are decomposed into separate aligned accesses
    --override refRoot/cpu/PMP_grain=0 (Uns32) (default=0) (override) Specify PMP region granularity, G (0 => 4 bytes, 1 => 8 bytes, etc)
    --override refRoot/cpu/PMP_initialparams=F (Boolean) (default=F) (default) Enable parameters to change the reset values for PMP CSRs
    --override refRoot/cpu/PMP_maskparams=F (Boolean) (default=F) (default) Enable parameters to change the read-only masks for PMP CSRs
    --override refRoot/cpu/PMP_max_page=0 (Uns32) (default=0) (default) Specify the maximum size of PMP region to map if non-zero (may improve performance; constrained to a power of two)
    --override refRoot/cpu/PMP_registers=64 (Uns32) (default=0) (override) Specify the number of implemented PMP address registers
    --override refRoot/cpu/PMP_undefined=F (Boolean) (default=F) (default) Whether accesses to unimplemented PMP registers are undefined (if True) or write ignored and zero (if False)

```

Log output after mismatch on first instruction, does not state that the value of the mseccfg register is 4, but rather 0
```diff
Info (IDV) Instruction executed prior to mismatch '0x80(_start+0): 00013197 auipc   x3,0x13'
Error (IDV) CSR register value mismatch (HartId:0, PC:0x00000080 _start+0):
Info (IDV)  0> CSR 747 (mseccfg)
+Info (IDV)   . dut:0x00000004
-Info (IDV)   . ref:0x00000000
UVM_ERROR @ 39.300 ns : rvvi-pkg.sv(94) reporter [] uvmt_cv32e40s_tb.imperas_dv.idv_trace2api.state_compare @ 39.000 ns: MISMATCH

Dump State GPR
 0: 00000000  1: 00000000  2: 00000000  3: 00013080 
 4: 00000000  5: 00000000  6: 00000000  7: 00000000 
 8: 00000000  9: 00000000 10: 00000000 11: 00000000 
12: 00000000 13: 00000000 14: 00000000 15: 00000000 
16: 00000000 17: 00000000 18: 00000000 19: 00000000 
20: 00000000 21: 00000000 22: 00000000 23: 00000000 
24: 00000000 25: 00000000 26: 00000000 27: 00000000 
28: 00000000 29: 00000000 30: 00000000 31: 00000000 

Dump State CSR
             jvt: 00000000          mstatus: 00001800             misa: 40901104              mie: 00000000 
           mtvec: 00000001       mcounteren: 00000000          menvcfg: 00000000        mstateen0: 00000000 
       mstateen1: 00000000        mstateen2: 00000000        mstateen3: 00000000         mstatush: 00000000 
        menvcfgh: 00000000       mstateen0h: 00000000       mstateen1h: 00000000       mstateen2h: 00000000 
[... 27 lines cut ...]
       pmpaddr56: 00000000        pmpaddr57: 00000000        pmpaddr58: 00000000        pmpaddr59: 00000000 
       pmpaddr60: 00000000        pmpaddr61: 00000000        pmpaddr62: 00000000        pmpaddr63: 00000000 
-         mseccfg: 00000000         mseccfgh: 00000000          tselect: 00000000           tdata1: 28001000 
          tdata2: 00000000           tdata3: 00000000            tinfo: 00008064         tcontrol: 00000000 
            dcsr: 40000413              dpc: 00000000        dscratch0: 00000000        dscratch1: 00000000 
          mcycle: 00000000         minstret: 00000000     mhpmcounter3: 00000000     mhpmcounter4: 00000000 
    mhpmcounter5: 00000000     mhpmcounter6: 00000000     mhpmcounter7: 00000000     mhpmcounter8: 00000000 
    mhpmcounter9: 00000000    mhpmcounter10: 00000000    mhpmcounter11: 00000000    mhpmcounter12: 00000000 
   mhpmcounter13: 00000000    mhpmcounter14: 00000000    mhpmcounter15: 00000000    mhpmcounter16: 00000000 
   mhpmcounter17: 00000000    mhpmcounter18: 00000000    mhpmcounter19: 00000000    mhpmcounter20: 00000000 
   mhpmcounter21: 00000000    mhpmcounter22: 00000000    mhpmcounter23: 00000000    mhpmcounter24: 00000000 
   mhpmcounter25: 00000000    mhpmcounter26: 00000000    mhpmcounter27: 00000000    mhpmcounter28: 00000000 
   mhpmcounter29: 00000000    mhpmcounter30: 00000000    mhpmcounter31: 00000000          mcycleh: 00000000 
       minstreth: 00000000    mhpmcounterh3: 00000000    mhpmcounterh4: 00000000    mhpmcounterh5: 00000000 
   mhpmcounterh6: 00000000    mhpmcounterh7: 00000000    mhpmcounterh8: 00000000    mhpmcounterh9: 00000000 
  mhpmcounterh10: 00000000   mhpmcounterh11: 00000000   mhpmcounterh12: 00000000   mhpmcounterh13: 00000000 
  mhpmcounterh14: 00000000   mhpmcounterh15: 00000000   mhpmcounterh16: 00000000   mhpmcounterh17: 00000000 
  mhpmcounterh18: 00000000   mhpmcounterh19: 00000000   mhpmcounterh20: 00000000   mhpmcounterh21: 00000000 
  mhpmcounterh22: 00000000   mhpmcounterh23: 00000000   mhpmcounterh24: 00000000   mhpmcounterh25: 00000000 
  mhpmcounterh26: 00000000   mhpmcounterh27: 00000000   mhpmcounterh28: 00000000   mhpmcounterh29: 00000000 
  mhpmcounterh30: 00000000   mhpmcounterh31: 00000000          cpuctrl: 00000019      secureseed0: 00000000 
     secureseed1: 00000000      secureseed2: 00000000        mvendorid: 00000602          marchid: 00000015 
          mimpid: 00000000          mhartid: 00000000       mconfigptr: 00000000 

```

eroom1966, 2023-05-09:
Hi @silabs-hfegran 
This override value was not being applied correctly.
I have a fix will allow the override to be set as intended, I have tested this and it works correctly with your testcases

I have uploaded a fix to the ImperasDV repo for the CV32E40S

commit 41ba68958cb996eeeba792c2a344bfda48933067 (HEAD -> main, origin/main, origin/HEAD)

Thx
Lee

silabs-hfegran, 2023-05-10:
Thank you for the fix @eroom1966, I believe this is resolved now.

closed 2023-05-10 by silabs-hfegran