openhwfoundation/core-v-verif #1878: Verification goal for FENCEI feature in CVA6 DV plan

MikeOpenHWGroup, 2023-05-05:
The [verif goals](https://github.com/openhwgroup/core-v-verif/blob/ddcc187fdf34901dd22967a5656a910ddc850258/cva6/docs/VerifPlans/FENCEI/VP_IP001.yml#L25) for this feature does not guarantee that the fencei instruction has worked. If the instruction following the fencei has not yet been pre-fetched then the newy stored value will always be executed. A proper test of a fencei requires the following scenario:

- The instruction immediately before fencei, the fencei instruction and the instruction immediately following fencei are fetched into the CVA6.
- The fencei instruction causes the instruction immediately following fencei to be fetched a second time.
- The test-program checks that the modified, not original instruction immediately following fencei is executed.

YoussraBELAMKADEM, 2023-05-08:
If I really assimilate your note, you are suggesting the hole verif plan should be updated (this is exacly what I'm working on ; updating in parallel with the verification process).

MikeOpenHWGroup, 2023-09-14:
Hi @YoussraBELAMKADEM, what is the status of this Issue?