openhwfoundation/core-v-verif #1920: Hardcoded ISA string to compile riscv-arch-tests

Abdulwadoodd, 2023-05-22:
Each test suite from [riscv-arch-tests](https://github.com/riscv-non-isa/riscv-arch-test) has ISA string that is only required by that specific extension and that string is scanned by RISCOF and passed to the `march` flag of toolchain while compiling the test.

For example, c.ebreak test has the ISA string `RV64IC_Zicsr` available [here](https://github.com/riscv-non-isa/riscv-arch-test/blob/main/riscv-test-suite/rv64i_m/C/src/cebreak-01.S#L13), and when it is compiled using RISCOF, ISA string will pass to the march flag as: `-march=rv64ic_zicsr`.

As core-v-verif does not use RISCOF to run arch-tests, it has a hardcoded ISA string (for example [here](https://github.com/openhwgroup/core-v-verif/blob/cva6/dev/cva6/sim/cva6.py#L859)) containing all the extension supported by the target configuration
I may explicitly mention that this is not a bug and it is working fine this way. 
The only little problem is that when it compiles the tests that do not intend to include any `C` instructions, this hardcoded ISA string (rv64gc) gets compiled such that it may compress some I-type instructions.
Another thing is that when we want to run tests of some non-general extensions such as BitManip, the toolchain requires to have corresponding zba, zbb, zbc, zbs as part of ISA string.
So, it is just a better approach that `march` comes from a specific test suite that needs to get compiled. Since core-v-verif uses yaml based framework instead of RISCOF, this can be adapted into that yaml-based framework.

MikeOpenHWGroup, 2023-05-23:
Hi @Abdulwadoodd.  This is a known issue with the method used by the CVA6 to compile test-programs.  The method used by CV32E cores in CORE-V-VERIF has solved this issue and I plan to migrate the CVA6 to this method in the coming months.  (Having said that, it is a low priority task at the moment and does not appear on @JeanRochCoulon's task list for CV32A60X verification.)

If you would like to know more about this, have a look at the [test program definitions](https://github.com/openhwgroup/core-v-verif/blob/master/mk/TOOLCHAIN.md#test-program-definitions) section of the [TOOLCHAIN](https://github.com/openhwgroup/core-v-verif/blob/master/mk/TOOLCHAIN.md) README and [Writing Tests for CORE-V-VERIF](https://docs.google.com/presentation/d/18IHmCRNLtPp16tTtUlwtVCwZwDvlgtmu/edit?usp=sharing&ouid=112401338207375688045&rtpof=true&sd=true) (starting on slide #14).

Abdulwadoodd, 2023-08-08:
Completed by PR #2075

closed 2023-08-08 by Abdulwadoodd