openhwfoundation/core-v-verif #1933: Mention of UVM run_phase in uvm_tests.rst is incorrect

MikeOpenHWGroup, 2023-05-25:
Email from dave.rich@siemens.com:
```
I was reading the text on this page: https://github.com/openhwgroup/core-v-verif/blob/master/docs/VerifStrat/source/uvm_tests.rst

In the run phase the base test will assert the fetch_en input to the core which signals it to start running. The timing of this is randomized but keep in mind that it will always happen after reset is de-asserted (because resets are done in the reset phase, which always executes before the run phase).

This is not correct. The UVM’s run_phase starts at time 0 in parallel with the reset_phase. You use run_phase with components that do not need to be aware of other phases, or if your sequences are set up to be unaware of phases. (i.e. you would start a reset_sequence in the run_phase, followed by other sequences. Usually monitors are not aware of other phases.

But if you do choose to use the reset_phase, then you would use the main_phase instead of the run_phase.

See https://www.chipverify.com/uvm/uvm-phases

-Dave
```

subbu009, 2024-05-15:
As Dave said, in UVM the `run_phase `starts in parallel with other run time phases like `reset_phase `and `configure_phase`. One of the base testcase [here ](https://github.com/openhwgroup/core-v-verif/blob/master/cv32e40p/tests/uvmt/base-tests/uvmt_cv32e40p_base_test.sv) uses `run_phase `which only contains `watchdog_timer()`, which is essentially used to stop simulation if timeout is reached. So, according to this testcase it works fine even the other run time phases are used along with `run_phase`. But if some functional sequence needs to be started in `run_phase`, then as Dave said `main_phase `must be used to run after `configure_phase`. Correct me if I am wrong.

subbu009, 2024-05-16:
Yes, I am interested. Help me get started

subbu009, 2024-05-29:
Hi @MikeOpenHWGroup , 
If we replace the `run_phase `with `main_phase`, then all the three phases `reset_phase`, `configure_phase `and `main_phase `will run one after the other as expected. I guess this will fix the issue explained by Dave.

About the "on-the-fly" reset, could you please elaborate more on the task as what is expected?

MikeOpenHWGroup, 2024-05-29:
> If we replace the run_phase with main_phase, then all the three phases reset_phase, configure_phase and main_phase will run one after the other as expected.

Yes, that is my understanding as well.

> About the "on-the-fly" reset, could you please elaborate more on the task as what is expected?

Essentially this means randomly asserting reset at some time after the initial reset is de-asserted.  This implies that the RTL will be executing a test-program and then jump back to the initial boot-address and re-start executing the same test-program.  These on-the-fly resets are handy for detecting un-initialized conditions in the RTL.  Typically, the testbench will have a large amount of state and trying to figure out which should be discarded and which should be kept (if any) when an on-the-fly reset occurs is a difficult problem.