openhwfoundation/core-v-verif #1935: ISS: 0.9.0, variant CV32E40S/40S_DEV mismatch

silabs-hfegran, 2023-05-26:
## hello-world mismatches on first instruction with ISS binary supporting v.0.9.0 when run in 0.8.0 mode

### Type
* Functionally incorrect behavior

### Steps to Reproduce
1. core-v-verif - cv32e40s/dev hash: 0ee61705054a00c32515df47931e13b8baa30a11
2. cherry-pick commit 62277456b7b2e7393e5ee22f4f956ab13e466123 from PR #1929 
3. IMPERAS_HOME pointing to hash version b7a211b52cebbf92065e7c907509e2b1da7b9b75  (branch: b_0_9_0) 
4. No other changes to any source files
5. make test TEST=hello_world USE_ISS=1
6. Logfile below

If running the above mentioned hash with the old ISS binary, no issues are observed and the test passes.

### Interesting/Notable events in log:

#### GPR does not get updated in ref. model 
This error is consistent on same test for variant CV32E40S_DEV if test bench and RTL is updated to match 0.9.0 spec, but I will keep both issues in one single issue as I do not have an easy way to reproduce the same errors on the CV32E40S_DEV at this point. (I do not want to merge in the 0.9.0 changes until I am certain that basic tests can be run in that configuration)
All logs are from runs with the CV32E40S variant (0.8.0)
```
Info (IDV) Instruction executed prior to mismatch '0x80(_start+0): 00013197 auipc   x3,0x13'
Error (IDV) GPR register value mismatch (HartId:0, PC:0x00000080 _start+0):
Info (IDV)  0> GPR x3
Info (IDV)   . dut:0x00013080
Info (IDV)   . ref:0x00000000
```

```
Warning (RISCV_ITT) CPU 'refRoot/cpu' 0x00000000 0000     c.illegal: Illegal attempt to use unsupported trigger type 2
```
#### From --showoverrides:
```
--override refRoot/cpu/extension_CVE4S/tdata1_reset=0x28001000 (Uns32) (default=0x68001044) (override) reset value of tdata1
```
#### does not match actual compared value:
```
Info (IDV)  1> CSR 7a1 (tdata1)
Info (IDV)   . dut:0x28001000
Info (IDV)   . ref:0x58000201 type:5(etrigger) dmode:1 hit:0 nmi:0 m:1 u:0 action:1(enter debug mode)
```

### Full log:
[log.txt](https://github.com/openhwgroup/core-v-verif/files/11573928/log.txt)

eroom1966, 2023-05-29:
@silabs-hfegran 
Can you provide some instructions on what commands to run (from my linux shell) in order to reproduce your testcase, I am a little unfamiliar how to apply the cherry pick

Did you try applying my changes without the cherry picking - I cannot see how that would have failed, I have run hundreds of testcases with those changes

silabs-hfegran, 2023-05-30:
> @silabs-hfegran Can you provide some instructions on what commands to run (from my linux shell) in order to reproduce your testcase, I am a little unfamiliar how to apply the cherry pick

```
git checkout cv32e40s/dev 
```

> Did you try applying my changes without the cherry picking - I cannot see how that would have failed, I have run hundreds of testcases with those changes

eroom1966, 2023-06-03:
@silabs-hfegran 

Following your instructions then comparing with my tree - I see the following difference - could this be an issue ?
I have no clue how this got into my clone, I simply took what was in the openhwgroup code

diff -r ./bin/clonetb /home/moore/git/eroom1966/openhwgroup/cv32e40s-dev/cv32e40s_dev/bin/clonetb
80,81c80,81
<   VERIF_ENV_REPO=https://github.com/openhwgroup/cv32e40x-dv.git
<   VERIF_ENV_BRANCH=main
---
>   VERIF_ENV_REPO=https://github.com/silabs-robin/core-v-verif.git
>   VERIF_ENV_BRANCH=cv32e40xdev_subtree_gitmerge

eroom1966, 2023-06-03:
@silabs-hfegran 
Lets have a shared desktop session on Monday to see what is different in our environments

silabs-hfegran, 2023-06-05:
> @silabs-hfegran
> Lets have a shared desktop session on Monday to see what is different in our environments

@eroom1966, Yes, lets discuss at our regular meeting today

It is really odd that our results differ - the diff you pasted above should have no impact whatsoever. Could anything have been omitted from your repository?

silabs-hfegran, 2023-06-05:
Configuration issue - closing

Thank you for your help resolving this issue @eroom1966