openhwfoundation/core-v-verif #1948: RVFI agent : compilation error when disable debug mode

AyoubJalali, 2023-05-31:
Hello,
As i know the debug mode isn't supported in the STEP1 configuration of the CVA6, so i try to disable it from the [uvme_cva6_cfg.sv](https://github.com/openhwgroup/core-v-verif/blob/cva6/dev/cva6/env/uvme/uvme_cva6_cfg.sv), but i got a compilation error says : 
### Error-[NOA] Null object access
### /home/ajalali/core_cntl_agent/core-v-verif/cva6/sim//../../lib/uvm_agents/uvma_rvfi/uvma_rvfi_instr_mon.sv, 208
  The object is being used before it was constructed/allocated.
  Please make sure that the object is allocated before using it.

so i think the problem is we should place the construction of the debug csrs, if where in debug mode (after the condition in [line](https://github.com/openhwgroup/core-v-verif/blob/e3ec071f66542c16a75c3ecee61e0efa62b76ae4/lib/uvm_agents/uvma_rvfi/uvma_rvfi_instr_mon.sv#L210)

MikeOpenHWGroup, 2023-06-05:
Good catch @AyoubJalali.  Actually, I _think_ the best solution here is to only **_check_** debug CSRs if they created.  Your suggested modification seems to achideve that.

@silabs-robin what do you think?

AyoubJalali, 2023-07-03:
Solved PR #1960

commit e54b7af9 (2023-06-12): Fix #1948
closed 2023-07-03 by AyoubJalali