openhwfoundation/core-v-verif #1963: ISS: mcontrol6.hit0/hit1 behavior incorrect

silabs-hfegran, 2023-06-08:
## mcontrol6.hit0/hit1 behavior incorrect

### Type
* Functionally incorrect behavior

### Description
2 issues: 
* mcontrol6.hit is disabled by default ( needs `--override cpu/no_hit=F` to enable the functionality described in UM
* Enabling the hit-bits with `--override cpu/no_hit=T` causes both bits to be set to 1 (2'b11) when an enabled, type 6 address match trigger causes debug entry

### Background
mcontrol6.hit0/hit1 should, according to UM form a 2-bit hit field with WARL { 2'b00, 2'b01 } behavior - I suspect that there has been some confusion with user manual wording for these two bits, 2'b11 is not a legal value.

eroom1966, 2023-06-08:
Hi Henrik, This was partially a hangover from the 0.8.0 spec where it was hardwired 0
However, there is no entry in the "R/W" column in the User Manual
What should this be ?

eroom1966, 2023-06-08:
OK, this is detailed further down as a NOTE, I missed this

silabs-hfegran, 2023-06-08:
> OK, this is detailed further down as a NOTE, I missed this, so to clarify
this 2 bit field should always have the value 2'b01 ?

hit1 should never be 1, hit0 is set by the CPU when we enter debug due to a trigger match with mcontrol6.
Software is able to write 2'b00 or 2'b01 to this field, any other values will cause the register to retain its previous value (WARL)

> I am confused by your statement about 'disabled by default'

--override cpu/no_hit is true by default, causing these bits not to be set when a trigger matches. Does that clear up things?

silabs-hfegran, 2023-06-14:
Resolved, thanks for the fix @eroom1966