openhwfoundation/core-v-verif #1987: ISS: Warning (IDV) WFI has completed, yet no net changes were detected to cause this

silabs-hfegran, 2023-06-19:
### Type
* Warning messages

### Steps to Reproduce
Please provide:
1. cv32e40s/dev https://github.com/openhwgroup/core-v-verif/commit/29653b907e41d691e27e41d0da07211acfce437f
2. Command line  (e.g. `make xrun-riscv-compliance`)
```
make comp_corev-dv gen_corev-dv TEST=corev_rand_interrupt_wfi CV_CORE=cv32e40s CFG=clic_default SIMULATOR=xrun USE_ISS=YES COV=YES RUN_INDEX=51 GEN_START_INDEX=51 RNDSEED=1303599584
make test TEST=corev_rand_interrupt_wfi CV_CORE=cv32e40s CFG=clic_default SIMULATOR=xrun USE_ISS=YES COV=YES RUN_INDEX=51 GEN_START_INDEX=51 RNDSEED=1303599584
```

Below also exhibit issue, but will take a relatively long time to complete:
```
make comp_corev-dv gen_corev-dv TEST=corev_rand_interrupt_wfi_mem_stress CV_CORE=cv32e40s CFG=clic_default SIMULATOR=xrun USE_ISS=YES COV=YES RUN_INDEX=259 GEN_START_INDEX=259 RNDSEED=-1728189559 
make test TEST=corev_rand_interrupt_wfi_mem_stress CV_CORE=cv32e40s CFG=clic_default SIMULATOR=xrun USE_ISS=YES COV=YES RUN_INDEX=259 GEN_START_INDEX=259 RNDSEED=-1728189559 
```
4. Logfile and/or wave-dump info (screen shots can be useful)
```
UVM_INFO @ 16.800 ns : uvmt_cv32e40s_base_test.sv(294) uvm_test_top [BASE TEST] set load_instr_mem
UVM_INFO @ 114.300 ns : uvmt_cv32e40s_firmware_test.sv(167) uvm_test_top [TEST] Started RUN
Warning (IDV) WFI has completed, yet no net changes were detected to cause this
Warning (IDV) WFI has completed, yet no net changes were detected to cause this
Warning (IDV) WFI has completed, yet no net changes were detected to cause this
Warning (IDV) WFI has completed, yet no net changes were detected to cause this
Warning (IDV) WFI has completed, yet no net changes were detected to cause this
Warning (IDV) WFI has completed, yet no net changes were detected to cause this
Warning (IDV) WFI has completed, yet no net changes were detected to cause this
Warning (IDV) WFI has completed, yet no net changes were detected to cause this
Warning (IDV) WFI has completed, yet no net changes were detected to cause this
Warning (IDV) WFI has completed, yet no net changes were detected to cause this
Warning (IDV) WFI has completed, yet no net changes were detected to cause this
Warning (IDV) WFI has completed, yet no net changes were detected to cause this
Warning (IDV) WFI has completed, yet no net changes were detected to cause this
Warning (IDV) WFI has completed, yet no net changes were detected to cause this
Warning (IDV) WFI has completed, yet no net changes were detected to cause this
Warning (IDV) WFI has completed, yet no net changes were detected to cause this
Warning (IDV) WFI has completed, yet no net changes were detected to cause this
Warning (IDV) WFI has completed, yet no net changes were detected to cause this
Warning (IDV) WFI has completed, yet no net changes were detected to cause this
Warning (IDV) WFI has completed, yet no net changes were detected to cause this
Warning (IDV) WFI has completed, yet no net changes were detected to cause this
Warning (IDV) WFI has completed, yet no net changes were detected to cause this

CV32 DONE

Warning (IDV) WFI has completed, yet no net changes were detected to cause this
UVM_INFO @ 115560.300 ns : uvmt_cv32e40s_firmware_test.sv(176) uvm_test_top [TEST] Finished RUN: exit status is 0
```

### Additional context
This should be investigated, as a warning such as this should most likely either not occur or be interpreted as an error.

eroom1966, 2023-06-20:
@silabs-hfegran 
I have taken a look at this, maybe I am misunderstanding the issue.
I see that the following WFI instruction completes, but I cannot see any interrupts changing which would cause this to happen.
Could you take a look at time 70542ns, I see the following

70542.000 ns: RET,0,2238,000015dc,"10500073 wfi                    ",,,,,
70554.000 ns: RET,0,2239,000015e0,"91f6     c.add   x3,x29         ",x3

eroom1966, 2023-06-20:
re-reading the CLIC spec its quite open what can cause a hart to resume execution after a WFI instruction

```
4.9.3. CLIC events that cause the hart to resume execution after Wait for Interrupt (WFI) Instruction
...
WFI can be a NOP and not actually pause hart execution. In addition, implementations can resume execution after a WFI for any other reason. 
```

**for any other reason**

eroom1966, 2023-08-16:
Fixed in ImperasDV hash : 20410de

silabs-hfegran, 2024-01-11:
Resolved

closed 2024-01-11 by silabs-hfegran