openhwfoundation/core-v-verif #2007: Regression script in corev-dv/corev_fp_zfinx_instr_test_for_regr?

MikeOpenHWGroup, 2023-06-23:
As of hash [25c4199](https://github.com/openhwgroup/core-v-verif/commit/25c4199488abcd99afcbd8a2b86d2266dc2a23e0) (on branch cv32e40p/dev) there is a regression script in [cv32e40p/tests/programs/corev-dv/corev_fp_zfinx_instr_test_for_regr/regr.sh](https://github.com/openhwgroup/core-v-verif/blame/318a4df4e3b7055dc5af5a9ebb4c4eeff2f95c9c/cv32e40p/tests/programs/corev-dv/corev_fp_zfinx_instr_test_for_regr/regr.sh#L1).  I know that I reviewed and approved the pull-request that added this file, but now I have some concerns.  Adding a script to run test-programs in the test-program directory is a significant change to the way tests are compiled and run today, so I have a few questions:

1. Is the use if this script documented anywhere (such as a README)?
2. Why was this changed introduced?
3. Do we need a new script to run the `corev_fp_zfinx_instr_test_for_regr` test-program, or can we update an existing script to support it?
4. Can we put the run script in the `bin` with the others?

dd-baoshan, 2023-06-26:
Hi @MikeOpenHWGroup 
Please refer to my inline response. 

1) Is the use if this script documented anywhere (such as a README)?
No, Currently I only put some notes in this script to describe how this script works.

2) Why was this changed introduced?
3) Do we need a new script to run the corev_fp_zfinx_instr_test_for_regr test-program, or can we update an existing script to support it?
During the initial stage when I tried to construct the yaml file that needed for cv_regress, I noticed that it has some limitations in current flow whereby it could not meet my test intention. 
e.g the testcase corev_fp_zfinx_instr_test_for_regr can use to run with different random stream by using runargs for test overriding, I choose in this way is to reduce the number of tests folder into one test folder only.
Current cv_regress requires a yaml file and ths yaml file takes in one testname for a testcase (in this case corev_fp_zfinx_instr_test_for_regr). This also mean if there are many different runargs, all of them are still using the same testname (users can put multiple same testname with different runargs but it will takes one out of them)
BUT if let said  multiple testname is allow and this will lead to another issue whereby same testname using same result path regardless the runargs input. Currently the yaml file does not have a way to define different run folder.

During that time, I did not spend much time to further root casue the flow limitation and improve it becasue i wanted to focus on tests development, regression, debug and coverage for project delivery. 

4) Can we put the run script in the bin with the others?
Yes, we can but is not recomended because this script is very specific to one particular test and it is not for common use on other tests.
I agreed that this script might cause unneccessary concerns to others. I am fine to remove it from dev and keep in my local space for local regression run.
As for now, the cv_regress is not able to generate the script for regression run for my testcase. I heard from Xavier that he is adding some addtional features for cv_regress flow but I do not think it can resolve my current issue. I will check with him and in case it still cannot resolve, the only way to fix this is to break the single test folder into multiple tests folder.

Please advise further on this. Thanks.

dd-vaibhavjain, 2023-06-26:
I agree with @MikeOpenHWGroup , this script is something which is divergent from the existing regression flow.

To summarize the problem : you want to run 1 test with multiple command line PLUSARGS without needing to create unique "tests" for each of these. Since this is only a runtime argument and can be handled by PLUSARGS.
 
The way this is done in test dir and also the script itself is not a good mechanism as i see. If @dd-baoshan we do this way, consider a case where there is a test failure and you want to reproduce the failure.
Now by using this mechanism, you cant reproduce the failure just by Test name and the seed only. You will have to read the run command arguments and check if something was passed directly in command line (such as what you have done) using CFG_PLUSARGS. And this is not a good solution.

Ideally in UVM we can create a sequence lib and use sequence lib with different configs to achieve what you are trying here.
But for corev-dv there is NO sequence lib, infact there is no run_phase itself, everything is just limited to randomization to dump out a list of random ASM instruction string!

I actually was also facing a similar issue  and have come up with 2 solutions for this and have tried them locally over last week:
I just want to give a brief description of these 2 solutions below here: (Both are independent of each other with no dependency)

1.)  We create unique test run dir, logs, etc based on CFG_PLUSARGS and achieve this by a grouping plusargs under a name. For example:
ifeq ($(CFG_GROUP_ALIAS), float_en)
CFG_PLUSARGS += +enable_floating_point=1
endif
And use this group name as suffix to the test run dir, logs , coverage db etc.
For this we need to just update the flow a little and setup a location to create a list of these CFG_PLUSARGS, possibly also under regress directory.

I had done this locally recently and it works, so i can propose this by a new PR and see if this seems like an acceptable solution to everyone.

2.) I have also tried another solution for this, by picking a random Stream from a list of library streams mentioned in corev.dv.yaml file (mimicking a bit of sequence lib concept) and just letting randomization "seed" pick a random stream each time.
This way a Test and seed are enough to re-run/re-create any failure and also keeps current flow as it is.

XavierAubert, 2023-06-26:
Hello everyone, 

I think they may be also a 3rd option. Back in april, in PR #1786, I added a mechanism to load different test configuration yaml that somehow works like RTL configuration "loading", i.e. specifying a base filename as a configuration, and the Makefile will use that configuration to input whatever is inside the file. 

I only "restricted" that for riscvdv tests at first (by its name), but this behaviour can also be extended to handle tests configurations for sure. 
The main issue is that we need to create one file per configuration for each test, so that might lead to lot of work, but we have a fine grain control for each test. It somehow works like your first solution, but is not packed inside the same directory

For the already present solutions, I like the less the way of picking the random stream from a list, because in some cases maybe only 20-30 tests can be enough for one stream to cover everything needed, and one other stream should run 100s of time and I think we won't be able to control that with this solution.

dd-vaibhavjain, 2023-06-27:
I had an offline discussion with @XavierAubert and wanted to clarify a bit. 
What he suggested as 3rd option was infact same as what was suggested as option 1, with all these common test configs coming through a yaml file <test_cfg>.yaml file and i agree this is a cleaner way. 
@XavierAubert is working on this to update the flow for this. 
These test_cfg.yaml are more like some additional run time configs that can be used to create more combinations with the main test.

MikeOpenHWGroup, 2023-07-05:
@dd-vaibhavjain has mentioned a significant problem with corev-dv (rsicv-dv).  It is _not_ driven by sequences, as it should be.  This is because riscv-dv is a UVM _Object_ (extended from uvm_object), not a UVM _Component_ (extended from uvm_component).  There is a solution to this, but it would mean OpenHW would need to fork and maintain its own implementation of riscv-dv.

I think the workaround suggested by @XavierAubert is probably the best solution.  (FWIW, the CV32E40S team is doing the same thing.)

dd-vaibhavjain, 2023-07-13:
PR #2029 has the the updates to core-v-verif make files to support "test configuration" from a separate yaml file.
It is doing what we have discussed above. 
Also verified the flow with simulations for different combinations of:
- CFG= 
- corev_dv tests
- custom tests
- simulators (vsim, xrun, vcs)
- coverage with vsim
- waves
- TEST_CFG_FILE(new)

At the same time the backward compatibility of current flow is maintained and the exisiting commands/structure etc are still the same. The new updates don't alter anything which exists now.

dd-vaibhavjain, 2023-07-13:
@dd-baoshan If these changes are approved and merged, we can probably remove the regr.sh scripts and create new yaml files for any additional runtime args that needs to be passed and combine them in a new regress/ .yaml file.

dd-baoshan, 2023-07-13:
Yes, I will create new test yaml for fpu once the combination updates from PR #2029 and Xavier are in.
The local regress script will be deleted as well.

dd-baoshan, 2023-08-10:
this can be closed once PR #2117 is meregd

MikeOpenHWGroup, 2023-08-16:
Agreed.  Thanks all.

closed 2023-08-16 by MikeOpenHWGroup