openhwfoundation/core-v-verif #2013: Zcmb no longer present in CV32E40X - but appears in configuration

eroom1966, 2023-06-27:
The Zcmb extension was removed in version 1.0.0
But it still appears in the testbench and breaks the simulation run when trying to apply a non-existant configuration setting

https://github.com/openhwgroup/core-v-verif/blob/cv32e40x/dev/cv32e40x/env/uvme/uvma_cv32e40x_core_cntrl_agent.sv
```
  if (cfg.ext_zcmb_supported) begin
    $fwrite(fh, $sformatf("--override %s/Zcmb=1\n", refpath));
  end else begin
    $fwrite(fh, $sformatf("--override %s/Zcmb=0\n", refpath));
  end
```

silabs-hfegran, 2023-06-27:
Noted, I will fix this

silabs-hfegran, 2023-06-27:
### Workflow change REQUIRED for 40X

This is already fixed when running with the appropriate code-base. Note that 40X simulations require a modified flow, I presume you are running on the cv32e40x/dev-branch, which is deprecated. This change was introduced very recently.

To run tests on the X core now, you will have to checkout the latest `cv32e40s/dev`-branch (this is not an error, the S branch is intentional), and from top level run
 `./bin/clonetb -x`
 Then, you can proceed to run tests as normal from `./cv32e40x/sim/uvmt` as long as your CV_CORE is set to the appropriate core. 
 
 The change does not apply to the 40S as of now. 
 
 X-specific PRs should also be made to the repo that is cloned in by the aforementioned clonetb script . 
 ( ` https://github.com/openhwgroup/cv32e40x-dv `)

eroom1966, 2023-06-27:
OK, does this mean there is also a related change to the cv32e40s workflow ?

silabs-hfegran, 2023-06-27:
> OK, does this mean there is also a related change to the cv32e40s workflow ?

No, no change to 40s as of now (There may be in the future, but we have no immediate plans)

eroom1966, 2023-06-28:
This is not working for me - this is what I did, but I still get an error

```
git clone https://github.com/Imperas/ImperasDV-OpenHW.git
THISDIR=$(pwd)/ImperasDV-OpenHW
export IMPERAS_HOME=${THISDIR}/Imperas
export IMPERAS_ARCH=Linux64
export IMPERAS_VLNV=${IMPERAS_HOME}/lib/${IMPERAS_ARCH}/ImperasLib
export IMPERAS_RUNTIME=CpuManager
export IMPERAS_PERSONALITY=CPUMAN_DV_ASYNC

# setup -xcelium_2107
git clone https://github.com/openhwgroup/core-v-verif -b cv32e40s/dev cv32e40x_dev
export SIMULATOR=xrun
export PATH=${PATH}:$(pwd)/${CV_CORE}_dev/bin
export COREV=YES
export CV_SW_TOOLCHAIN=/home/moore/riscv/riscv
export CV_SW_PREFIX=riscv32-corev-elf-
export CV_SW_VENDOR=corev
pushd cv32e40x_dev
    ./bin/clonetb -x
popd
pushd cv32e40x_dev/cv32e40x/sim/uvmt
     make sanity CV_CORE=cv32e40x
popd
```

I get a lot of errors, I will include the log file here
[sanity.log](https://github.com/openhwgroup/core-v-verif/files/11893213/sanity.log)

silabs-robin, 2023-07-03:
Hi @eroom1966 .
We are still transitioning to this new "[x-dv](https://github.com/openhwgroup/cv32e40x-dv)" repo, and we have not ironed out every detail yet.
Sorry that you got caught in that.

I will send you a workaround shortly.
(And after that I'll PR a fix.)

--------

Hi @MikeOpenHWGroup, I'm just pinging you to show that this case is a good example of why `ci_check` on Github is a good idea.
The problem could have been caught by automation instead of being caught by Lee.

silabs-robin, 2023-07-04:
Update1:

Use hash `bcfeb056f665f6c769e160d4067aad3ab77933df` of core-v-verif.
On that commit, using `./bin/clonetb -x` etc should work like Henrik described it.

With that, the original problem should be gone.
The only `zcm*` in https://github.com/openhwgroup/cv32e40x-dv/blob/main/env/uvme/uvma_cv32e40x_core_cntrl_agent.sv is `zcmp` and `zcmt`.

--------

Update 2:
_(I apparently missed the "comment" button after writing Update1 this morning.)_
We'll soon have a better fix after https://github.com/openhwgroup/core-v-verif/pull/2025 and other upcoming PRs.

silabs-robin, 2023-07-06:
Hi @eroom1966, @silabs-hfegran, and @MikeOpenHWGroup.
Final update about `cv32e40x/dev` problems:

The following standard flow should work:
1. Check out `cv32e40x/dev`
2. Run `./bin/clonetb -x`

Extra info:
Thanks to https://github.com/openhwgroup/core-v-verif/pull/2030 `cv32e40x/dev` should be comfortably decoupled from both `cv32e40s/dev` and `cv32e40x-dv`.
We are working on 1) deleting the `core-v-verif/cv32e40x/` directory, and 2) adding a readme that explain the above two-step workflow.

--------

This ticket can be closed, AFAICT.
But I'll defer closing to the creator of the ticket.

silabs-robin, 2023-12-11:
@eroom1966 It's ok if I close this, right?

eroom1966, 2023-12-11:
fine to close - thx

closed 2023-12-11 by eroom1966