openhwfoundation/core-v-verif #2015: ImperasDV metrics should be considered in simulation pass/fail determination

aimeepsutton, 2023-06-27:
Dolphin Design recently experienced an issue where a bug in their testbench resulted in zero instruction retirements reported to ImperasDV. The check-in sanity tests they ran all passed (vacuously) and so they were unaware of the problem.

This situation can be avoided by including ImperasDV metrics in the [pass/fail determination function](https://github.com/openhwgroup/core-v-verif/blob/38b1745c79b2286a523b636802c6c7bfbb50c6df/cv32e40x/tb/uvmt/uvmt_cv32e40x_tb.sv#L776C1-L776C1) in core-v-verif. Currently only UVM errors/warnings are being considered in this function, but it should be updated to check, for example, that the number of instructions retired is non-zero.

The RVVI API provides a function called rvviRefMetricGet() that can be used to obtain simulation metrics from ImperasDV. Here is the function prototype, and the metrics which can be queried. 

[image]

[image]

Documentation above comes from  [https://www.riscv-verification.org/docs/rvvi/doxygen](https://www.riscv-verification.org/docs/rvvi/doxygen)

MikeOpenHWGroup, 2023-06-29:
Back in the OVPsim days, the UVM environment maintained a count of retired instructions and as part of the end-of-sim checks would call `uvm_error()` if the number of retired instructions was zero.  This should be reinstated so that any runs of CORE-V-VERIF that exclude ImperasDV will also fail if no instructions are retired.

It would also be a good idea to implement the rvviRefMetricGet() query as well since most users of CORE-V-VERIF for the CV32E4 cores will be (already are) using ImperasDV.  Just a note about implementation:

> This situation can be avoided by including ImperasDV metrics in the [pass/fail determination [...]

It is important that any and all conditions that could cause a failing test be flagged as UVM errors or warnings. So the pass/fail function mentioned by @aimeepsutton should check the state of ImperasDV and then raise errors/warnings as required.

dd-vaibhavjain, 2023-07-20:
I have the updates for this. 
Due to lot of backlog of open PRs for core-v-verif cv32e40p/dev I have not pushed my updates for this,  just waiting for those open ones to close for now.

dd-vaibhavjain, 2023-07-27:
Thanks @aimeepsutton @MikeOpenHWGroup !
This checking for rvvi metrics is now added to the **core-v-verif**'s cv32e40p testbench (currently in branch cv32e40p/dev).
For every test that runs with reference model (i.e, with USE_ISS=yes) , these checks will be enabled and would check for various non-zero results from rvviRefMetricGet() at the end of the simulation, and would otherwise raise an error and cause Simulation end-result to indicate a "simulation failure".

XavierAubert, 2023-08-10:
@MikeOpenHWGroup If I'm not mistaken, this issue can be closed because https://github.com/openhwgroup/core-v-verif/pull/2081 has already been merged.

pascalgouedo, 2023-08-10:
Hi,
Only done for E40P. Need to be done for other cores...
Just wondering if it needs to be done core by core or even better in central area files...

MikeOpenHWGroup, 2023-08-14:
Thanks everyone.  @pascalgouedo is correct that this work should also be done for the other projects that are using ImperasDV.  However, this specific issue, for the CV32E40P environment, can be closed and I will open a similar issue for the other cores.

closed 2023-08-14 by MikeOpenHWGroup