openhwfoundation/core-v-verif #2034: Errror while "make veri-test TEST=hello-world"

monahashemii, 2023-07-07:
Hi, 
I'm new to core-v-verif and trying to make a test simulation by 'make' inside 'core-v-verif/cv32e40p/sim/core', but facing this message:
'**Error: Exiting due to 7 warning(s)**'.

I was wondering if you could help me...

**These are warnings:**

*******************************************************************************************
* Compiling CORE TB and CV32E40P with Verilator
*******************************************************************************************
verilator --cc --sv --exe \
         \
        --Wno-lint --Wno-UNOPTFLAT \
        --Wno-MODDUP --top-module \
        tb_top_verilator /home/core-v-verif/cv32e40p/tb/core/tb_top_verilator.sv /home/core-v-verif/cv32e40p/tb/core/cv32e40p_tb_wrapper.sv /home/core-v-verif/cv32e40p/tb/core/tb_riscv                          /riscv_rvalid_stall.sv /home/core-v-verif/cv32e40p/tb/core/tb_riscv/riscv_gnt_stall.sv /home/core-v-verif/cv32e40p/tb/core/mm_ram.sv /home/core-v-verif/cv32e40p/tb/core/dp_ram.sv \
        -f /home/core-v-verif/core-v-cores/cv32e40p/cv32e40p_manifest.flist \
        /home/core-v-verif/core-v-cores/cv32e40p/bhv/cv32e40p_core_log.sv \
        /home/core-v-verif/cv32e40p/tb/core/tb_top_verilator.cpp --Mdir cobj_dir \
        -CFLAGS "-std=gnu++11 -O2" \
        -Wno-BLKANDNBLK
%Warning-MULTIDRIVEN: /home/core-v-verif/cv32e40p/tb/core/mm_ram.sv:291:9: Variable written to in always_comb also written by other process (IEEE 1800-2017 9.2.2.2): 'data_req_dec'
                                                                                  : ... In instance tb_top_verilator.cv32e40p_tb_wrapper_i.ram_i
                      /home/core-v-verif/cv32e40p/tb/core/mm_ram.sv:291:9:
  291 |         data_req_dec        = '0;
      |         ^~~~~~~~~~~~
                      /home/core-v-verif/cv32e40p/tb/core/mm_ram.sv:196:8: ... Location of other write
  196 |        data_req_dec = data_req_i;
      |        ^~~~~~~~~~~~
                      ... For warning description see https://verilator.org/warn/MULTIDRIVEN?v=5.012
                      ... Use "/* verilator lint_off MULTIDRIVEN */" and lint_on around source to disable this message.
%Warning-MULTIDRIVEN: /home/core-v-verif/cv32e40p/tb/core/mm_ram.sv:292:9: Variable written to in always_comb also written by other process (IEEE 1800-2017 9.2.2.2): 'data_addr_dec'
                                                                                  : ... In instance tb_top_verilator.cv32e40p_tb_wrapper_i.ram_i
                      /home/core-v-verif/cv32e40p/tb/core/mm_ram.sv:292:9:
  292 |         data_addr_dec       = '0;
      |         ^~~~~~~~~~~~~
                      /home/core-v-verif/cv32e40p/tb/core/mm_ram.sv:205:11: ... Location of other write
  205 |           data_addr_dec  = data_addr_i[RAM_ADDR_WIDTH-1:0];
      |           ^~~~~~~~~~~~~
%Warning-MULTIDRIVEN: /home/core-v-verif/cv32e40p/tb/core/mm_ram.sv:293:9: Variable written to in always_comb also written by other process (IEEE 1800-2017 9.2.2.2): 'data_wdata_dec'
                                                                                  : ... In instance tb_top_verilator.cv32e40p_tb_wrapper_i.ram_i
                      /home/core-v-verif/cv32e40p/tb/core/mm_ram.sv:293:9:
  293 |         data_wdata_dec      = '0;
      |         ^~~~~~~~~~~~~~
                      /home/core-v-verif/cv32e40p/tb/core/mm_ram.sv:208:8: ... Location of other write
  208 |        data_wdata_dec = data_wdata_i;
      |        ^~~~~~~~~~~~~~
%Warning-MULTIDRIVEN: /home/core-v-verif/cv32e40p/tb/core/mm_ram.sv:294:9: Variable written to in always_comb also written by other process (IEEE 1800-2017 9.2.2.2): 'data_we_dec'
                                                                                  : ... In instance tb_top_verilator.cv32e40p_tb_wrapper_i.ram_i
                      /home/core-v-verif/cv32e40p/tb/core/mm_ram.sv:294:9:
  294 |         data_we_dec         = '0;
      |         ^~~~~~~~~~~
                      /home/core-v-verif/cv32e40p/tb/core/mm_ram.sv:209:8: ... Location of other write
  209 |        data_we_dec    = data_we_i;
      |        ^~~~~~~~~~~
%Warning-MULTIDRIVEN: /home/core-v-verif/cv32e40p/tb/core/mm_ram.sv:295:9: Variable written to in always_comb also written by other process (IEEE 1800-2017 9.2.2.2): 'data_be_dec'
                                                                                  : ... In instance tb_top_verilator.cv32e40p_tb_wrapper_i.ram_i
                      /home/core-v-verif/cv32e40p/tb/core/mm_ram.sv:295:9:
  295 |         data_be_dec         = '0;
      |         ^~~~~~~~~~~
                      /home/core-v-verif/cv32e40p/tb/core/mm_ram.sv:210:8: ... Location of other write
  210 |        data_be_dec    = data_be_i;
      |        ^~~~~~~~~~~
%Warning-MULTIDRIVEN: /home/core-v-verif/cv32e40p/tb/core/mm_ram.sv:312:9: Variable written to in always_comb also written by other process (IEEE 1800-2017 9.2.2.2): 'transaction'
                                                                                  : ... In instance tb_top_verilator.cv32e40p_tb_wrapper_i.ram_i
                      /home/core-v-verif/cv32e40p/tb/core/mm_ram.sv:312:9:
  312 |         transaction         = T_PER;
      |         ^~~~~~~~~~~
                      /home/core-v-verif/cv32e40p/tb/core/mm_ram.sv:211:8: ... Location of other write
  211 |        transaction    = T_RAM;
      |        ^~~~~~~~~~~
%Warning-COMBDLY: /home/core-v-verif/core-v-cores/cv32e40p/rtl/../bhv/cv32e40p_sim_clock_gate.sv:25:31: Non-blocking assignment '<=' in combinational logic process
                                                                                                               : ... This will be executed as a blocking assignment '='!
   25 |     if (clk_i == 1'b0) clk_en <= en_i | scan_cg_en_i;
      |                               ^~
                  *** See https://verilator.org/warn/COMBDLY before disabling this,
                  else you may end up with different sim results.
%Error: Exiting due to 7 warning(s)
make[1]: *** [Makefile:492: testbench_verilator] Error 1
make[1]: Leaving directory '/home/core-v-verif/cv32e40p/sim/core'
make: *** [Makefile:486: sanity-veri-run] Error 2

taylortempleton, 2023-07-07:
These are Verilator warnings.  My "solution" was to ignore them, for now.
(Admittedly it's a terrible idea to just ignore warnings one doesn't fully
understand, but I did so in service of 'just getting it to run' on my
system.)

In CV32E40P/Sim/Core/Makefile, I added the below warning ignore flags:

# verilator configuration
VERILATOR = verilator
VERI_FLAGS +=
VERI_COMPILE_FLAGS += -Wno-BLKANDNBLK -Wno-MULTIDRIVEN -Wno-COMBDLY $(
SV_CMP_FLAGS) # hope this doesn't hurt us in the long run

monahashemii, 2023-07-07:
Thank you so much.
But now, it results another error after 'make' inside 'core-v-verif/cv32e40p/sim/core'. 

I just want to test by 'make veri-test TEST=hello-world'...

*******************************************************************************************
* Compiling the BSP
*******************************************************************************************
mkdir -p simulation_results/hello-world/0/test_program/bsp
cp /home/core-v-verif/cv32e40p/bsp/Makefile simulation_results/hello-world/0/test_program/bsp
make -C simulation_results/hello-world/0/test_program/bsp \
        VPATH=/home/core-v-verif/cv32e40p/bsp \
        RISCV=/opt/riscv \
        RISCV_PREFIX=riscv32-unknown-elf- \
        RISCV_EXE_PREFIX=/opt/riscv/bin/riscv32-unknown-elf- \
        RISCV_MARCH=rv32imc \
        RISCV_CC=gcc \
        RISCV_CFLAGS="" \
        all
make[3]: Entering directory '/home/core-v-verif/cv32e40p/sim/core/simulation_results/hello-world/0/test_program/bsp'
/opt/riscv/bin/riscv32-unknown-elf-gcc -Os -g -static -mabi=ilp32 -march= -Wall -pedantic -c /home/core-v-verif/cv32e40p/bsp/crt0.S -o crt0.o
riscv32-unknown-elf-gcc: error: missing argument to '-march='
make[3]: *** [Makefile:20: crt0.o] Error 1
make[3]: Leaving directory '/home/core-v-verif/cv32e40p/sim/core/simulation_results/hello-world/0/test_program/bsp'
make[2]: *** [/home/core-v-verif/mk/Common.mk:515: bsp] Error 2
make[2]: Leaving directory '/home/core-v-verif/cv32e40p/sim/core'
make[1]: *** [/home/core-v-verif/mk/Common.mk:486: /home/core-v-verif/cv32e40p/tests/programs/custom/hello-world/hello-world.elf] Error 2
make[1]: Leaving directory '/home/core-v-verif/cv32e40p/sim/core'
make: *** [Makefile:489: sanity-veri-run] Error 2

taylortempleton, 2023-07-07:
Sure I had the same set of errors.  You're almost there.

Your next issue is described here:

https://github.com/openhwgroup/core-v-verif/issues/1594

The lazy solution is to set a variable by typing:
export CV_SW_MARCH=rv32imc_zicsr
at the command line, and then trying your make command again. You'd have to
set this variable every time you re-open the environment.

The more robust solution, as Mike pointed out, is to use the latest version
of the Embecosm pre-built toolchain.  I struggled with this, and used a
variable, for now.

taylortempleton, 2023-07-07:
Apologies, I retract my last answer.  This may be a different issue.

MikeOpenHWGroup, 2023-07-12:
Hi @monahashemii, I notice that your compile of the BSP is not setting MARCH at all.  As @taylortempleton suggest, you should set a shell environment variable `export CV_SW_MARCH=rv32imc_zicsr`.

monahashemii, 2023-07-19:
Thanks @MikeOpenHWGroup but not solved and the same errors are still there...

MikeOpenHWGroup, 2023-07-19:
This is surprising.  Looking at [core-v-verif/cv32e40p/bsp/Makefile](https://github.com/openhwgroup/core-v-verif/blob/8fb291707e97a82582435bf845d8c5f0dc7c3e17/cv32e40p/bsp/Makefile#L9) we see that `march` is set to the value of `$CV_SW_MARCH`:
```
CFLAGS ?= -Os -g -static -mabi=ilp32 -march=$(CV_SW_MARCH) -Wall -pedantic
```
Your logfile indicates that `$CV_SW_MARCH` is not set:
```
/opt/riscv/bin/riscv32-unknown-elf-gcc -Os -g -static -mabi=ilp32 -march= -Wall -pedantic -c /home/core-v-verif/cv32e40p/bsp/crt0.S -o crt0.o
```
In the above, `-march=` is not what you want to see.  You need `-march=rv32imc_zicsr`.

As for the Verilator warnings, these are know-and-willfully-ignored issues.  The "core" testbench is not our production environment and we do not intend to maintain it (we would accept a pull-request if you are interested).

monahashemii, 2023-07-20:
I tried that and set march but not solved and the same errors are still there...:

>echo $CV_SW_MARCH
rv32imc_zicsr

>make
Running with RISCV=/root/riscv-tools-install
mkdir -p /root/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig
cd /root/chipyard && java -Xmx8G -Xss8M -XX:MaxPermSize=256M -Djava.io.tmpdir=/root/chipyard/.java_tmp -jar /root/chipyard/generators/rocket-chip/sbt-launch.jar -Dsbt.sourcemode=true -Dsbt.workspace=/root/chipyard/tools  ";project chipyard; runMain chipyard.Generator  --target-dir /root/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig --name chipyard.TestHarness.RocketConfig --top-module chipyard.TestHarness --legacy-configs chipyard:RocketConfig"
OpenJDK 64-Bit Server VM warning: Ignoring option MaxPermSize; support was removed in 8.0
[info] welcome to sbt 1.4.9 (Ubuntu Java 11.0.11)
[info] loading settings for project chipyard-build from plugins.sbt ...
[info] loading project definition from /root/chipyard/project
[info] loading settings for project chipyardRoot from build.sbt ...
[info] loading settings for project hardfloat from build.sbt ...
[info] loading settings for project rocketConfig from build.sbt ...
[info] loading settings for project firrtl_interpreter from build.sbt ...
[info] loading settings for project treadle from build.sbt ...
[info] loading settings for project chisel_testers from build.sbt ...
[info] loading settings for project testchipip from build.sbt ...
[info] loading settings for project icenet from build.sbt ...
[info] loading settings for project hwacha from build.sbt ...
[info] loading settings for project boom from build.sbt ...
[info] loading settings for project cva6 from build.sbt ...
[info] loading settings for project sodor from build.sbt ...
[info] loading settings for project gemmini from build.sbt ...
[info] loading settings for project mdf from build.sbt ...
[info] loading settings for project barstoolsMacros from build.sbt ...
[info] loading settings for project sim-build from plugins.sbt ...
[info] loading project definition from /root/chipyard/sims/firesim/sim/project
[info] loading settings for project firesim from build.sbt ...
[info] loading settings for project targetutils from build.sbt ...
[info] loading settings for project midas from build.sbt ...
[info] loading settings for project chisel3-build from plugins.sbt ...
[info] loading project definition from /root/chipyard/tools/chisel3/project
[info] loading settings for project chisel from build.sbt ...
[info] loading settings for project firrtl-build from plugins.sbt ...
[info] loading project definition from /root/chipyard/tools/firrtl/project
[info] loading settings for project firrtl from build.sbt ...
[info] resolving key references (55373 settings) ...
[info] set current project to chipyardRoot (in build file:/root/chipyard/)
[info] set current project to chipyard (in build file:/root/chipyard/)
[info] compiling 69 Scala sources to /root/chipyard/tools/treadle/target/scala-2.12/classes ...
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/Driver.scala:6:8: object ExecutionOptionsManager is not a member of package firrtl
[error] import firrtl.{AnnotationSeq, ExecutionOptionsManager, HasFirrtlOptions}
[error]        ^
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/Driver.scala:32:20: type ComposableOptions is not a member of package firrtl
[error]     extends firrtl.ComposableOptions {
[error]                    ^
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/Driver.scala:34:41: not found: type ExecutionOptionsManager
[error]   def vcdOutputFileName(optionsManager: ExecutionOptionsManager): String = {
[error]                                         ^
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/Driver.scala:73:9: not found: type ExecutionOptionsManager
[error]   self: ExecutionOptionsManager =>
[error]         ^
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/Driver.scala:77:3: not found: value parser
[error]   parser.note("treadle-options")
[error]   ^
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/Driver.scala:79:3: not found: value parser
[error]   parser
[error]   ^
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/Driver.scala:87:3: not found: value parser
[error]   parser
[error]   ^
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/Driver.scala:95:3: not found: value parser
[error]   parser
[error]   ^
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/Driver.scala:103:3: not found: value parser
[error]   parser
[error]   ^
[... 132 lines cut ...]
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/vcd/VCDConfig.scala:48:3: not found: value parser
[error]   parser
[error]   ^
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/vcd/VCDConfig.scala:56:3: not found: value parser
[error]   parser
[error]   ^
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/vcd/VCDConfig.scala:64:3: not found: value parser
[error]   parser
[error]   ^
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/vcd/VCDConfig.scala:73:3: not found: value parser
[error]   parser
[error]   ^
[error] /root/chipyard/tools/treadle/src/main/scala/treadle/vcd/VCDConfig.scala:82:57: no arguments allowed for nullary constructor Object: ()Object
[error] class VCDOptionsManager extends ExecutionOptionsManager("vcd") with HasVCDConfig
[error]                                                         ^
[warn] /root/chipyard/tools/treadle/src/main/scala/treadle/vcd/VCDConfig.scala:5:15: Unused import
[warn] import firrtl.ExecutionOptionsManager
[warn]               ^
[warn] four warnings found
[error] 55 errors found
[error] (treadle / Compile / compileIncremental) Compilation failed
[error] Total time: 8 s, completed Jul 20, 2023, 12:42:14 AM
/root/chipyard/common.mk:105: recipe for target 'generator_temp' failed
make: *** [generator_temp] Error 1
root@fe0642bf7bd2:~/chipyard/sims/verilator#

monahashemii, 2023-07-20:
> > make
> > Running with RISCV=/root/riscv-tools-install
> > mkdir -p /root/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig
> > cd /root/chipyard && java -Xmx8G -Xss8M -XX:MaxPermSize=256M -Djava.io.tmpdir=/root/chipyard/.java_tmp -jar /root/chipyard/generators/rocket-chip/sbt-launch.jar -Dsbt.sourcemode=true -Dsbt.workspace=/root/chipyard/tools ";project chipyard; runMain chipyard.Generator --target-dir /root/chipyard/sims/verilator/generated-src/chipyard.TestHarness.RocketConfig --name chipyard.TestHarness.RocketConfig --top-module chipyard.TestHarness --legacy-configs chipyard:RocketConfig"
> 
> These are not errors from CORE-V-VERIF. It looks like [Chipyard](https://github.com/ucb-bar/chipyard)?

Sorry, you're right. Working on two projects at the same time :D 
Thanks, that works now.

closed 2023-07-20 by monahashemii