openhwfoundation/core-v-verif #2066: ISS: Reserved csr instructions with mnxti are not considered illegal instructions

silabs-hfegran, 2023-07-20:
## Executing illegal csr-instructions with mnxti is not handled as an illegal instruction by the ISS

### Type
* Functionally incorrect behavior

### Steps to Reproduce
1. cv32e40s/dev, hash: f9c3848dcacbb4cb5d50f20de311e6f0ef82819e
2. Command line: `make test TEST=clic CFG=clic_default USE_ISS=1 CV_CORE=cv32e40s`

### Additional context
All  CSR-instructions with `mnxti` except `CSRRS rd, csr, x0`, `CSRRSI`, `CSRRCI` is reserved and should be considered illegal instructions.

From the clic spec (and repeated in the user manual):
```
These CSRs are only designed to be used with the CSRR (CSRRS rd,csr,x0), CSRRSI, and CSRRCI instructions. Accessing the xnxti CSR using any other CSR instruction form (CSRRW/CSRRS,rs1!=x0/CSRRC/CSRRWI) is reserved. Note: Use of xnxti with CSRRSI with non-zero uimm values for bits 0, 2, and 4 are reserved for future use.
```

User manual adds:
```
(...) and will also be treated as illegal instructions.
```

eroom1966, 2023-07-20:
Hi Henrik, can you please provide a URL to the document you are referring to ?

eroom1966, 2023-07-20:
@silabs-hfegran 
looking at the cv32e40s/x additions to the RISCV international spec. It appears that there are similar constraints for 
- mnxti
- mscratchscwl
- mscratchcsw 

Is there a directed unit test that exists today, that targets these registers with the illegal instructions and illegal values?

silabs-hfegran, 2023-07-21:
> @silabs-hfegran looking at the cv32e40s/x additions to the RISCV international spec. It appears that there are similar constraints for
>
> * mnxti
> * mscratchscwl
> * mscratchcsw
>
> Is there a directed unit [...]

the aforementioned test here does that - this part of the test was added recently

silabs-hfegran, 2023-07-21:
There is an issue with the hash in question here, that may affect other tests (certain c.addi-instructions causing issues with isacov) - issue should have been resolved in this hash: a32459e075c59636300f0e8843476ee3fc481117

silabs-hfegran, 2023-07-31:
Resolved, Thanks @eroom1966

closed 2023-07-31 by silabs-hfegran