openhwfoundation/core-v-verif #2073: ISACOV : No rs2 for c.lw

AyoubJalali, 2023-07-24:
Hello,
I think we should remove [rs2_toggle](https://github.com/openhwgroup/core-v-verif/blob/8fb291707e97a82582435bf845d8c5f0dc7c3e17/lib/uvm_agents/uvma_isacov/cov/uvma_isacov_cov_model.sv#L1198) from **cg_cl** because **c.lw** does not have rs2

MikeOpenHWGroup, 2023-07-24:
Good catch @AyoubJalali.   I agree.  @silabs-robin, what say you?

AyoubJalali, 2023-07-24:
> Good catch @AyoubJalali. I agree. @silabs-robin, what say you?

I suggest a quick fix for this

MikeOpenHWGroup, 2023-07-24:
> I suggest a quick fix for this

I agree but want to hear from @silabs-robin first.

silabs-robin, 2023-07-24:
> I think we should remove [rs2_toggle](https://github.com/openhwgroup/core-v-verif/blob/8fb291707e97a82582435bf845d8c5f0dc7c3e17/lib/uvm_agents/uvma_isacov/cov/uvma_isacov_cov_model.sv#L1198) from cg_cl because c.lw does not have rs2

I agree.

There is no `rs2` in `CL` at all.
And I can only find that `C.LW` and friends are the only users of `CL`, so there shouldn't be any exceptions to this either.

[image]

MikeOpenHWGroup, 2023-07-24:
Awesome!  I looked at the same Figure in the ISA and came to the same conclusion.

@AyoubJalali, can you issue a PR to fix this?  I'll approve/merge as soon as I see it.   Thanks!!

AyoubJalali, 2023-07-24:
> Awesome! I looked at the same Figure in the ISA and came to the same conclusion.
> 
> @AyoubJalali, can you issue a PR to fix this? I'll approve/merge as soon as I see it. Thanks!!

on it

closed 2024-07-10 by AyoubJalali