openhwfoundation/core-v-verif #2079: ISS mismatch after debug+nmi + nmi_handler corrupting mtvec + interrupt + trapped handler + debug request

silabs-hfegran, 2023-07-25:
## ISS mismatch
In this test, the following sequence of events occur:

1. Haltrequest is pending immediately on start
2. During debug, an store zero to an address occurs (no state change of memory, dcsr.nmip gets flagged)
3. Load from aforementioned address is OK, no adverse effects as the write value was already zero
4. Dret is followed by an immediate jump to nmi-handler, whose non-initialized stack pointer (@0x0000_0000) gets overwritten with all zeroes
5. nmi handler completes its execution, but has already corrupted the exception handler.
6. Regular initialization code gets run, followed by setting of mpie and finally an mret-instruction
7. There is a pending, non-shv interrupt, code attempts to jump to address zero (mtvec, previously corrupted by nmi-handler)
8. Core stalls in a continuous sequence of illegal instruction traps, and jumps to address zero (which contains an illegal instruction)
9. After a long sequence of these illegal instruction traps, a debug haltreq becomes pending
10. => ISS fails to find a valid state that matches the DUT.

From what I can see, the RTL appears to behave as expected

### Type
* Functionally incorrect behavior

### Steps to Reproduce
Please provide:
1. cv32e40s/dev - hash 6a8c5d0c401b
2. Command line:
```
make comp_corev-dv gen_corev-dv TEST=corev_rand_data_obi_err_debug CFG=clic_default SIMULATOR=xrun USE_ISS=YES COV=YES RUN_INDEX=122 GEN_START_INDEX=122 RNDSEED=2033414162
make test TEST=corev_rand_data_obi_err_debug CFG=clic_default SIMULATOR=xrun USE_ISS=YES COV=YES RUN_INDEX=122 GEN_START_INDEX=122 RNDSEED=2033414162
```

silabs-hfegran, 2023-07-25:
The following test exhibits similar behavior:
```
make comp_corev-dv gen_corev-dv TEST=corev_rand_data_obi_err_debug CFG=dummy_instr SIMULATOR=xrun USE_ISS=YES COV=YES RUN_INDEX=608 GEN_START_INDEX=608 RNDSEED=9086385 
make test TEST=corev_rand_data_obi_err_debug CFG=dummy_instr SIMULATOR=xrun USE_ISS=YES COV=YES RUN_INDEX=608 GEN_START_INDEX=608 RNDSEED=9086385
```

eroom1966, 2023-08-01:
@silabs-hfegran 
what a testcase !

OK, what is happening here is that there are a large number of illegal instructions that are trapping, basically an illegal at address 0x0, so the traps are effectively re-entrant

At the same time, a number of interrupts are going off.
this increases the search space for the lead up to the next retired instruction

At the moment we have a limit on the search space, and for this testcase it has exceeded it, if I relaxed the limit - the test then passes. 

The short term fix will be to relax the limit, but in addition, we are also optimizing the search space algorithm.
Either way, we have a fix for this problem that I can release shortly.

MikeOpenHWGroup, 2023-08-01:
@eroom1966, @silabs-hfegran, I've been waiting to for the "search space" problem to appear.  :-)  It is a old-and-recurring problem with reference models that need to produce lists of multiple legal outcomes.  A few suggestions:
- @eroom1966 it would be good if ImperasDV generated an unambiguous "search space limit exceeded" error to stdout.
- @silabs-hfegran, we should consider whether this test-program is "excessively re-entrant.  What item from the DVplan requires this many nested traps?

silabs-hfegran, 2023-08-02:
> * @silabs-hfegran, we should consider whether this test-program is "excessively re-entrant.  What item from the DVplan requires this many nested traps?

@MikeOpenHWGroup, This random test should not be, but breaks down under one condition, and that is when we start using the stack prior to its initialization. Debug straight out of reset is fair enough, the issue occur when we get an nmi in this section; the nmi handler is shared with the regular exception handler, and that handler involves backing up the gpr registers to stack. With an uninitialized SP, i.e. address 0x0000_0000, we end up allocating space on the stack downwards from address 0 (inclusive), which causes the exception handler to write data to the first code word of the exception handler address (mtvec, address 0x0000_0000, which in our case is a jump to the actual handler). Returning from the nmi-handler, we return to the regular code flow, where we initialize the registers, then upon enabling interrupts we take a non-vectored interrupts, which should take us to address zero (That got overwritten earlier). The cpu naturally stalls in a loop here, as it is not possible to progress into the handler - it basically excepts on itself; the only possible progression from this state is a debug entry due to a halt-request. 

This is one thing I really do like about random tests - it stimulates cases that would not be considered under any normal circumstances. While this case is firmly in unlikely corner-case land, it is still something we have to spend time on debugging, because as with any mismatch, the reason for it is normally not obvious.

MikeOpenHWGroup, 2023-08-07:
> This random test should not be, but breaks down under one condition, and that is when we start using the stack prior to its initialization.
...
The cpu naturally stalls in a loop here, as it is not possible to progress into the handler - it basically excepts on itself; the only possible progression from this state is a debug entry due to a halt-request.

Yes, this is a "fascinating problem" (if your mind works that way).  Without wanting to create a never-ending debate, I worry that trying to resolve such issues leads down a path of over-engineering with ever-diminishing returns.  Would it not be better to create a restriction in the User Manual to say that "any use of the stack prior to its initialization leads to undefined behaviour"?  That allows us to implement "reasonable constraints" in test-program generation that avoid this scenario.

eroom1966, 2023-08-15:
@silabs-hfegran can this now be closed - I think the (initial) issue is resolved ?

closed 2023-08-18 by silabs-hfegran