openhwfoundation/core-v-verif #2082: ISS: [Mismatch] Pending debug request with nmip and dret fails to detect debug mode as valid next state

silabs-hfegran, 2023-07-26:
When there is a pending nmi in debug mode, upon dret, the ISS fails evaluate the state where the jump to the exception handler is interrupted by this pending debug halt-request

### Type
* Functionally incorrect behavior

### Steps to Reproduce
Please provide:
1. cv32e40s/dev - hash https://github.com/openhwgroup/core-v-verif/commit/6a8c5d0c401b7ba2d04212f630d5bac14c65f927
2. Command line:
```
make comp_corev-dv gen_corev-dv TEST=corev_rand_data_obi_err_debug CFG=dummy_instr SIMULATOR=xrun USE_ISS=YES COV=YES RUN_INDEX=733 GEN_START_INDEX=733 RNDSEED=489439175 
make test TEST=corev_rand_data_obi_err_debug CFG=dummy_instr SIMULATOR=xrun USE_ISS=YES COV=YES RUN_INDEX=733 GEN_START_INDEX=733 RNDSEED=489439175 
```

4. Logfile and/or wave-dump i
[image]
nfo (screen shots can be useful)

```diff
UVM_INFO @ 5946.300 ns : uvma_rvfi_instr_mon.sv(213) uvm_test_top.env.rvfi_agent.instr_monitor0 [RVFIMON] Debug NMIP
Error (IDV) The reference model is unable to find a legal set of actions to converge to the provided DUT state.
Info (IDV) Reference model mode (HartId: 0):
Info (IDV)   Machine - Machine mode
Info (IDV) Currently applied nets:
Info (IDV)  - nmi => 1
Info (IDV)  - irq_id_i => 17
Info (IDV)  - irq_lev_i => be
Info (IDV)  - irq_sec_i => 3
Info (IDV)  - haltreq => 1
Info (IDV) Pending net changes:
Info (IDV)  - Group 3:
Info (IDV)  -          nmi_cause => 1025,  age = 0 cycles, 0 retires
Info (IDV)  - Group 4:
Info (IDV)  -            haltreq => 0,  age = 10 cycles, 0 retires
Info (IDV) DUT state change (hartId: 0):
Info (IDV)  - dut.pc               - 0x1a110800 debug_rom+0
Info (IDV)  - dut.mstatus          - 0x00001880 SD:0 TW:0 MPRV:0 XS:0(Off) FS:0(Off) MPP:3 VS:0(Off) MPIE:1 UBE:0 MIE:0
Info (IDV)  - dut.mepc             - 0x0001c072
Info (IDV)  - dut.mcause           - 0xb8000401 Interrupt:1 minhv:0 mpp:3 mpie:1 mpil:0 Code:1025
Info (IDV)  - dut.dcsr             - 0x400004d3 debugver:4 ebreakm:0 ebreaku:0 stepie:0 stopcount:1 stoptime:0 cause:3(haltreq) mprven:1 nmip:0 prv:3(M-mode)
Info (IDV)  - dut.dpc              - 0x00000000
Info (IDV) Reference state change evaluation tree (HartId: 0):
Info (IDV)  +  Net haltreq => 0x0 { when:2980 }
Info (IDV)     +  Net nmi_cause => 0x401 { when:2990 }
Info (IDV)        +  Exception { pc:0x1c072(sub_3_6_0_t+4), mcause:0xb8000401, mstatus:0x1880 }
Info (IDV)           +  Retire { pc:0x0(__jvt_base$+0) }
Info (IDV)              > dut.pc           - 0x1a110800 debug_rom+0
Info (IDV)                ref.pc           - 0x00000000 __jvt_base$+0
Info (IDV)              > dut.dpc          - 0x00000000
Info (IDV)                ref.dpc          - 0x0001c072
Info (IDV)             
Info (IDV)  +  Net haltreq => 0x0 { when:2980 }
Info (IDV)     +  Exception { pc:0x1c072(sub_3_6_0_t+4), mcause:0xb8000fff, mstatus:0x1880 }
Info (IDV)        +  Net nmi_cause => 0x401 { when:2990 }
Info (IDV)           +  Retire { pc:0x0(__jvt_base$+0) }
Info (IDV)              > dut.pc           - 0x1a110800 debug_rom+0
Info (IDV)                ref.pc           - 0x00000000 __jvt_base$+0
Info (IDV)              > dut.mcause       - 0xb8000401 Interrupt:1 minhv:0 mpp:3 mpie:1 mpil:0 Code:1025
Info (IDV)                ref.mcause       - 0xb8000fff Interrupt:1 minhv:0 mpp:3 mpie:1 mpil:0 Code:4095
Info (IDV)              > dut.dpc          - 0x00000000
Info (IDV)                ref.dpc          - 0x0001c072
Info (IDV)             
Info (IDV)  +  Net haltreq => 0x0 { when:2980 }
Info (IDV)     +  Exception { pc:0x1c072(sub_3_6_0_t+4), mcause:0xb8000fff, mstatus:0x1880 }
Info (IDV)        +  Retire { pc:0x0(__jvt_base$+0) }
Info (IDV)           > dut.pc           - 0x1a110800 debug_rom+0
Info (IDV)             ref.pc           - 0x00000000 __jvt_base$+0
Info (IDV)           > dut.mcause       - 0xb8000401 Interrupt:1 minhv:0 mpp:3 mpie:1 mpil:0 Code:1025
Info (IDV)             ref.mcause       - 0xb8000fff Interrupt:1 minhv:0 mpp:3 mpie:1 mpil:0 Code:4095
Info (IDV)           > dut.dpc          - 0x00000000
[... 118 lines cut ...]
  mhpmcounterh11: 00000000   mhpmcounterh12: 00000000   mhpmcounterh13: 00000000   mhpmcounterh14: 00000000 
  mhpmcounterh15: 00000000   mhpmcounterh16: 00000000   mhpmcounterh17: 00000000   mhpmcounterh18: 00000000 
  mhpmcounterh19: 00000000   mhpmcounterh20: 00000000   mhpmcounterh21: 00000000   mhpmcounterh22: 00000000 
  mhpmcounterh23: 00000000   mhpmcounterh24: 00000000   mhpmcounterh25: 00000000   mhpmcounterh26: 00000000 
  mhpmcounterh27: 00000000   mhpmcounterh28: 00000000   mhpmcounterh29: 00000000   mhpmcounterh30: 00000000 
  mhpmcounterh31: 00000000          cpuctrl: 000f001f      secureseed0: 00000000      secureseed1: 00000000 
     secureseed2: 00000000      hpmcounter3: 00000000      hpmcounter4: 00000000      hpmcounter5: 00000000 
     hpmcounter6: 00000000      hpmcounter7: 00000000      hpmcounter8: 00000000      hpmcounter9: 00000000 
    hpmcounter10: 00000000     hpmcounter11: 00000000     hpmcounter12: 00000000     hpmcounter13: 00000000 
    hpmcounter14: 00000000     hpmcounter15: 00000000     hpmcounter16: 00000000     hpmcounter17: 00000000 
    hpmcounter18: 00000000     hpmcounter19: 00000000     hpmcounter20: 00000000     hpmcounter21: 00000000 
    hpmcounter22: 00000000     hpmcounter23: 00000000     hpmcounter24: 00000000     hpmcounter25: 00000000 
    hpmcounter26: 00000000     hpmcounter27: 00000000     hpmcounter28: 00000000     hpmcounter29: 00000000 
    hpmcounter30: 00000000     hpmcounter31: 00000000     hpmcounterh3: 00000000     hpmcounterh4: 00000000 
    hpmcounterh5: 00000000     hpmcounterh6: 00000000     hpmcounterh7: 00000000     hpmcounterh8: 00000000 
    hpmcounterh9: 00000000    hpmcounterh10: 00000000    hpmcounterh11: 00000000    hpmcounterh12: 00000000 
   hpmcounterh13: 00000000    hpmcounterh14: 00000000    hpmcounterh15: 00000000    hpmcounterh16: 00000000 
   hpmcounterh17: 00000000    hpmcounterh18: 00000000    hpmcounterh19: 00000000    hpmcounterh20: 00000000 
   hpmcounterh21: 00000000    hpmcounterh22: 00000000    hpmcounterh23: 00000000    hpmcounterh24: 00000000 
   hpmcounterh25: 00000000    hpmcounterh26: 00000000    hpmcounterh27: 00000000    hpmcounterh28: 00000000 
   hpmcounterh29: 00000000    hpmcounterh30: 00000000    hpmcounterh31: 00000000        mvendorid: 00000602 
         marchid: 00000015           mimpid: 00000000          mhartid: 00000000       mconfigptr: 00000000 
      mintstatus: 00000000 

```

MikeOpenHWGroup, 2023-07-28:
...love the IDV error reporting...

eroom1966, 2023-07-28:
OK, so I am confused a little by these retirements

```
RET,0,791,1a110c7c,"07cc0c13 addi    x24,x24,124    ",x24=00041904,,,,
RET,0,793,1a110800,"000c1663 bnez    x24,1a11080c   ",,,,CSR300(mstatus)=00001880 CSR341(mepc)=0001c072 CSR342(mcause)=b8000401 CSR7b0(dcsr)=400004d3 CSR7b1(dpc)=00000000,
```

why has the dpc transitioned to 0x00000000 ?

silabs-hfegran, 2023-07-31:
> OK, so I am confused a little by these retirements
> 
> ```
> RET,0,791,1a110c7c,"07cc0c13 addi    x24,x24,124    ",x24=00041904,,,,
> RET,0,792,1a110c80,"7b200073 dret                   ",,,,,
> RET,0,793,1a110800,"000c1663 bnez    x24,1a11080c   ",,,,CSR300(mstatus)=00001880 CSR341(mepc)=0001c072 CSR342(mcause)=b8000401 CSR7b0(dcsr)=400004d3 CSR7b1(dpc)=00000000,
> ```
> 
> why has the dpc transitioned to 0x00000000 ?

@Silabs-ArjanB, I need some clarification on this, is my interpretation below correct?
My impression is that dret had a pending nmi (dcsr.nmip), which implies that a return from debug mode we should jump to the nmi-handler (mtvec + 0x0, i.e. address 0x0), but this is interrupted by another debug request prior to retirement of the first instruction in the handler. Thus, dpc is set to the nmi handler address (0x0) as we took the nmi with the dret.

Silabs-ArjanB, 2023-07-31:
Hi @silabs-hfegran That RTL behavior would be consistent with the following part from https://docs.openhwgroup.org/projects/cv32e40x-user-manual/en/latest/debug.html

[image]

@silabs-oysteink Could you please confirm that this is what the RTL is actually doing?

eroom1966, 2023-07-31:
Hmmm, actually there is something else to consider
https://github.com/riscv/riscv-debug-spec/issues/796

This is not specifically referring to NMI, but should it be the same ?
ie, upon execution of xRET (presumably including DRET)
`must also be evaluated immediately following the execution of an xRET instruction or an explicit write to a CSR on which these interrupt trap conditions expressly depend (including mip, mie, mstatus, and mideleg).`

silabs-oysteink, 2023-07-31:
I haven't reproduced the issue myself, but from the information given in this issue it seems like the RTL takes the NMI immediately after the dret instruction, but before that first NMI handler instructions retires a new debug request is asserted. The RTL then re-enters debug mode with dpc pointing at the fetch address of the NMI handler.

Silabs-ArjanB, 2023-07-31:
Hi @eroom1966 The RISC-V debug issue you refer to has no associated debug spec update as far as I can see. Nonetheless, isn't the RTL doing exactly that (for a different reason though)? We take the NMI immediately following the execution of the DRET. However, the NMI later happens to get killed itself by another debug entry.

eroom1966, 2023-07-31:
@Silabs-ArjanB 
Yes I think this is what I was trying to say. Although there is no definitive description, it is almost implied that the pending nmi should be seen on the dret, so I think the RTL is behaving as expected.

Referring back to your earlier post regarding priority of NMI over haltreq. 
https://github.com/openhwgroup/core-v-verif/issues/2082#issuecomment-1657921198
What exactly does this mean ?
`CV32E40X prioritizes debug mode entry below NMIs, but above regular interrupts and synchronous exceptions.`
Does it mean that once the NMI has been raised, the haltreq should not be taken ?

Silabs-ArjanB, 2023-07-31:
Hi @eroom1966 

>> Does it mean that once the NMI has been raised, the haltreq should not be taken ?

If both an NMI and something leading to debug entry are pending then NMI entry is done first.

eroom1966, 2023-07-31:
just to be clear, the NMI entry is taken but does not execute, as it immediately switches into debug mode
This is what we are seeing in this particular case.

eroom1966, 2023-08-15:
@Silabs-ArjanB 
We are discussing this internally, and we are trying to interpret the rationale as this is not part of the specification
Can I ask
Why have you made NMI higher priority than debug events like haltreq when other interrupts are lower priority. 
Is there a good reason for this, or is it just what dropped out?

Silabs-ArjanB, 2023-08-15:
Hi @eroom1966 It is our interpretation of what 'non-maskable' is supposed to mean. It would be weird if there are higher prioritiy events that mask out so-called 'non-maskable' things.

eroom1966, 2023-08-16:
@Silabs-ArjanB 
I have a further question, please let me have your thoughts.

What would you expect to happen if we are running in single-step mode, ie, dcsr.step==1, but you have disabled interrupts by setting dcsr.stepie=0 (which also disables NMI)

What should happen next if there is an NMI pending, and you execute a DRET ?
You have said that NMI **cannot be masked**, but now you have a lower priority debug event (single-step) occurring, and the single step mode is configured to mask interrupts

In this case is the NMI taken or not ?
If it is not taken, this seems to contravene what you have said regarding masking out of 'non-maskable' things

Silabs-ArjanB, 2023-08-16:
Hi @eroom1966 Upon the DRET the NMI will not be taken as the Debug spec states that 'interrupts (including NMI) are disabled during single stepping with step set'.

When I said that NMIs should not be masked then I of course did not mean we are allowed to break existing RISC-V specs. That remark only applies to scenarios where the official RISC-V specs do not clearly mandate priorities (e.g. between external debug request and NMI).

eroom1966, 2023-08-16:
Fixed in ImperasDV hash : 20410de

closed 2023-08-23 by silabs-hfegran