openhwfoundation/core-v-verif #2088: ISS/TB: Properly enable mnxti/mscratchcsw/mscratcswl-checking with ISS

silabs-hfegran, 2023-07-27:
### Task Outcome
Currently `mnxti`, `mscratchcsw` and `mscratchcswl` are marked as volatile, which implies that their value will be copied from the values that the RTL reports. This needs a proper fix, so this task is here to keep this in mind.
Proper fix: The read value of these registers is only valid when there is a valid csr read that accesses these registers. At any other point in time, the value of these "registers" can not be compared. The ISS needs to take this into account, and when that is in place, the volatile-statement in the `uvmt_cv32e40<s/x>_imperas_dv_wrap` for these registers must be removed.

### Location Information
`core-v-verif/cv32e40s/tb/uvmt/uvmt_cv32e40s_imperas_dv_wrap.sv`
`core-v-verif/cv32e40s/tb/uvmt/uvmt_cv32e40x_imperas_dv_wrap.sv`

### Completion Criteria
Registers are no longer marked as volatile in the test bench, and there should be no related mismatches during test-runs that are caused by a failure to properly compare valid values.

silabs-robin, 2023-12-12:
I'm not sure if/how this relates to https://github.com/openhwgroup/core-v-verif/issues/2147

silabs-robin, 2023-12-12:
Seems this is not fixed yet:
[image]