openhwfoundation/core-v-verif #2107: ISS: WFE wakeup event skipped by ISS

silabs-hfegran, 2023-08-07:
## ISS misses wfe wakeup event
ISS misses wakeup event to instruction at address `0x4ee` while the RTL wakes up and executes next sleep instruction at `0x4f2`

### Type
* Functionally incorrect behavior

### Steps to Reproduce
Please provide:
1. cv32e40s/dev hash: 72e162a835ac2c8168c2282c16d947cfe42e43e8
2. `make test TEST=wfe_test USE_ISS=1 SEED=0` (presumably any seed)
3. ISS trace log leading up to the error:
```
Info 'refRoot/cpu', 0x000000000000044c(main+4c): 4562     c.lwsp  x10,24(x2)
Info 'refRoot/cpu', 0x000000000000044e(main+4e): 9782     c.jalr  x15
Info 'refRoot/cpu', 0x00000000000004c0(wfe_wakeup): b856     cm.push {x1,x8},-32
Info 'refRoot/cpu', 0x00000000000004c2(wfe_wakeup+2): 000127b7 lui     x15,0x12
Info 'refRoot/cpu', 0x00000000000004c6(wfe_wakeup+6): 000105a3 sb      x0,11(x2)
Info 'refRoot/cpu', 0x00000000000004ca(wfe_wakeup+a): 1c078793 addi    x15,x15,448
Info 'refRoot/cpu', 0x00000000000004ce(wfe_wakeup+e): c63e     c.swsp  x15,12(x2)
Info 'refRoot/cpu', 0x00000000000004d0(wfe_wakeup+10): cd89     c.beqz  x11,4ea
Info 'refRoot/cpu', 0x00000000000004ea(wfe_wakeup+2a): 8c000073 wfe
Error (IDV) The reference model is unable to find a legal set of actions to converge to the provided DUT state.
Info (IDV) Reference model mode (HartId: 0):
Info (IDV)   Machine - Machine mode
Info (IDV) Pending net changes:
Info (IDV)  - Group 0:
Info (IDV)  -           wu_wfe_i => 1,  age = 10 cycles, 1 retires
Info (IDV) DUT state change (hartId: 0):
Info (IDV)  - dut.pc               - 0x000004f2 wfe_wakeup+50
Info (IDV) Reference state change evaluation tree (HartId: 0):
Info (IDV)  +  Net wu_wfe_i => 0x1 { when:6931 }
Info (IDV)     +  Retire { pc:0x4ee(wfe_wakeup+46) }
Info (IDV)        > dut.pc           - 0x000004f2 wfe_wakeup+50
Info (IDV)          ref.pc           - 0x000004ee wfe_wakeup+46
Info (IDV)       
Info (IDV)  +  Retire { pc:0x4ee(wfe_wakeup+46) }
Info (IDV)     > dut.pc           - 0x000004f2 wfe_wakeup+50
Info (IDV)       ref.pc           - 0x000004ee wfe_wakeup+46
Info (IDV)    
Info (IDV) No state change found for reference to converge to DUT state.
Warning (IDV) The reference is being forced to converge to the DUT state, results that follow should be treated with caution.
UVM_ERROR @ 20841.300 ns : idvPkg.sv(92) reporter [] uvmt_cv32e40s_tb.imperas_dv.idv_trace2api.state_compare @ 20841.000 ns: MISMATCH

Dump Reference State GPR
 0: 00000000  1: 00000450  2: 003fffb0  3: 00013500 
 4: 00000000  5: 00000d58  6: 000018ec  7: 00000000 
 8: 00000000  9: 00000000 10: 00000000 11: 00000000 
12: 00000001 13: 0000000a 14: 003ffff0 15: 000121c0 
16: 80808080 17: fefefeff 18: 00000000 19: 00000000 
20: 00000000 21: 00000000 22: 00000000 23: 00000000 
24: 00000000 25: 00000000 26: 00000000 27: 00000000 
28: 00000000 29: 00000000 30: 00000000 31: 000121ac 

Dump Reference State CSR
             jvt: 00000000          mstatus: 00001800             misa: 40901104              mie: 00000000 
           mtvec: 00000001       mcounteren: 00000000          menvcfg: 00000000        mstateen0: 00000000 
       mstateen1: 00000000        mstateen2: 00000000        mstateen3: 00000000         mstatush: 00000000 
        menvcfgh: 00000000       mstateen0h: 00000000       mstateen1h: 00000000       mstateen2h: 00000000 
      mstateen3h: 00000000    mcountinhibit: 00000005       mhpmevent3: 00000000       mhpmevent4: 00000000 
      mhpmevent5: 00000000       mhpmevent6: 00000000       mhpmevent7: 00000000       mhpmevent8: 00000000 
      mhpmevent9: 00000000      mhpmevent10: 00000000      mhpmevent11: 00000000      mhpmevent12: 00000000 
     mhpmevent13: 00000000      mhpmevent14: 00000000      mhpmevent15: 00000000      mhpmevent16: 00000000 
     mhpmevent17: 00000000      mhpmevent18: 00000000      mhpmevent19: 00000000      mhpmevent20: 00000000 
     mhpmevent21: 00000000      mhpmevent22: 00000000      mhpmevent23: 00000000      mhpmevent24: 00000000 
     mhpmevent25: 00000000      mhpmevent26: 00000000      mhpmevent27: 00000000      mhpmevent28: 00000000 
     mhpmevent29: 00000000      mhpmevent30: 00000000      mhpmevent31: 00000000         mscratch: 00000000 
            mepc: 00000000           mcause: 00000000            mtval: 00000000              mip: 00000000 
         pmpcfg0: 00000000          pmpcfg1: 00000000          pmpcfg2: 00000000          pmpcfg3: 00000000 
         pmpcfg4: 00000000          pmpcfg5: 00000000          pmpcfg6: 00000000          pmpcfg7: 00000000 
         pmpcfg8: 00000000          pmpcfg9: 00000000         pmpcfg10: 00000000         pmpcfg11: 00000000 
        pmpcfg12: 00000000         pmpcfg13: 00000000         pmpcfg14: 00000000         pmpcfg15: 00000000 
        pmpaddr0: 00000000         pmpaddr1: 00000000         pmpaddr2: 00000000         pmpaddr3: 00000000 
        pmpaddr4: 00000000         pmpaddr5: 00000000         pmpaddr6: 00000000         pmpaddr7: 00000000 
        pmpaddr8: 00000000         pmpaddr9: 00000000        pmpaddr10: 00000000        pmpaddr11: 00000000 
       pmpaddr12: 00000000        pmpaddr13: 00000000        pmpaddr14: 00000000        pmpaddr15: 00000000 
       pmpaddr16: 00000000        pmpaddr17: 00000000        pmpaddr18: 00000000        pmpaddr19: 00000000 
       pmpaddr20: 00000000        pmpaddr21: 00000000        pmpaddr22: 00000000        pmpaddr23: 00000000 
       pmpaddr24: 00000000        pmpaddr25: 00000000        pmpaddr26: 00000000        pmpaddr27: 00000000 
       pmpaddr28: 00000000        pmpaddr29: 00000000        pmpaddr30: 00000000        pmpaddr31: 00000000 
       pmpaddr32: 00000000        pmpaddr33: 00000000        pmpaddr34: 00000000        pmpaddr35: 00000000 
       pmpaddr36: 00000000        pmpaddr37: 00000000        pmpaddr38: 00000000        pmpaddr39: 00000000 
       pmpaddr40: 00000000        pmpaddr41: 00000000        pmpaddr42: 00000000        pmpaddr43: 00000000 
       pmpaddr44: 00000000        pmpaddr45: 00000000        pmpaddr46: 00000000        pmpaddr47: 00000000 
       pmpaddr48: 00000000        pmpaddr49: 00000000        pmpaddr50: 00000000        pmpaddr51: 00000000 
       pmpaddr52: 00000000        pmpaddr53: 00000000        pmpaddr54: 00000000        pmpaddr55: 00000000 
       pmpaddr56: 00000000        pmpaddr57: 00000000        pmpaddr58: 00000000        pmpaddr59: 00000000 
       pmpaddr60: 00000000        pmpaddr61: 00000000        pmpaddr62: 00000000        pmpaddr63: 00000000 
         mseccfg: 00000000         mseccfgh: 00000000          tselect: 00000000           tdata1: 28001000 
          tdata2: 00000000            tinfo: 01008064             dcsr: 40000413              dpc: 00000000 
       dscratch0: 00000000        dscratch1: 00000000           mcycle: 00000000         minstret: 00000000 
    mhpmcounter3: 00000000     mhpmcounter4: 00000000     mhpmcounter5: 00000000     mhpmcounter6: 00000000 
    mhpmcounter7: 00000000     mhpmcounter8: 00000000     mhpmcounter9: 00000000    mhpmcounter10: 00000000 
   mhpmcounter11: 00000000    mhpmcounter12: 00000000    mhpmcounter13: 00000000    mhpmcounter14: 00000000 
   mhpmcounter15: 00000000    mhpmcounter16: 00000000    mhpmcounter17: 00000000    mhpmcounter18: 00000000 
   mhpmcounter19: 00000000    mhpmcounter20: 00000000    mhpmcounter21: 00000000    mhpmcounter22: 00000000 
   mhpmcounter23: 00000000    mhpmcounter24: 00000000    mhpmcounter25: 00000000    mhpmcounter26: 00000000 
   mhpmcounter27: 00000000    mhpmcounter28: 00000000    mhpmcounter29: 00000000    mhpmcounter30: 00000000 
   mhpmcounter31: 00000000          mcycleh: 00000000        minstreth: 00000000    mhpmcounterh3: 00000000 
   mhpmcounterh4: 00000000    mhpmcounterh5: 00000000    mhpmcounterh6: 00000000    mhpmcounterh7: 00000000 
   mhpmcounterh8: 00000000    mhpmcounterh9: 00000000   mhpmcounterh10: 00000000   mhpmcounterh11: 00000000 
  mhpmcounterh12: 00000000   mhpmcounterh13: 00000000   mhpmcounterh14: 00000000   mhpmcounterh15: 00000000 
  mhpmcounterh16: 00000000   mhpmcounterh17: 00000000   mhpmcounterh18: 00000000   mhpmcounterh19: 00000000 
  mhpmcounterh20: 00000000   mhpmcounterh21: 00000000   mhpmcounterh22: 00000000   mhpmcounterh23: 00000000 
  mhpmcounterh24: 00000000   mhpmcounterh25: 00000000   mhpmcounterh26: 00000000   mhpmcounterh27: 00000000 
  mhpmcounterh28: 00000000   mhpmcounterh29: 00000000   mhpmcounterh30: 00000000   mhpmcounterh31: 00000000 
         cpuctrl: 00000019      secureseed0: 00000000      secureseed1: 00000000      secureseed2: 00000000 
       mvendorid: 00000602          marchid: 00000015           mimpid: 00000000          mhartid: 00000000 
      mconfigptr: 00000000 

Info (IDV) hart0 PC has been resynchronized to 000004f6

```
4. Waveform:
[image]

### Additional context
WFE signalling was introduced by the wfe-agent to the ISS and dut in this commit.

eroom1966, 2023-08-15:
Hi Henrik

I see something slightly different.

I see a number of wfe instructions (.word 0x8c000073)
```
     4e6:       4501                    c.li    x10,0
     4e8:       a09d                    c.j     54e <wfe_wakeup+0x8e>
     4ea:       8c000073                .word   0x8c000073
     4ee:       8c000073                .word   0x8c000073
     4f2:       8c000073                .word   0x8c000073
     4f6:       8c000073                .word   0x8c000073
```
But I only see one event of a transition from 0->1 on the wakeup net, are you saying it is level sensitive, rather than edge sensitive ?

I have to go back and read the WFI (for comparison), but I thought this was edge based not level

eroom1966, 2023-08-15:
If I change the wu_wfe_i signal to be level sensitive (as appears to be your requirement from the RTL/testcase) then the test matches with the REF model
Its not clear from the spec that it is level sensitive - could the spec be updated to indicate this is the case ?

@Silabs-ArjanB could you please comment ?

eroom1966, 2023-08-15:
I should also add that in the ARM architecture the WFE is edge sensitive not level, so if the intent is to make this same feature available in RISCV, this is not equivalent

silabs-hfegran, 2023-08-16:
> If I change the wu_wfe_i signal to be level sensitive (as appears to be your requirement from the RTL/testcase) then the test matches with the REF model Its not clear from the spec that it is level [...]

Hi @eroom1966, the spec will be updated to clarify the behavior of this signal, and it is my understanding that we intentionally defined it as level sensitive.

silabs-hfegran, 2023-08-18:
Resolved, thanks for this fix @eroom1966

closed 2023-08-18 by silabs-hfegran