openhwfoundation/core-v-verif #2130: CV32E40S simulation

adrianjfallas, 2023-08-17:
Hi guys!

I'm having some trouble in compiling CV32E40S uvm simulation (starting from /sim/uvmt folder). I have not encountered this issue with CV32E40P or CV32E40X, both of those I have been able to run the hello-world test successfully.

I am using the cv32e40s/release branch. After setting the env variables, I run "make SIMULATOR=xrun hello-world USE_ISS=NO" and everything seems to be working fine and as expected until compiling the BSP.

```
mkdir -p $CORE_V_HOME/cv32e40s/sim/uvmt/xrun_results/default/hello-world/0/test_program
make bsp
make[2]: Entering directory '$CORE_V_HOME/cv32e40s/sim/uvmt'
$CORE_V_HOME/mk/Common.mk:300: CV_SW_CC not defined in either the shell environment, test.yaml or cfg.yaml
$CORE_V_HOME/mk/Common.mk:312: CV_SW_CFLAGS not defined in either the shell environment, test.yaml or cfg.yaml
$CORE_V_HOME/mk/Common.mk:327: RISCV set to /nis/asic/cr_dump2/fallasad/SeedSelector_Ubuntu/corev-openhw-gcc-ubuntu2004-20230623
$CORE_V_HOME/mk/Common.mk:328: RISCV_PREFIX set to riscv32-corev-elf-
$CORE_V_HOME/mk/Common.mk:329: RISCV_EXE_PREFIX set to /nis/asic/cr_dump2/fallasad/SeedSelector_Ubuntu/corev-openhw-gcc-ubuntu2004-20230623/bin/riscv32-corev-elf-
$CORE_V_HOME/mk/Common.mk:330: RISCV_MARCH set to rv32imc_zicsr
$CORE_V_HOME/mk/Common.mk:331: RISCV_CC set to gcc
$CORE_V_HOME/mk/Common.mk:332: RISCV_CFLAGS set to 
*******************************************************************************************
* Compiling the BSP
*******************************************************************************************
mkdir -p $CORE_V_HOME/cv32e40s/sim/uvmt/xrun_results/default/hello-world/0/test_program/bsp
cp $CORE_V_HOME/cv32e40s/bsp/Makefile $CORE_V_HOME/cv32e40s/sim/uvmt/xrun_results/default/hello-world/0/test_program/bsp
make -C $CORE_V_HOME/cv32e40s/sim/uvmt/xrun_results/default/hello-world/0/test_program/bsp \
	VPATH=$CORE_V_HOME/cv32e40s/bsp \
	RISCV=$TOOLCHAIN/corev-openhw-gcc-ubuntu2004-20230623 \
	RISCV_PREFIX=riscv32-corev-elf- \
	RISCV_EXE_PREFIX=$TOOLCHAIN/corev-openhw-gcc-ubuntu2004-20230623/bin/riscv32-corev-elf- \
	RISCV_MARCH=rv32imc_zicsr \
	RISCV_CC=gcc \
	RISCV_CFLAGS="" \
	all
make[3]: Entering directory '$CORE_V_HOME/cv32e40s/sim/uvmt/xrun_results/default/hello-world/0/test_program/bsp'
$TOOLCHAIN/corev-openhw-gcc-ubuntu2004-20230623/bin/riscv32-corev-elf-gcc -Os -g -static -mabi=ilp32 -march=rv32imc_zicsr -Wall -pedantic -mno-relax  -c $CORE_V_HOME/cv32e40s/bsp/crt0.S -o crt0.o
$CORE_V_HOME/cv32e40s/bsp/crt0.S: Assembler messages:
$CORE_V_HOME/cv32e40s/bsp/crt0.S:33: **Error: unknown CSR `jvt'**
make[3]: *** [Makefile:22: crt0.o] Error 1
make[3]: Leaving directory '$CORE_V_HOME/cv32e40s/sim/uvmt/xrun_results/default/hello-world/0/test_program/bsp'
make[2]: *** [$CORE_V_HOME/mk/Common.mk:524: bsp] Error 2
make[2]: Leaving directory '$CORE_V_HOME/cv32e40s/sim/uvmt'
make[1]: *** [$CORE_V_HOME/mk/Common.mk:495: $CORE_V_HOME/cv32e40s/sim/uvmt/xrun_results/default/hello-world/0/test_program/hello-world.elf] Error 2
make[1]: Leaving directory '$CORE_V_HOME/cv32e40s/sim/uvmt'
make: [$CORE_V_HOME/mk/uvmt/xrun.mk:306: hello-world] Error 2 (ignored)

```

I am getting this **Error: unknown CSR `jvt'** , have you guys seen this before?

I am using an embescom toolchain, and as said before, I have not encountered this issue or any other on the CV32E40P or CV32E40X cores, only on CV32E40S.

The version of xrun that I am using is 20.09.011. 

Thanks in advance!

silabs-hfegran, 2023-08-18:
Hi @adrianjfallas, 
Firstly, as @MikeOpenHWGroup stated, I would recommend you use the `cv32e40s/dev` branch, as the release branch is not technically for releases but a staging branch that is used for merges between the two closely related cores, cv32e40x and cv32e40s. Apologies for the unfortunate naming. The dev branch is the one that is most up-to-date with latest developments and the only one actively maintained. 

Secondly, the compiler flags that you are using are outdated (This is fixed on the dev branch), the c-extension has been split up into `zca` (old c ext), `zcb`, `zcmp` and `zcmt`. I did not think that there would be any zcmt-related instructions in the bsp, are you by any chance using a release of the embecosm toolchain with zcmt disabled? if so, that can explain why the jvt register is unknown, as that is a part of that extension. There are a whole lot of issues with compiler-support for zcmt currently, but as it has been necessary for our verification efforts, we have tried our best to work around the various issues that have occurred, so most of the directed tests should work fine provided that the correct flags are used.

adrianjfallas, 2023-08-18:
Thank you both for the quick response.

Understood, I am using the `cv32e40s/dev` branch now. 

And you are right @silabs-hfegran, I was using a version of the toolchain with the zcmt disabled, they actually placed this warning about that: 
**Warning: This build does not include Zcmt table generation functionality. If this is needed please use the build dated 22 Jun 2023.**
When using that other toolchain version and setting the `CV_SW_MARCH` to what @MikeOpenHWGroup suggested, the problem was solved.

Again thanks a lot.

closed 2023-08-18 by adrianjfallas