openhwfoundation/core-v-verif #2134: Merge Script Improvements

silabs-robin, 2023-08-18:
### Task Outcome
Merge script gets additional features and tweaks (as per list below).

### Background information
We have started using the new script for doing S->X merges, and ideas for improvements are appearing.

### Location Information
https://github.com/openhwgroup/core-v-verif/blob/cv32e40s/dev/bin/merge.sh

### Completion Criteria
All feature suggestions are either implemented or rejected.

### Additional context
* Option to generate "rejection diff".
    * Rationale: x-dv PR reviews are much easier to do with a rejection diff.
* `git branch --unset-upstream` after cloning/branching operations.
    * Rationale: The default tracking is the upstream repo, and it is slightly impractical.
* Generated branch names should be PR-title'able.
    * Rationale: Github defaults the PR title to the merge commit description, no need to manually come up with a title.
* Echo what command is ran ("set -x").
    * Rationale: It is not easy to understand what the script does when it runs.
* "Press any key to continue" before major operations.
    * Rationale: Sometimes one would like more control and monitoring of what the script is about to do.
    * Comment: Could be an option to enable it (default is off).

kristinedosvik, 2023-08-18:
Agree with all the suggustions!

silabs-krdosvik, 2023-08-31:
Also agreeing with another suggestion from @silabs-hfegran: The script executes instructions that require no prior setup. If there is already a setup, the instructions that are executed may give warnings or error messages. It can be confusing and should be avoided.

silabs-hfegran, 2023-08-31:
Another suggestion: perhaps checkout the cv32e40x-dv repo into a folder not called cv32e40x, this is to preserve any possible work-in-progress files or repo changes that may be present in the cv32e40x folder.

silabs-robin, 2023-08-31:
> The script executes instructions that require no prior setup. If there is already a setup, the instructions that are executed may give warnings or error messages. It can be confusing and should be avoided.

I struggle to understand.

1. Does the script today give warnings, and the warnings are confusing?
2. Or is it confusing that the script today does not give warnings, and we want to add warnings?

Further:
* Should the script stop when it encounters reason to give warnings?
* What kind of warnings are we talking about?

silabs-robin, 2023-08-31:
> Another suggestion: perhaps checkout the cv32e40x-dv repo into a folder not called cv32e40x, this is to preserve any possible work-in-progress files or repo changes that may be present in the cv32e40x folder.

I added this to the list.

silabs-hfegran, 2024-01-11:
Are all these suggestions implemented and can this issue be closed? @silabs-robin @silabs-krdosvik

silabs-krdosvik, 2024-01-12:
The "new suggestions" are not implemented yet. I can look into it now.

silabs-robin, 2024-01-15:
My immediate unfiltered thoughts:

* Leave it up, because it is genuinely a good list of improvement-ideas.
    * Ask @MikeOpenHWGroup if he sees value for OpenHW group in leaving the issue open until it is implemented. If not, then close the issue.
* Don't implement anything before it is necessary (or an occasion arises where it seems useful again).
* Mark it with some kind of "waived" label, with regards to TRL 5 sign-off.

MikeOpenHWGroup, 2024-01-15:
From #2349 we, the following merge script improvements have been implemented:
- Add check if merge is up to date
- Better branch names,
- Add checks befor/after operations.
- Remove unused option

I lost the thread to see which other items have not yet been implemented.

closed 2024-01-16 by silabs-robin