openhwfoundation/core-v-verif #2137: ISACOV : Ignore WILDCARD Bins

AyoubJalali, 2023-08-18:
Hello @silabs-robin @MikeOpenHWGroup ,
In the ISACOV agent there's some macros that define wildcard bins for toggling GPR (rx), the macros content 3 arguments : 
- name
- field
- iff_exp
as I understand the **_iff_exp_** argument is the one that decide if we create the Coverpoint or not, in the example the c.mv instruction does not have rs1, so it make sense to disable rs1_value coverpoint &  toggling rs1 ->   **`ISACOV_CP_BITWISE(cp_rs1_toggle, instr.rs1_value, has_rs1)** (has_rs1 = 0 for c.mv & has_rs1 = 1 for c.add)
what i got in the coverage report is : 
[image]

as you can see the rs1_value is successfully ignore, but rs1_toggle NO !! and it count in the coverage.

So I wanna know what you got int this covergoupe in the report, is that a tool behavior for **synopsys** or a BUG in these macros.

silabs-robin, 2023-08-18:
From what I can tell, the `iff` only sets a qualifying condition to decide when the coverpoint can be sampled.
Similar to how `iff` works for bins.
So the coverpoint would still exist, AFAICT.
(I only read a bit of coverage code within core-v-verif and didn't confer the LRM now.)

Here is what it looks like for us:
[image]

silabs-robin, 2023-08-18:
The CP in question: https://github.com/openhwgroup/core-v-verif/blob/master/lib/uvm_agents/uvma_isacov/cov/uvma_isacov_cov_model.sv#L977C4-L977C21

The macro in question: https://github.com/openhwgroup/core-v-verif/blob/master/lib/uvm_agents/uvma_isacov/uvma_isacov_macros.sv

I think the only way to remove the coverpoint might be to use the `with` syntax that was discussed previously. (At least I can't think of anything else at the moment.)
And I don't remember how well it went when you used `iff` instead.

AyoubJalali, 2023-08-18:
So it count also in coverage.
I don't think using **iff** with WILDCARD bins is the same as using **iff** with bins.
can't we use option.weight = iff_exp in the macros, that way we're sure that does not count

AyoubJalali, 2023-08-18:
> I think the only way to remove the coverpoint might be to use the `with` syntax that was discussed previously. (At least I can't think of anything else at the moment.)

could you test it to confirm that **with** works in that case

silabs-robin, 2023-08-21:
> could you test it to confirm that with works in that case

I couldn't get it to work.
(Not sure why, because the LRM grammar seems to say it is okay, but Xcelium thought otherwise.)

@MikeOpenHWGroup do you have any ideas?

MikeOpenHWGroup, 2023-08-21:
> I couldn't get it to work.
(Not sure why, because the LRM grammar seems to say it is okay, but Xcelium thought otherwise.)

Can you expand on this @silabs-robin?  Does it compile and run under Xcelium?  If so, what is the resulting coverage look like?

silabs-robin, 2023-08-21:
I didn't manage to add `with` expressions to the wildcard bins, because Xcelium got an error saying it was not ok to do so.

So then I can't really think of any ideas atm for how to prevent such coverpoints from showing in coverage results.

silabs-robin, 2023-08-22:
> can't we use option.weight = iff_exp in the macros, that way we're sure that does not count

@AyoubJalali Yeah, maybe this. I have not tried it.

AyoubJalali, 2023-08-24:
> I couldn't get it to work.

try it on WILDCARD BINS not on the coverpoint

silabs-robin, 2023-08-28:
> > I couldn't get it to work.
> 
> try it on WILDCARD BINS not on the coverpoint

That is what I did.