openhwfoundation/core-v-verif #2148: ISS: debug_test_trigger tdata2 mismatch with trigger type 5, bits 4 and 6

silabs-hfegran, 2023-08-23:
## tdata2 mismatch with trigger type 5, bits 4 and 6

### Type
* Functionally incorrect behavior

### Steps to Reproduce
Please provide:
1. [cv32e40x/dev 81604f4faecbeb863fdcf7](https://github.com/openhwgroup/core-v-verif/commit/81604f4faecbeb863fdcf7b8bc6138720409e367)
2. ` make test TEST=debug_test_trigger CV_CORE=cv32e40x CFG=debug_trigger_cfg1 USE_ISS=YES COV=YES RNDSEED=0 `
3.
```
Info (IDV) Instruction executed prior to mismatch '0x1a1109ea(_debug_mode_register_test+356): 7a249073 csrw    tdata2,x9'
Error (IDV) CSR register value mismatch (HartId:0, PC:0x1a1109ea _debug_mode_register_test+356):
Info (IDV)  0> CSR 7a2 (tdata2)
Info (IDV)   . dut:0x010008fe
Info (IDV)   . ref:0x010008ae
UVM_ERROR @ 22287.300 ns : idvPkg.sv(92) reporter [] uvmt_cv32e40x_tb.imperas_dv.idv_trace2api.state_compare @ 22287.000 ns: MISMATCH

Dump Reference State GPR
 0: 00000000  1: 1a110896  2: 003fff30  3: 00014328 
 4: 00000000  5: 00000000  6: 00000003  7: 00000000 
 8: 58000001  9: ffffffff 10: 0000000a 11: 000145a9 
12: 00000021 13: 0000000a 14: 00800000 15: 00000000 
16: 00000000 17: 00000020 18: 00000000 19: 00000000 
20: 00013000 21: 00000000 22: 00000000 23: 00000000 
24: 00000000 25: 00000000 26: 00000000 27: 00000000 
28: 00000003 29: 0001459d 30: 00000000 31: 00012de4 

Dump Reference State CSR
             jvt: 00000000          mstatus: 00001880             misa: 40801104              mie: 00000000 
           mtvec: 00000001         mstatush: 00000000    mcountinhibit: 0000000d       mhpmevent3: 00000000 
      mhpmevent4: 00000000       mhpmevent5: 00000000       mhpmevent6: 00000000       mhpmevent7: 00000000 
      mhpmevent8: 00000000       mhpmevent9: 00000000      mhpmevent10: 00000000      mhpmevent11: 00000000 
     mhpmevent12: 00000000      mhpmevent13: 00000000      mhpmevent14: 00000000      mhpmevent15: 00000000 
     mhpmevent16: 00000000      mhpmevent17: 00000000      mhpmevent18: 00000000      mhpmevent19: 00000000 
     mhpmevent20: 00000000      mhpmevent21: 00000000      mhpmevent22: 00000000      mhpmevent23: 00000000 
     mhpmevent24: 00000000      mhpmevent25: 00000000      mhpmevent26: 00000000      mhpmevent27: 00000000 
     mhpmevent28: 00000000      mhpmevent29: 00000000      mhpmevent30: 00000000      mhpmevent31: 00000000 
        mscratch: 00000000             mepc: 00001880           mcause: 00000002            mtval: 00000000 
             mip: 00000000          tselect: 00000000           tdata1: 58000001           tdata2: 010008ae 
           tinfo: 01008064             dcsr: 400004d3              dpc: 00001632        dscratch0: 00000000 
       dscratch1: 00000000           mcycle: 00000000         minstret: 00000000     mhpmcounter3: 00000000 
    mhpmcounter4: 00000000     mhpmcounter5: 00000000     mhpmcounter6: 00000000     mhpmcounter7: 00000000 
    mhpmcounter8: 00000000     mhpmcounter9: 00000000    mhpmcounter10: 00000000    mhpmcounter11: 00000000 
   mhpmcounter12: 00000000    mhpmcounter13: 00000000    mhpmcounter14: 00000000    mhpmcounter15: 00000000 
   mhpmcounter16: 00000000    mhpmcounter17: 00000000    mhpmcounter18: 00000000    mhpmcounter19: 00000000 
   mhpmcounter20: 00000000    mhpmcounter21: 00000000    mhpmcounter22: 00000000    mhpmcounter23: 00000000 
   mhpmcounter24: 00000000    mhpmcounter25: 00000000    mhpmcounter26: 00000000    mhpmcounter27: 00000000 
   mhpmcounter28: 00000000    mhpmcounter29: 00000000    mhpmcounter30: 00000000    mhpmcounter31: 00000000 
         mcycleh: 00000000        minstreth: 00000000    mhpmcounterh3: 00000000    mhpmcounterh4: 00000000 
   mhpmcounterh5: 00000000    mhpmcounterh6: 00000000    mhpmcounterh7: 00000000    mhpmcounterh8: 00000000 
   mhpmcounterh9: 00000000   mhpmcounterh10: 00000000   mhpmcounterh11: 00000000   mhpmcounterh12: 00000000 
  mhpmcounterh13: 00000000   mhpmcounterh14: 00000000   mhpmcounterh15: 00000000   mhpmcounterh16: 00000000 
  mhpmcounterh17: 00000000   mhpmcounterh18: 00000000   mhpmcounterh19: 00000000   mhpmcounterh20: 00000000 
  mhpmcounterh21: 00000000   mhpmcounterh22: 00000000   mhpmcounterh23: 00000000   mhpmcounterh24: 00000000 
  mhpmcounterh25: 00000000   mhpmcounterh26: 00000000   mhpmcounterh27: 00000000   mhpmcounterh28: 00000000 
  mhpmcounterh29: 00000000   mhpmcounterh30: 00000000   mhpmcounterh31: 00000000            cycle: 00000000 
         instret: 00000000      hpmcounter3: 00000000      hpmcounter4: 00000000      hpmcounter5: 00000000 
     hpmcounter6: 00000000      hpmcounter7: 00000000      hpmcounter8: 00000000      hpmcounter9: 00000000 
    hpmcounter10: 00000000     hpmcounter11: 00000000     hpmcounter12: 00000000     hpmcounter13: 00000000 
    hpmcounter14: 00000000     hpmcounter15: 00000000     hpmcounter16: 00000000     hpmcounter17: 00000000 
    hpmcounter18: 00000000     hpmcounter19: 00000000     hpmcounter20: 00000000     hpmcounter21: 00000000 
    hpmcounter22: 00000000     hpmcounter23: 00000000     hpmcounter24: 00000000     hpmcounter25: 00000000 
    hpmcounter26: 00000000     hpmcounter27: 00000000     hpmcounter28: 00000000     hpmcounter29: 00000000 
    hpmcounter30: 00000000     hpmcounter31: 00000000           cycleh: 00000000         instreth: 00000000 
    hpmcounterh3: 00000000     hpmcounterh4: 00000000     hpmcounterh5: 00000000     hpmcounterh6: 00000000 
    hpmcounterh7: 00000000     hpmcounterh8: 00000000     hpmcounterh9: 00000000    hpmcounterh10: 00000000 
   hpmcounterh11: 00000000    hpmcounterh12: 00000000    hpmcounterh13: 00000000    hpmcounterh14: 00000000 
   hpmcounterh15: 00000000    hpmcounterh16: 00000000    hpmcounterh17: 00000000    hpmcounterh18: 00000000 
   hpmcounterh19: 00000000    hpmcounterh20: 00000000    hpmcounterh21: 00000000    hpmcounterh22: 00000000 
   hpmcounterh23: 00000000    hpmcounterh24: 00000000    hpmcounterh25: 00000000    hpmcounterh26: 00000000 
   hpmcounterh27: 00000000    hpmcounterh28: 00000000    hpmcounterh29: 00000000    hpmcounterh30: 00000000 
   hpmcounterh31: 00000000        mvendorid: 00000602          marchid: 00000014           mimpid: 00000000 
         mhartid: 00000000       mconfigptr: 00000000 

```

### Additional context
The same test with the following configurations exhibit the same issue (longer runtime than the above):
` make test TEST=debug_test_trigger CV_CORE=cv32e40x CFG=debug_trigger_cfg2 USE_ISS=YES COV=YES RNDSEED=0 `
` make test TEST=debug_test_trigger CV_CORE=cv32e40x CFG=debug_trigger_cfg3 USE_ISS=YES COV=YES RNDSEED=0 `
` make test TEST=debug_test_trigger CV_CORE=cv32e40x CFG=debug_trigger_cfg4 USE_ISS=YES COV=YES RNDSEED=0 `

[image]

These two bits are stated to be WARL with no value restriction, ISS appears to hardwire these to zero when A-extension is not enabled.

eroom1966, 2023-08-29:
OK, I think I see what is happening here,
The E40X and E40S support misaligned accesses in hardware, therefore these exceptions cannot be raised.
Is my understanding correct ?

This would explain why 4 and 6 cannot be set, as they cannot be raised

eroom1966, 2023-08-29:
From the Priv spec

5.6.15 Exception Trigger (etrigger, at 0x7a1)
...
Hardware may support only a subset of exceptions. A debugger must read back tdata2 after
writing it to confirm the requested functionality is actually supported.

silabs-hfegran, 2023-08-30:
Hi Lee, 

The privilege spec `3.6.3.3 Alignment` comments the following 
[image]

[image]

The X (but not S) supports raising these exceptions as the A-extension is supported. @Silabs-ArjanB, are there any differences to the WARL legal values for the tdata2 register if the A_EXT parameter is turned off?

Silabs-ArjanB, 2023-08-30:
Yes, as implied by the text in yellow these two bits are WARL (0x0) if A_EXT == A_NONE:

[image]

We could add a more specific note about that below that table if you want.

silabs-hfegran, 2023-08-30:
Ok so this appears to be an RTL issue, not an ISS issue. Closing

closed 2023-08-30 by silabs-hfegran