openhwfoundation/core-v-verif #2185: Questions Running Coremark Test

nerdylye, 2023-09-06:
Hi, I am new to this and followed the tutorial steps to run coremark test on cv32w40p core. The simulation can be run but the output is a little weird because the iterations/sec is 0 and the coremark score does not show. May I know if it is supposed to be like this? 

The command I used to run is `make veri-test TEST=coremark` with Verilator version 5.012

The toolchain I used is from [RISC-V Compiler Toolchain - lowRISC provides a pre-built GCC based toolchain](https://github.com/lowRISC/lowrisc-toolchains/releases)

[image]

pascalgouedo, 2023-09-06:
Hi @nerdylye 
I think this is linked to your toolchain and if it is supporting to print Floating-Point or not.
For instance pre-built RISC-V toolchain used for CV32E40Pv1 were not containing anything to allow any printf of Floating-Point numbers (mainly because it implies much larger newlib library).

pascalgouedo, 2023-09-06:
But you can still compute your Coremark/MHz score using the reported number of cycles (Total ticks).

nerdylye, 2023-09-07:
Hi @MikeOpenHWGroup, thank you for your reply. I think I'm using CV32E40Pv1 because I do not see this branch `cv32e40p/dev` in my directory.

Hi @pascalgouedo, thank you for your reply. I see, so the simulation is running fine am I correct?

I also tried doing the same thing on CV32E40S and CV32E40X cores but both are not working. Is it because Verilator is not supported for these cores?

pascalgouedo, 2023-09-07:
Hi @nerdylye  

Yes simulation is OK as confirmed by "Correct operation validated" sentence.

For core version you surely use cv32e40v2 RTL as I guess you are using master branch.

For the E40X and E40S I don't know.

nerdylye, 2023-09-07:
I'm getting error like this for CV32E40S and CV32E40X cores, even when running hello-world test. 

[image]

Who can I ask for assistance? Thank you

MikeOpenHWGroup, 2023-09-07:
> I'm getting error like this for CV32E40S and CV32E40X cores, even when running hello-world test.

Hi @nerdylye, we like to keep GitHub Issues focused on a single topic.  Please open a second issue for this one.

nerdylye, 2023-09-08:
Ok got it, thank you!

closed 2023-09-08 by nerdylye