openhwfoundation/core-v-verif #2230: MIssing test.yaml in building CV32E40P

baichen318, 2023-10-07:
Hi all,

I am new to `core-v-verif`, and I follow the guidelines documented in [Quick Start](https://docs.openhwgroup.org/projects/core-v-verif/en/latest/quick_start.html#core-v-verif-quick-start-guide).

However, I encountered an error suggesting missing `test.yaml` for `hello-world` in the repository by executing the following instructions.

```bash
$ cd $CORE_V_VERIF/cv32e40p/sim/core
$ make
```

Enclosed, please find the attached screenshot highlighting the error.
[image]

I was wondering how I could solve this problem.

Thank you all in advance.

Best regards,
Chen

MikeOpenHWGroup, 2023-10-10:
Hi @baichen318, thanks for your interest in CORE-V-VERIF.  It look like [yaml2make](https://github.com/openhwgroup/core-v-verif/blob/580ea6f94f36cf6c25faf6d4b4b0f30d3d97cef5/bin/yaml2make#L1) did not run as expected.  If you are interested in the details, check out [Test-program Definitions](https://github.com/openhwgroup/core-v-verif/blob/master/mk/TOOLCHAIN.md#test-program-definitions) section of `mk/TOOLCHAIN.md`.

Based on your description I can think of two things to try:
1. Your shell environment variables.  See [mk/README.md](https://github.com/openhwgroup/core-v-verif/blob/master/mk/README.md#required-corev-environment-variables).
2. You may not have all the python plugins.  Try this:
```
$ cd $CORE_V_VERIF/bin
$ pip3 install -r requirements.txt
```

MikeOpenHWGroup, 2023-11-15:
Hi @baichen318, is your problem fixed?  If so, please close this Issue.  If we have not heard from you in one week, we will close this Issue.

closed 2023-12-21 by MikeOpenHWGroup