openhwfoundation/core-v-verif #2286: Clarify new illegal handler code for HWLOOPs

MikeOpenHWGroup, 2023-11-08:
In pull-request #2280, the **corev-dv** function `cv32e40p_asm_program_gen::gen_illegal_instr_handler(int hart)` was substantially updated to add illegal handler code for HWLOOPs:
- If the illegal exception happened within a hwloop.
- If yes , check if the illegal was the last instruction of an hwloop with at least 1 more iteration left.
- If yes, then set MEPC to first instruction of hwloop instead of just incrementing by 4.
  Else MEPC is just incremented by 4.

The code is fine, but difficult for a human to follow (because it is SystemVerilog code that emits RISC-V assembler) so it would be good to add a comment header to the function to include the above information.

MikeOpenHWGroup, 2023-12-21:
Hi @dd-vaibhavjain, is there a chance you can get to this Issue before the end of this calendar year?

dd-vaibhavjain, 2023-12-29:
The relavant comments for hwloop handling were added for all the handlers in corev_dv for cv32e40p. 
This will be part of the next PR.

MikeOpenHWGroup, 2024-01-26:
Hi @dd-vaibhavjain, has the "next PR" been submitted?

XavierAubert, 2024-02-01:
Hi Mike ! 
PR #2357 has just been submitted, and it contains the commit 44647b9a6cbcb23519c534ce6177b5c22879123a to address this issue

MikeOpenHWGroup, 2024-02-02:
Thanks @XavierAubert.  PR #2357 has been merged, so I will close this Issue.

closed 2024-02-02 by MikeOpenHWGroup