openhwfoundation/core-v-verif #2299: VPTOOL does not load features

yesilzeytin, 2023-11-22:
## VPTOOL does not load features

### Type
* Functionally incorrect behavior

### Steps to Reproduce
1. URL to branch that exhibits the issue: **https://github.com/openhwgroup/core-v-verif**
2. Command line: **sh vptool-example/runme.sh**
3. Logfile and/or wave-dump info:
[image]
[image]

### Additional context
When tried to follow the VPTOOL example as a starting point, the tool opens but does not load any features as mentioned in the README. When they are tried to be added manually through .pck files, they do not load anyway.

An interesting observation that I had is when we follow all the steps in the README (**https://github.com/openhwgroup/core-v-verif/blob/67791932df6128671344015584109d7a177d9a9a/tools/vptool/README.md**), I observe the warning **"Locked Ip file has not been loaded"**, and see one **locked_ip.pick** being loaded anyway. This warning is not observed when the **locked_ip.pick** in **vptool/vptool-example/example-database/vptool** is copied to the **vptool/vptool-example/example-database/** in folder hierarchy and two locked IP files are being loaded. This may or may not be related to the issue, but wanted to share screenshots regarding both just in case.

MikeOpenHWGroup, 2023-11-22:
Hi @zchamski, I am not sure if you are the right person to assign this to.  Please feel free to re-assign if not.

zchamski, 2023-11-22:
Hi @yesilzeytin, can you please share the commit info of the VPTOOL you are using?

Also, is your DVplan still in the Pickle format, or was it already migrated to Yaml?  The Pickle format for storing features was phased out quite some time ago (Feb 2023), replaced by Yaml, and _by default_ VPTOOL will only read DVplans stored in Yaml files.

yesilzeytin, 2023-11-23:
Hi @zchamski, I'm not sure what exactly you meant with commit info, but the latest commits into my branch are in the link below since I tried to run VPTOOL with a newly cloned repository just yesterday to make sure that everything is up-to-date: https://github.com/openhwgroup/core-v-verif/commits/master

Actually, my DVplan is in Pickle format. That may be the issue, then. However, I couldn't see any example YAML files in the repository, nor details on how a YAML format should be in terms of README and so on. Is the current README for VPTOOL outdated, then? If so, are there any resources that I can use to examine the YAML format, such as a guideline or an example YAML that you know of and can provide?

zchamski, 2023-11-23:
Hi @yesilzeytin, the recommended approach in your case is to migrate the DVplan from Pickle to Yaml using VPTOOL itself.   There's a "magic" VPTOOL option `-p` which tells VPTOOL to read in Pickle files instead of the Yaml ones.  When you subsequently save the files, they will be saved in Yaml format.

This is intentionally a one-way migration path towards Yaml. There is no formalized specification of the Yaml file structure (yet).  However, care was taken to preserve the Yaml file structure in the load/save cycle so that any differences between file versions are solely linked to new/changed content.

yesilzeytin, 2023-11-24:
Hi @zchamski, Thanks a lot for the fast responses. I spent some time examining the environment as much as I could to make sure that I wasn't missing something simple, but unfortunately, I still couldn't fix the issue. In order to clearly identify the problem (on my part or the VPTOOL part), I wanted to provide the process I used step-by-step to provide replicability.

- Cloned the latest commit (9th of November) of the master branch.
- Installed all the prerequisites on README (https://github.com/openhwgroup/core-v-verif/blob/67791932df6128671344015584109d7a177d9a9a/tools/vptool/README.md).
- Run command:
`export PROJECT_IDENT=test`
- Update the runme.sh script to modify the line
`python3 $ROOTDIR/../vptool/vp.py -t winxpblue`
to
`python3 $ROOTDIR/../vptool/vp.py -t winxpblue -p` and `python3 $ROOTDIR/../vptool/vp.py -p -t winxpblue` in two different tests
- Run command:
`sh vptool-example/runme.sh`

Nothing loads. Also, when we tried to load it manually through the VPTOOL interface and manually select Pickle files, they still did not load.

In the interface, we can only add Features, and we cannot also modify the content on the right-hand side or add any Subfeatures. When we add Features that are totally empty and save them as YAML files, naturally, they only include Features, and therefore we still cannot see, add, or modify Subfeatures/Verification Items/Item Descriptions and see their YAML formats.

We tried the same steps on two different Ubuntu machines, and they both acted the same for reference.

abdullahyildiz, 2024-01-11:
Hi all,

I also have the same issue and cannot successfully load the example database.

zchamski, 2024-01-11:
Hi @abdullahyildiz and @yesilzeytin, I will try to get this fixed tomorrow.  If unsuccessful, I will come back to you Monday 22 January (I will be offline during the coming week.)

yesilzeytin, 2024-02-13:
Hi @zchamski, sorry to bother you but did you have any opportunity to examine the issue?

zchamski, 2024-02-14:
Hi @yesilzeytin, I've been squeezed time-wise, I can get back to working on VPTOOL starting next week (Mon 19 Feb) at the earliest.

yesilzeytin, 2024-02-15:
Hi @zchamski, thanks for the heads up. Good luck to you in your work/tasks.

zchamski, 2025-01-08:
Hi @yesilzeytin, AFAI remember we cleaned up this issue as being related to the Pickle-to-Yaml migration in VPTOOL DV plan files.    Can you please confirm that you can use VPTOOL in Yaml mode - or that you do not use it at all (and so #2299 is a non-issue)?

yesilzeytin, 2025-01-09:
Hi @zchamski. That is what I remember as well. Back then, we had a meeting where you showed us the fix on your local build, which worked smoothly for us.

However, I just cloned a fresh repository to determine whether this issue is "non-issue" anymore or "fixed". I didn't comprehensively test the tool, but I can say that as it currently stands, the issue seems to be fixed, and the README is consistent with the tool as well. Therefore, we can mark this issue as not "non-issue" but "fixed."

closed 2025-01-09 by yesilzeytin