openhwfoundation/core-v-verif #2324: Typo in constraint distribution ranges in cv32e40p_pulp_hwloop_instr_lib.sv

MikeOpenHWGroup, 2023-12-12:
In a discussion of pull-request #2305 there was mention of a set of typos in [cv32e40p_pulp_hwloop_instr_lib.sv](https://github.com/openhwgroup/core-v-verif/blob/c98ff2a4d747eec08514f1dec8664923694d92db/cv32e40p/env/corev-dv/instr_lib/cv32e40p_pulp_hwloop_instr_lib.sv#L1102).   For example:
```
foreach(hwloop_count[i])
          hwloop_count[i] dist {[0:400] := 10, [401:1023] := 300, [1024:2047] := 150,
                                [2048:4094] := 50, 4095 := 300};
```
A comment from @dd-vaibhavjain indicated that he noticed a typo just now for `hwloop_count` range.  As the typo was very minor, I approved and merged the change, so this typo is now at the head of the cv32e40p/dev branch.   The purpose of this Issue is to ensure we review this and either fix or accept the range.

dd-vaibhavjain, 2023-12-12:
Hi @MikeOpenHWGroup , this was already fixed in that PR.

MikeOpenHWGroup, 2023-12-12:
Really? Starting at [cv32e40p_pulp_hwloop_instr_lib.sv line #1101](https://github.com/openhwgroup/core-v-verif/blob/c98ff2a4d747eec08514f1dec8664923694d92db/cv32e40p/env/corev-dv/instr_lib/cv32e40p_pulp_hwloop_instr_lib.sv#L1101) we have:
```
          hwloop_counti[1] dist {[0:400] := 10, [401:1023] := 300, [1024:2047] := 150,
                                 [2048:4094] := 50, 4095 := 300};

          hwloop_count[1] dist {[0:400] := 10, [401:1023] := 300, [1024:2047] := 150,
                                [2048:4094] := 50, 4095 := 300};
```
Is that what you want?

MikeOpenHWGroup, 2023-12-12:
> if you are asking about the distribution itself, do you see any issue that we need to address?

Earlier I was concerned because it seemed that hits of `hwloop_count[1] == 4095` would be hard to achieve.  That seems to have been addressed.   The probability of `hwloop_count[1] == 0` seems low.

Anyway, you are happy with the distribution so I will close this issue.  If functional coverage has holes we may need to tune the distribution.

closed 2023-12-12 by MikeOpenHWGroup