openhwfoundation/core-v-verif #2333: ISS MISMATCH (minhv=1) mepc misalignment

silabs-krdosvik, 2023-12-14:
## ISS MISMATCH (minhv=1) mepc misalignment
Running modified clic test “mret_with_minhv” (with induced error) and get this mismatch:
[image]

The dut traps due to mepc being misaligned and therefore fetch a random word (that is an illegal instruction).
We think the dut traps correctly due to this part of the spec:
([https://github.com/riscv/riscv-fast-interrupt/blob/master/clic.adoc)](https://github.com/riscv/riscv-fast-interrupt/blob/master/clic.adoc)#) 
[image]

### Type
* Functionally incorrect behavior

### Steps to Reproduce
1)	git clone [git@github.com:openhwgroup/core-v-verif.git](mailto:git@github.com:openhwgroup/core-v-verif.git)
2)	cd core-v-verif/
3)	git checkout cv32e40x/dev
4)	rm -rf cv32e40x
5)	git clone git@github.com:silabs-krdosvik/cv32e40x-dv.git cv32e40x
6)	cd cv32e40x
7)	git checkout clic-iss-problem
8)	cd sim/uvmt/
9)	make test TEST=clic CFG=clic_default

eroom1966, 2023-12-18:
Hi Kristine,

Can I request that you run the simulation above, capturing the rvvi database.
In order to do this, you should only need to do the two following things

1. set environment variable in your shell
export IMPERAS_RVVI_DEBUG=1

2. add the following +define to your simulation run
+define+IDV_INCLUDE_TRACE2BIN

After running your simulation, you should have a directory called rvvi.db
Please provide this to me for analysis

silabs-krdosvik, 2023-12-19:
Of course. 
I will send you an email with the rvvi.db folder.

eroom1966, 2023-12-20:
@silabs-krdosvik 
Apologies in advance, the database produced is a little short - can I ask you to increase the number of MISMATCHES, so that the simulation continues for a few more cycles.
In my execution it exits exactly on the instruction here 

```
Info   mstatus 00001800 -> 00001880 [SD:0 XS:0(Off) FS:0(Off) MPP:3 VS:0(Off) MPIE:0->1 MIE:0]
Info   mcause 70390002 -> 78390002 [Interrupt:0 minhv:1 mpp:3 mpie:0->1 mpil:57 Code:2(Virtual supervisor software interrupt)]
Info (IDV) rvviDecode.rvvi_trace2log.process_event @ 1439630: RET,0,29212,00006132,"30200073 mret                   ",,,,CSR300(mstatus)=0000000000001880 CSR342(mcause)=0000000078390002,
Info (IDV) rvviDecode.rvvi_trace2log.process_event @ 1439650: EXC,0,29213,62460f0e,"0000     c.illegal              ",,,,CSR300(mstatus)=0000000000001800 CSR341(mepc)=0000000062460f0e CSR342(mcause)=0000000030390002,
rvviRefShutdown
```

Please increase the MISMATCHES to 100 and re-run
sorry for the inconvenience

thx
Lee

silabs-krdosvik, 2023-12-27:
I see, of course, no problem. 

I will send you an email with the new version of the rvvi.db folder and also the xrun-clic.log file, which is what I believe is the full log (correct me if I am mistaken).

> Please increase the MISMATCHES to 100 

I have set the IDV_MAX_ERRORS parameter to 100 and now the ImperasDV VERIFICATION REPORT reports 33 mismatches instead of only 6. I tell you this in case I somehow have misunderstood and there was another parameter you wanted me to change.

(The new configuration additions are now merged to the branch "clic-iss-problem" on my cv32e40x-dv fork.)

eroom1966, 2023-12-27:
Yes this sounds correct regarding the max errors
I am out of office till Jan/3, so will pick this up then

Thx
Lee

====================================================================
The information contained in this electronic mail message and any
attachments hereto
is privileged and confidential information intended only for the use of the
individual or
entity named above or their designee.  If the reader of this message is not
the intended
recipient, you are hereby notified that any dissemination, distribution or
copying of this
communication is strictly prohibited.  If you have received this
communication in error
please immediately notify us by return  message or by telephone and delete
the
original message from your mail system.  Thank you.
====================================================================

silabs-krdosvik, 2024-01-08:
The MISMATCH disappears when updated our ISS version.

closed 2024-01-08 by silabs-krdosvik

eroom1966, 2024-01-09:
@silabs-krdosvik can we close this issue ?

silabs-krdosvik, 2024-01-10:
Yes. the issue is closed.