openhwfoundation/core-v-verif #2381: Covergroups not in the DV plan

MikeOpenHWGroup, 2024-02-28:
In pull-request #2380 there was a [question](https://github.com/openhwgroup/core-v-verif/pull/2380/files/50bd147b911fd59f25478573cc33715f39a88837#r1504662897) regarding some new cross-coverage bins.  Although these bins look good, I am worried about two things:
1. They are not part of any DV plan.
2. There are a lot of cross-coverage bins which will be very difficult to hit.

My concern is that we now have a situation where the functional coverage score will be decreased and much effort will be needed to fill the new holes.  Before we do that, let's:
- review the coverage to ensure we need it.
- update the DV plan accordingly.

XavierAubert, 2024-06-13:
Covergroups links have been added in #2417, and after the last round of updates and reviews in #2455, I think you could have a look and decide if this issue can be closed.

MikeOpenHWGroup, 2024-06-14:
Thanks @XavierAubert and @dd-BeeNee, this looks good.  Closing this issue.

closed 2024-06-14 by MikeOpenHWGroup