openhwfoundation/core-v-verif #2386: uvmt_cv32e40p_tb_ifs.sv needs to be refactorted

MikeOpenHWGroup, 2024-03-14:
As mentioned in pull-request #2385, `cv32e40p/tb/uvmt/uvmt_cv32e40p_tb_ifs.sv` (on the **cv32e40p/dev** branch) has become very large, containing eight SV-interfaces and several of these interfaces support complex logic.  I think it is time to consider refactoring this file into two or more files.  Perhaps the interfaces used for coverage could be moved into a separate file.

XavierAubert, 2024-06-13:
Sorry, I totally forgot this about this issue... 
Is it too late considering how close project closure is, or should I take 10-20min to make the update you requested Mike? 
Since it's only moving interface definitions in another files, in my opinion, there is a low risk of breaking anything and/or changing any coverage metric...

MikeOpenHWGroup, 2024-06-13:
Hi @XavierAubert, given that we are very close to generating a release of CV32E40P, I recommend that we do not do this until after the release.  We can waive this issue for the release.