openhwfoundation/core-v-verif #2388: Discussion regarding DV plan annotation: Links to Coverage

MikeOpenHWGroup, 2024-03-19:
In #2385, @XavierAubert asked:

> Regarding DV plan annotation, the first batch targets plans of XPULP (non-hwloop) instructions. As you participated in the first round of reviews, @MikeOpenHWGroup and @MarioOpenHWGroup can you have a look and tell me if the way I did the annotation seems fine to you ? Please note that the file mentionned in the DV Plan that describes UVM TB plusargs is still a WIP, I will push it as soon as possible.

The purpose of this Issue is to discuss how to properly capture a "link to coverage" from the DV plan (typically in a spreadsheet) and the verification environment.  Note that I have labelled this as a `cv32e40p` issue only because the above PR is from a Contributor working on that Project.  In fact, this question affects all CORE-V cores.

MikeOpenHWGroup, 2024-03-19:
In @XavierAubert's update to the [CV32E40P Debug DV Plan](https://github.com/XavierAubert/core-v-verif/blob/cv32e40p/dev_dd_w11a/cv32e40p/docs/VerifPlans/Simulation/debug-trace/CV32E40P_debug.xlsx), you can see the following in row 3 under the `Link to Coverage` column:
```
CG: uvm_pkg.uvm_test_top.env.cov_model.debug_covg.cg_ebreak_execute_with_ebreakm
CG: uvm_pkg.uvm_test_top.env.cov_model.debug_covg.cg_cebreak_execute_with_ebreakm
A : uvmt_cv32_tb.u_debug_assert.a_cebreak_debug_mode
```

I believe this is a satisfactory way to capture the link from a specific entry in a DV plan to covergroup and cover-properties.

MikeOpenHWGroup, 2024-03-19:
> Please note that the file mentionned in the DV Plan that describes UVM TB plusargs is still a WIP, I will push it as soon as possible.

Do you mean `CV32E40Pv2_test_list.xlsx`?  It would be very useful to see an early draft of this file.

MikeOpenHWGroup, 2024-03-19:
>  I need to delete reference to v1 links in order to reflect v2 paths and I wanted more insight on how I should proceed.
> I also agree on the format to report the links

OK, given that we agree on the format, let me suggest a way to (perhaps) make entering all these paths to covergroups less painful by replacing `uvm_pkg.uvm_test_top.env.cov_model` with an alias such as `$CM` to yield:
```
CG: $CM.debug_covg.cg_ebreak_execute_with_ebreakm
CG: $CM.debug_covg.cg_cebreak_execute_with_ebreakm
```
If you follow this strategy, it would be good to add a new worksheet (tab) to the spreadsheet that defines what these aliases mean and how they should be interpreted.  For example, you could have something like this in a tab called "README":

| Alias        | Expands To           | Comment  |
| ------------- |-------------| -----|
| $CM      | uvm_pkg.uvm_test_top.env.cov_model | Functional Coverage Model Root |
| $DBGCM      | uvm_pkg.uvm_test_top.env.cov_model.debug_covg  | Root of Functional Coverage model for Debug  |

XavierAubert, 2024-05-17:
I pushed the DV plans without the aliases you advised in PR #2417 (commit 95ea5cf42cd21c4711a3317ab919c9dbc4c1728c), but we proceeded to separate v1 and v2 DV Plans to avoid confusion on paths to coverage PR #2425 (commit 538deaef5f5c97816b6a5b9091afb108406cb2d9).

Can we consider this is enough to close the issue @MikeOpenHWGroup ? Or do you want to keep it open and remove cv32e40p label as it may concern other cores ?

MikeOpenHWGroup, 2024-05-17:
Thanks @XavierAubert!   I'd like to keep this Issue open until our next meeting (2024-05-22) to give us a chance to discuss it "in person".