openhwfoundation/core-v-verif #2399: [TASK] How to config multi-core verification based on core-v-verif Project or CVA6 Project ?

charles7098, 2024-03-28:
### Is there an existing core-v-verif task for this?

- [X] I have searched the existing task issues

### Task Description

Hello, Developers. I'm currently using the CVA6 Project Verificatoin Testbench. I want to used this TestBench to verif multi-core verification. Does the Testbench itself support multi-core verification?

### Description of Done

None

MikeOpenHWGroup, 2024-03-28:
Hi @charles7098, thanks for your interest in CORE-V-VERIF.  This verification environment was specifically developed for a single-core, single-thread Design-Under-Test (DUT).  I would not use CORE-V-VERIF for a multi-core (or even a multi-threaded) DUT.

You could consider looking at the [OpenPiton](https://github.com/PrincetonUniversity/openpiton) project.   The goals of OpenPiton are not the same as CORE-V-VERIF, but it could be the basis of a multi-core testbench.

closed 2024-03-28 by MikeOpenHWGroup

charles7098, 2024-03-29:
OK，thanks for your answer!