openhwfoundation/core-v-verif #2403: [cv_dv_utils AXI2MEM] Ready signals are stuck at 1.

khandelwaltanuj, 2024-04-05:
Hello,

While working on the verification of HPDcache, I observed that ready signals of AXI2MEM are internally driven to 1. The readme of AXI2MEM says that user should drive these signals. 

Thanks and Regards
Tanuj Khandelwal

MikeOpenHWGroup, 2024-04-05:
Thanks for the issue. Can you point to the specific file and line number?

khandelwaltanuj, 2024-04-08:
Hi @MikeOpenHWGroup 

The file is: 
https://github.com/openhwgroup/core-v-verif/blob/master/lib/cv_dv_utils/uvm/memory_rsp_model/axi2mem/axi_intf.sv

line 82 to 86; I need to comment out these lines. 

Regards
Tanuj

MikeOpenHWGroup, 2024-04-08:
Hi @ludovicpion, I think @khandelwaltanuj is correct.  The file and line numbers he references hardcode `aw_ready`, `ar_ready` and `assign w_ready` to 1'b1.  Can you submit a pull-request to remove these assignments and allow the user to control them via a sequence?