openhwfoundation/core-v-verif #2407: [cv_dv_utils pulse gen] To add pulse generator(new uvm driver)  in cv_dv_utils

khandelwaltanuj, 2024-04-11:
Hello,

A new uvm driver pulse generator is added in the cv_dv_utils. 

Here is a little detail. 
The Pulse Gererator is a SystemVerilog UVM module which is used to configure and generate pulses. Pulse Generator can be configured to generate multiple pulses. A pulse can be a sysncronous pulse with respect to a clock or an asynchronous pulse. Timeunit of pico second is used to genarate an asynchronous pulse.

The new driver has been tested in the TB of HPDCache, where it is used to generator the signal flush. And we would like to commit it to the open source. 

Thanks and regards
Tanuj Khandelwal

khandelwaltanuj, 2024-04-11:
Hi @MikeOpenHWGroup 

It is a drived class from uvm_driver. 
It contains following components 

pulse_gen_driver extend uvm_driver (dummy, dummy)
--> pulse gen configuration class 
--> pulse gen interface 

I hope you have now a clear picture of it. 

Thanks and regards
Tanuj Khandelwal

MikeOpenHWGroup, 2024-04-11:
Thanks @khandelwaltanuj.  As mentioned in #2408, lets move this discussion the [TWG:Verification channel on MatterMost](https://mattermost.openhwgroup.org/all-users/channels/twg--verification) and we can discuss it there.
 
When you are ready to make your contribution, please create a pull-request and be sure to follow the contribution guidelines above.

closed 2024-04-11 by MikeOpenHWGroup