openhwfoundation/core-v-verif #2429: Immediate assertion without an else clause

MikeOpenHWGroup, 2024-05-13:
Pull-request #2428 introduced an immediate assertion without an else clause.  In CORE-V-VERIF immediate assertions require an `else` clause that invokes `uvm_error` to ensure an unambiguous failure in simulation should the assertion fire.  This can easily be fixed as below: 
```
@@ -607,7 +607,8 @@ interface uvmt_cv32e40p_cov_if
         detect_apu_rvalid   = 1;
       end
       else begin
-          assert (clk_cycle_window <= MAX_FP_XACT_CYCLE); 
+          assert (clk_cycle_window <= MAX_FP_XACT_CYCLE)
+           else `uvm_error("uvmt_cv32e40p_cov_if", "clk_cycle_window > MAX_FP_XACT_CYCLE"); 
           if (apu_req && apu_gnt && apu_rvalid_i && detect_apu_rvalid) begin : IS_0_CYC_FPU
             clk_cycle_window  = 0;
             detect_apu_rvalid = 0;
```

dd-baoshan, 2024-05-14:
Thanks for the feedback. I overlooked the guideline when applied the assertion. I have PR the fix on this.
#2431

MikeOpenHWGroup, 2024-05-14:
Fixed in #2431.

closed 2024-05-14 by MikeOpenHWGroup