openhwfoundation/core-v-verif #2435: docs: Need to update respective core user manual links in intro.rst

subbu009, 2024-05-15:
## Task Title
Need to update respective core user manual links in core-v-verif/docs/VerifStrat/source/intro.rst

### Task Outcome
The existing links for user manuals of
cv32e40x https://cv32e40x.readthedocs.io/en/latest/
cv32e40s https://cv32e40s.readthedocs.io/en/latest/
cva6 https://cva6.readthedocs.io/en/latest/intro.html
are not found.
I found modified links point to the latest docs from the respective core repos.
cv32e40x https://docs.openhwgroup.org/projects/cv32e40x-user-manual/en/latest/
cv32e40s https://docs.openhwgroup.org/projects/cv32e40s-user-manual/en/latest/
cva6 https://docs.openhwgroup.org/projects/cva6-user-manual/

MikeOpenHWGroup, 2024-05-15:
Hi @subbu009, thanks for reporting this issue.  I have created pull-request #2436 to resolve.  Please have a look.

MikeOpenHWGroup, 2024-05-15:
Hi @subbu009, #2436 has been merged. Please close this issue if you are satisfied.

closed 2024-05-16 by subbu009