openhwfoundation/core-v-verif #2444: [Spike tandem] Runtime exceptions in CVA6 Spike + VCS-UVM tandem simulations

zchamski, 2024-05-21:
Issue observed and reproducible on recent versions of `core-v-verif` code including the latest master (commit 4e6e8604f): when attemptring a CVA6 Spike tandem run with VCS-UVM (envariable `DV_SIMULATORS` contains `vcs-uvm` and `SPIKE_TANDEM=1`), VCS fails with an uncaught DPI runtime exception:

```
Error-[DPI-UED] C++ Exception detected                                                                                                                                        
  Import DPI routine invoked at file
  '...../cva6/verif/sim//../core-v-verif/lib/uvm_components/uvmc_rvfi_reference_model/rvfi_spike.sv'(line
  76) has C++ exceptions not caught. C++ exceptions shall not propagate out of
  any imported subroutine.
  Fix it in DPI-C code before running simulation.
```
### Type

* Functionally incorrect behavior

### Steps to Reproduce

1. URL to branch that exhibits the issue: `master` branch at https://github.com/openhwgroup/core-v-verif (the result above was obtained with https://github.com/openhwgroup/core-v-verif/commit/4e6e8604fa295e9974924abcf3e447ab60805f88).
2. Command line: at cva6 toplevel `export DV_SIMULATORS=vcs-uvm SPIKE_TANDEM=1 ; bash verif/regress/smoke-tests.sh`

zchamski, 2024-05-21:
Further investigation gives more insigth into the nature of the failure.  Here's a relevant `.log.iss` logfile fragment with extra messages added:
```
2368 [SPIKE] Starting 'spike_create'...
2369 [SPIKE] htif_args = {
2370   /........../cva6/verif/sim/out_2024-05-21/directed_asm_tests/add.o,
2371 }
2372 [SPIKE] Proc 0 | ISA: RV64IMAFDC_zba_zbb_zbc_zbs_zcb_zicsr_zicntr PRIV: MSU
2373 <stdin>:39.20-43.7: Warning (simple_bus_reg): /soc/clint@33554432: simple-bus unit address format error, expected "2000000"
2374 [SPIKE] PASS: Got back default /top/dram_base value = 0x80000000!
2375 [SPIKE] Simulation::Simulation after parse_params: /top/dram_base value = 0x69a098211923f469!
2376 [SPIKE] Simulation::Simulation: Before make_mems
2377 [SPIKE] Initializing memories...
2378 [SPIKE] ROM content head(8) = 13 04 10 00 13 14 f4 01
2379 [SPIKE] BootROM initialization done.
2380 [SPIKE] Setting up DRAM... Got size == 6542914149569485696 (0x5acd19c0d0f75b80)
2381 terminate called after throwing an instance of 'std::runtime_error'
2382   what():  memory size must be a positive multiple of 4 KiB
2383 
2384 Error-[DPI-UED] C++ Exception detected
2385   Import DPI routine invoked at file
2386   '/........../cva6/verif/sim//../core-v-verif/lib/uvm_components/uvmc_rvfi_reference_model/rvfi_spike.sv'(line
2387   76) has C++ exceptions not caught. C++ exceptions shall not propagate out of
2388   any imported subroutine.
2389   Fix it in DPI-C code before running simulation.
```

Clearly, DRAM base address and DRAM size values are not valid, pointing to Spike parameter management.

zchamski, 2024-05-21:
This issue blocks openhwgroup/cva6/issues/1898.

AyoubJalali, 2024-05-27:
> The fix in PR [openhwgroup/cva6#2153](https://github.com/openhwgroup/cva6/pull/2153) solves the problems of random (sic!) runtime exceptions in Spike/VCS-UVM tandem code caused by invalid DRAM parameters. @AyoubJalali , can you please rebase your IRQ code (or cherry-pick the commit from [openhwgroup/cva6#2153](https://github.com/openhwgroup/cva6/pull/2153)) and check if the DPI interface works OK now?

on it !!
Thanks

zchamski, 2024-06-06:
Hi @AyoubJalali, any updates on the status?

closed 2024-06-14 by AyoubJalali