openhwfoundation/core-v-verif #2450: QUESTASIM_HOME Neither Documented nor Checked

silabs-robin, 2024-05-29:
### Type
* Compile error (hopefully nobody has committed anything that doesn't compile!)
* Confusing or extraneous status or error messages
* Other.

### Steps to Reproduce
Try to build `spike_lib`.
E.g. approximately `cd cv32e40s/sim/uvmt && make spike_lib`.

### Additional context
Here is the error message you get:
`../fesvr/fesvr_dpi.cc:22:10: fatal error: svdpi.h: No such file or directory`
There is no clue that QUESTASIM_HOME is missing.
`grep`'ing from top-level reveals no documentation.
What should happen is that whenever you run a target that requires this env var to be set, then it should automatically be checked and then printed an error if it is not set ([example](https://github.com/openhwgroup/core-v-verif/blob/cv32e40s/dev/bin/clonetb#L51), in shell but you can do the same in make or any other language).

MarioOpenHWGroup, 2024-05-29:
Hi Robin, makes sense I will open a PR with the check :+1:

MarioOpenHWGroup, 2024-05-29:
Probably the best solution is to not let the user compile spike_lib if the SIMULATOR var is not set as it requires any of the three big ones

silabs-robin, 2024-05-29:
I had SIMULATOR set

MarioOpenHWGroup, 2024-05-29:
Okeey I will submit a fix :+1:

MikeOpenHWGroup, 2024-05-29:
A little bit of digging produced this:
```
ifeq ($(ENVXXX),)
   $(error ENVXXX is undefined)
endif
```
Adding this to the top of `vsim.mk` will ensure that the check runs for every invocation of make, regardless of the target. **_Important_**: the `ifeq` and `endif` must start in column 0 and the indentation of `$error` must be with spaces, not tabs.

closed 2024-05-29 by MikeOpenHWGroup