openhwfoundation/core-v-verif #2467: [BUG] : mstatus.TW is set to 1

AyoubJalali, 2024-06-10:
Hello, the [cv32a65x spec](https://docs.openhwgroup.org/projects/cva6-user-manual/06_cv32a65x_riscv/index.html#virt-control) the TW filed in MSTATUS is read-only zero, but Spike allow the read 1 in TW filed after a write into it in both modes TANDEM & SOLO

MarioOpenHWGroup, 2024-06-14:
Can you share more info about the specific test and specific platform ?

AyoubJalali, 2024-06-14:
> Can you share more info about the specific test and specific platform where you can see this issue ?

This issue based on the new [CV32A65X Spec](https://docs.openhwgroup.org/projects/cva6-user-manual/04_cv32a65x/riscv/priv.html#_machine_status_mstatus_and_mstatush_registers), and the issue appears using the csr_test, or just write a 0xffff_ffff and read it.

zchamski, 2024-06-14:
Just checked Spike source (both "our" vendorized code and upstream master), and `mstatus.TW` is writable irrespective of implemented extensions `U`/`S`, whereas the `status.MPRV` bit is duly guarded by the presence of the `U` extension.

zchamski, 2024-06-14:
@MarioOpenHWGroup: AFAI can tell, here too the features introduced in cvv #2445 should solve the issue (with proper mask, ideally derived from riscv-config info). Do you confirm?

MarioOpenHWGroup, 2024-06-14:
The #2465 should give the mechanism to solve this yes :) But it should be configured in the cva6utils sv code

closed 2024-07-08 by AyoubJalali