openhwfoundation/core-v-verif #2468: Bug : MTVAL is not read-only zero

AyoubJalali, 2024-06-11:
Hello, in the CV32A65X configuration the mtval is read-only zero (doesn't not raise an exception if we write into it), is means we only have the zero value in it, but Spike isn't aligned and store the instruction causes the exception into it, the bug appears in both solo and tandem mode.

TANDEM :
[image]

SOLO : 
[image]

MikeOpenHWGroup, 2024-06-11:
The [CVA3265X User Manual](https://docs.openhwgroup.org/projects/cva6-user-manual/06_cv32a65x_riscv/index.html#_machine_trap_value_mtval_register) states that the mtval register is an MXLEN-bit read-only 0 register, so I think the RTL is doing the right thing.

This is yet-another Spike configuration issue.

zchamski, 2024-06-14:
Just checked Spike source (both "our" vendorized code and upstream master), and `mstatus.TW` is writable irrespective of implemented extensions `U`/`S`, whereas the `status.MPRV` bit is duly guarded by the presence of the `U` extension.

MarioOpenHWGroup, 2024-06-14:
It should be configurable using the new PR on cvv #2465

closed 2024-07-08 by AyoubJalali