openhwfoundation/core-v-verif #2484: [BUG] UVMA core cntrl agent is not aware of H (hypervisor) extension

zchamski, 2024-06-27:
**Functionally incorrect behavior:**  the core cntrl cfg agent does not mark CSRs `mtinst` and `mtval2` as unsupported in the absence of the `H` (hypervisor) extension.  As a result, Spike tandem simulations of CVA6 consider these registers as implemented in RTL (**incorrect** behavior) while Spike reports them as not found (correct behavior).

The effect can be seen in tandem scoreboard checks.

### Steps to Reproduce
Please provide:
1. URL to the branch that exhibits the issue: [cva6 master@33ab2efa83454e5b4f90c2156863b47adb0c56ba](https://gitlab.thales-invia.fr/riscv-ci/cva6/-/commit/33ab2efa83454e5b4f90c2156863b47adb0c56ba)
2. Command line: `export DV_SIMULATORS=vcs-testharness SPIKE_TANDEM=1; bash verif/regress/smoke-tests.sh`, target `cv32a65x`.
3. Logfile excerpt  from `verif/sim/out_2024-06-27/vcs-testharness_sim/rv32i-I-ADD-01.cv32a65x.log.iss`, CSRs 34a (`mtinst`) and 34b (`mtval2`):
```
UVM_INFO /gitlab-runner/runner_riscv-public/builds/yD5zmwgi3/1/riscv-ci/cva6/verif/core-v-verif/lib/uvm_components/uvmc_rvfi_scoreboard/uvmc_rvfi_scoreboard_utils.sv(198) @ 12210: reporter [spike_tandem] 
CSR 341  Mismatch   [REF]: 0xffffffff8000031a [CORE]: 0x8000031a        
CSR 34a  not found  [REF]: 0x0                [CORE]: 0x0               
CSR 34b  not found  [REF]: 0x0                [CORE]: 0x0               
UVM_INFO /gitlab-runner/runner_riscv-public/builds/yD5zmwgi3/1/riscv-ci/cva6/verif/core-v-verif/lib/uvm_components/uvmc_rvfi_scoreboard/uvmc_rvfi_scoreboard_utils.sv(199) @ 12210: reporter [spike_tandem]                12210 | RVFI |        3 |      0 | 80000004 | 34202f73 | M | x0  | 00000000 | x2  | 00000000 | x30 | 000000000000000b | csrrs   t5, mcause, zero
UVM_ERROR /gitlab-runner/runner_riscv-public/builds/yD5zmwgi3/1/riscv-ci/cva6/verif/core-v-verif/lib/uvm_components/uvmc_rvfi_scoreboard/uvmc_rvfi_scoreboard_utils.sv(200) @ 12210: reporter [spike_tandem]                12210 | RVFI |        0 |      0 | 80000004 | 34202f73 | M | x0  | 00000000 | x2  | 00000342 | x30 | 000000000000000b | csrrs   t5, mcause, zero <- CORE
```

### Additional context

- The root cause of the incorrect behavior is the initialization of the `unsupported_csr_ mask` in file https://github.com/openhwgroup/core-v-verif/blob/master/lib/uvm_agents/uvma_core_cntrl/uvma_core_cntrl_cfg.sv which does NOT mark CSrs  `mtinst` and `mtval2` as unsupported if `H` extension is not present.
- The discrepancy on the value of CSR 341 (`mepc`) is a false negative and is being addressed.

AyoubJalali, 2024-07-03:
Hello @zchamski , can you confirm that the bug is fixed

MarioOpenHWGroup, 2024-08-23:
Solved, H bit included on the `core_cntrl_cfg`

closed 2024-08-23 by MarioOpenHWGroup