openhwfoundation/core-v-verif #2491: How can I use Spike ISS?

coolderin, 2024-07-03:
Hi y'all,

I'm trying to verify CV32E40P core. 
I can't use/purchase Imperas solutions due to economic issues. So, I have to use freeware options. Is there any configuration/settings/files about to use Spike ISS (or any other free ISS) as golden model?

MarioOpenHWGroup, 2024-07-03:
Hi @coolderin,
We have an implementation that allows us to use Spike ISS with some of our cores but it is not yet implemented for CV32E40P. I think can be easily implemented looking to the other implementations but is not yet done.

pascalgouedo, 2024-07-03:
Hi @coolderin 
We are in the phase of pushing all the final reports for the full verification of v2 step of CV32E40P.
What do you intend to verify on that core?
Additional custom instructions not part of CV32E40Pv2?

As mentioned @MarioOpenHWGroup it could be possible to use Spike ISS for CV32E40P but it will require some efforts to implement that in makefiles without breaking Imperas model usage.

coolderin, 2024-07-03:
We're trying to run torture test using riscv-dv. We've xcelium simulator only.

pascalgouedo, 2024-07-03:
It was already quite hard to synchronize RTL and Imperas Model on asynchronous events while Imperas model is way smarter than Spike.
So good luck.

coolderin, 2024-07-11:
Oh Alright, does this IP create any assembly files temporarily and where does it keep?

If it does I'll run these assembly files and testbenches by manual in spike and maybe in verilator.

MikeOpenHWGroup, 2024-07-11:
> does this IP create any assembly files temporarily and where does it keep?

As explained in the README, the testbench for this core is in the [CORE-V-VERIF](https://github.com/openhwgroup/core-v-verif/tree/cv32e40p/dev) repo (cv32e40p/dev branch).  There is a small set of human-generated C and RISC-V assembly files there that are used as test-programs for verification.  However, most of the test-programs used to verify this core have been generated with a CV32E40P-specific set of extensions to riscv-dv.   We do not keep these machine generated files in the repo.

> If it does I'll run these assembly files and testbenches by manual in spike and maybe in verilator.

This is not a good use of your time, and will not allow you to fully verify the core.  The v1.0.0 release of CV32E40P is considered "fully verified", so unless you make modifications to the RTL, there is little need to verify it.  If you want to see it running, disable the Imperas ISS (see CORE-V-VERIF documentation for more information about this).

If you want floating point and/or pulp instructions, the v1.8.3 is considered "almost fully verified".

If you are serious about a verification effort for your own variant of CV32E40P, please reach out to me at mike@openhwgroup.org and we can discuss getting you set up.

closed 2024-07-11 by MikeOpenHWGroup