openhwfoundation/core-v-verif #2500: Bug : Spike doesn't raise exception

AyoubJalali, 2024-07-08:
This is a bug on tandem and solo mode in Spike compare to the [CV32A65X spec](https://docs.openhwgroup.org/projects/cva6-user-manual/04_cv32a65x/riscv/priv.html) : 
Spike doesn't raise exception on some no implement CSRs in CV32A65X configuration : 

1. menvcfg -> U-mode not supported in CV32A65X
2. menvcfgh -> U-mode not supported in CV32A65X
3. mcounteren -> U-mode not supported in CV32A65X
4. mcountinhibit 
5. tinfo (0x7a4)
6. mscontext (0x7aa)

AyoubJalali, 2024-08-23:
the issue fixed for those CSRs : 
menvcfg
menvcfgh
mcounteren

MikeOpenHWGroup, 2024-08-24:
Hi @AyoubJalali, thanks for re-starting the discussion on his issue.  Are you saying the the Spike Tandem model for CV32A65X now works for the `menvcfg`, `menvcfgh` and `mcounteren` CSRs, but still does not work for `mcountinhibit`, `tinfo` and `mscontext`?

In which PR(s) where the  `menvcfg`, `menvcfgh` and `mcounteren` CSRs fixed?

AyoubJalali, 2024-08-26:
> Hi @AyoubJalali, thanks for re-starting the discussion on his issue. Are you saying the the Spike Tandem model for CV32A65X now works for the `menvcfg`, `menvcfgh` and `mcounteren` CSRs, but still [...]

No it's the opposite, the don't really have the PR number

AyoubJalali, 2024-08-26:
update : there was a misunderstanding of the Specification on the `mcountinhibit` csr, and it seems it's an RTL issue, the CVA6 should not raise an exception if we don't support perf counter, but the `mcountinhibit` should has zero in it.

zchamski, 2024-08-26:
@AyoubJalali: the privileged spec of CV32A65X has no `mscontext` CSR (address 0x7AA). Can you check if the test is correct in this respect?

zchamski, 2024-08-26:
`tinfo` is a debug register and debug mode is not supported on CV32A65X, so the CSR should be disabled on the Spike side (alongside other debug-related CSRs).  The proper way to do this is yet TBD; we could rely on register grouping specified by the privileged spec plus a single "debug enable" param or disable the registers individually using per-register parameters.

AyoubJalali, 2024-08-27:
> @AyoubJalali: the privileged spec of CV32A65X has no `mscontext` CSR (address 0x7AA). Can you check if the test is correct in this respect?

i had the same question, but spike doesn't raise exception on this CSR (0x7AA)
[image]
[image]

The test that a run, create a sequence of read/write on unmapped CSRs (like 0x7aa), to be sure the it raise exception on every one of them !

AyoubJalali, 2024-08-27:
> Neither the current nor the previous privileged spec (v. 20211203) contains this register. However, it is a "historical" CSR, it exists in upstream Spike and according to @MarioOpenHWGroup it is used in CVE2. PR [openhwgroup/cva6#2407](https://github.com/openhwgroup/cva6/pull/2407) marks it as non-accessible so that Spike will trap on accessing this CSR.

ok got it !

AyoubJalali, 2024-08-27:
A reminder : the bug still on only 2 CSRs : 

1. tinfo 
2. mscontext

AyoubJalali, 2024-08-28:
fixed due to PR cva6 #2407

closed 2024-08-28 by AyoubJalali