openhwfoundation/core-v-verif #2517: Add capability to tp_translator to automatically determine value of CV_DV_UTILS_DIR.

MikeOpenHWGroup, 2024-07-29:
Pull-Request #2511 introduces a new shell environment variable, `CV_DV_UTILS_DIR`, which the user is required to define in their shell environment.  We strive to keep the number of shell variables to a minimum, so our Makefiles typically determine the value of `CORE_V_VERIF` ([example](https://github.com/openhwgroup/core-v-verif/blob/8e25885aa9596fce0a47c53eb0965c981bf4047c/cv32e40p/sim/uvmt/Makefile#L48)) and then define tool or manifest dependent variables based on that ([example](https://github.com/openhwgroup/core-v-verif/blob/8e25885aa9596fce0a47c53eb0965c981bf4047c/mk/Common.mk#L339)).

The purpose of this issue is to add similar functionality to `tp_translator.`

khandelwaltanuj, 2024-08-08:
Hi @MikeOpenHWGroup 

In the DV UTILS we have added the File.f to facilitate the compilation of cv_dv_utils in a new project. It is in this file that the variable CV_DV_UTILS_DIR is used. For the moment I am not sure, how it is possible to do away with this variable. 

The example that yu have pointed to is of a makefile but we are not using makefile to compile cv_dv_utils. Hence I am not so sure how it is possible to do away with this variable. 

Thanks and Regards
Tanuj Khandelwal

MikeOpenHWGroup, 2024-08-08:
> The example that yu have pointed to is of a makefile but we are not using makefile to compile cv_dv_utils. 

Yes, that is true, but a Python script can do what a Makefile can do (and in a way that is much easier to read!).  I suggest that the Python script should do the following:
- determine the value of `CORE_V_VERIF` as an absolute (not relative) path
- define `CV_DV_UTILS_DIR` using `CORE_V_VERIF` as the base.

You can look at the Makefile to see an example and it will be similar in Python.

khandelwaltanuj, 2024-08-08:
Hi @MikeOpenHWGroup 

I am a little confused here. The tp_translator does not use cv_dv_utils_dir. 
The cv_dv_utils_dir variable is used in the files list that we have created.  

It was needed because an external user cannot user CEA makefile system. 

And there is no link between File.f and tp_translator. 

Thanks and Regards
Tanuj