openhwfoundation/core-v-verif #2519: Compilation fails

ilamparithy, 2024-08-02:
Hi Team ,
I checked out the verification environment, i ran the simulation as suggested in CORE-V-VERIF Quick Start Guide
it throws bellow error and stops please help me to resolve this  , I am using ventilator 

home/thiru/ilam/ophw/core-v-verif/mk/Common.mk:303: CV_SW_CFLAGS not defined in either the shell environment, test.yaml or cfg.yaml
/home/thiru/ilam/ophw/core-v-verif/mk/Common.mk:318: RISCV set to /home/thiru/ilam/riscv/bin/riscv32-unknown-elf-
/home/thiru/ilam/ophw/core-v-verif/mk/Common.mk:319: RISCV_PREFIX set to riscv32-unknown-elf-
/home/thiru/ilam/ophw/core-v-verif/mk/Common.mk:320: RISCV_EXE_PREFIX set to /home/thiru/ilam/riscv/bin/riscv32-unknown-elf-/bin/riscv32-unknown-elf-
/home/thiru/ilam/ophw/core-v-verif/mk/Common.mk:321: RISCV_MARCH set to rv32imc
/home/thiru/ilam/ophw/core-v-verif/mk/Common.mk:322: RISCV_CC set to gcc
/home/thiru/ilam/ophw/core-v-verif/mk/Common.mk:323: RISCV_CFLAGS set to 
make veri-test TEST=hello-world
make[1]: Entering directory '/home/thiru/ilam/ophw/core-v-verif/cv32e40p/sim/core'
/home/thiru/ilam/ophw/core-v-verif/mk/Common.mk:303: CV_SW_CFLAGS not defined in either the shell environment, test.yaml or cfg.yaml
/home/thiru/ilam/ophw/core-v-verif/mk/Common.mk:318: RISCV set to /home/thiru/ilam/riscv/bin/riscv32-unknown-elf-
/home/thiru/ilam/ophw/core-v-verif/mk/Common.mk:319: RISCV_PREFIX set to riscv32-unknown-elf-
/home/thiru/ilam/ophw/core-v-verif/mk/Common.mk:320: RISCV_EXE_PREFIX set to /home/thiru/ilam/riscv/bin/riscv32-unknown-elf-/bin/riscv32-unknown-elf-
/home/thiru/ilam/ophw/core-v-verif/mk/Common.mk:321: RISCV_MARCH set to rv32imc
/home/thiru/ilam/ophw/core-v-verif/mk/Common.mk:322: RISCV_CC set to gcc
/home/thiru/ilam/ophw/core-v-verif/mk/Common.mk:323: RISCV_CFLAGS set to 
*******************************************************************************************
* Cloning CV32E40P RTL model
*******************************************************************************************
git clone https://github.com/openhwgroup/cv32e40p /home/thiru/ilam/ophw/core-v-verif/core-v-cores/cv32e40p; cd /home/thiru/ilam/ophw/core-v-verif/core-v-cores/cv32e40p; git checkout fcd5968
fatal: destination path '/home/thiru/ilam/ophw/core-v-verif/core-v-cores/cv32e40p' already exists and is not an empty directory.
HEAD is now at fcd5968 deleted redundant mul custom instructions (#661)
*******************************************************************************************
* Compiling CORE TB and CV32E40P with Verilator
*******************************************************************************************
verilator --cc --sv --exe \
	 \
	--Wno-lint --Wno-UNOPTFLAT \
	--Wno-MODDUP --top-module \
	tb_top_verilator /home/thiru/ilam/ophw/core-v-verif/cv32e40p/tb/core/tb_top_verilator.sv /home/thiru/ilam/ophw/core-v-verif/cv32e40p/tb/core/cv32e40p_tb_wrapper.sv /home/thiru/ilam/ophw/core-v-verif/cv32e40p/tb/core/tb_riscv/riscv_rvalid_stall.sv /home/thiru/ilam/ophw/core-v-verif/cv32e40p/tb/core/tb_riscv/riscv_gnt_stall.sv /home/thiru/ilam/ophw/core-v-verif/cv32e40p/tb/core/mm_ram.sv /home/thiru/ilam/ophw/core-v-verif/cv32e40p/tb/core/dp_ram.sv \
	-f /home/thiru/ilam/ophw/core-v-verif/core-v-cores/cv32e40p/cv32e40p_manifest.flist \
	/home/thiru/ilam/ophw/core-v-verif/core-v-cores/cv32e40p/bhv/cv32e40p_core_log.sv \
	/home/thiru/ilam/ophw/core-v-verif/cv32e40p/tb/core/tb_top_verilator.cpp --Mdir cobj_dir \
	-CFLAGS "-std=gnu++11 -O2" \
	-Wno-BLKANDNBLK  
%Warning-MULTIDRIVEN: /home/thiru/ilam/ophw/core-v-verif/cv32e40p/tb/core/mm_ram.sv:291:9: Variable written to in always_comb also written by other process (IEEE 1800-2023 9.2.2.2): 'data_req_dec'
                                                                                         : ... note: In instance 'tb_top_verilator.cv32e40p_tb_wrapper_i.ram_i'
                      /home/thiru/ilam/ophw/core-v-verif/cv32e40p/tb/core/mm_ram.sv:291:9: 
  291 |         data_req_dec        = '0;
      |         ^~~~~~~~~~~~
                      /home/thiru/ilam/ophw/core-v-verif/cv32e40p/tb/core/mm_ram.sv:196:8: ... Location of other write
  196 |        data_req_dec = data_req_i;
      |        ^~~~~~~~~~~~
                      ... For warning description see https://verilator.org/warn/MULTIDRIVEN?v=5.025
                      ... Use "/* verilator lint_off MULTIDRIVEN */" and lint_on around source to disable this message.
%Warning-MULTIDRIVEN: /home/thiru/ilam/ophw/core-v-verif/cv32e40p/tb/core/mm_ram.sv:292:9: Variable written to in always_comb also written by other process (IEEE 1800-2023 9.2.2.2): 'data_addr_dec'
                                                                                         : ... note: In instance 'tb_top_verilator.cv32e40p_tb_wrapper_i.ram_i'
                      /home/thiru/ilam/ophw/core-v-verif/cv32e40p/tb/core/mm_ram.sv:292:9: 
  292 |         data_addr_dec       = '0;
      |         ^~~~~~~~~~~~~
                      /home/thiru/ilam/ophw/core-v-verif/cv32e40p/tb/core/mm_ram.sv:205:11: ... Location of other write
  205 |           data_addr_dec  = data_addr_i[RAM_ADDR_WIDTH-1:0];
      |           ^~~~~~~~~~~~~
%Warning-MULTIDRIVEN: /home/thiru/ilam/ophw/core-v-verif/cv32e40p/tb/core/mm_ram.sv:293:9: Variable written to in always_comb also written by other process (IEEE 1800-2023 9.2.2.2): 'data_wdata_dec'
                                                                                         : ... note: In instance 'tb_top_verilator.cv32e40p_tb_wrapper_i.ram_i'
                      /home/thiru/ilam/ophw/core-v-verif/cv32e40p/tb/core/mm_ram.sv:293:9: 
  293 |         data_wdata_dec      = '0;
      |         ^~~~~~~~~~~~~~
                      /home/thiru/ilam/ophw/core-v-verif/cv32e40p/tb/core/mm_ram.sv:208:8: ... Location of other write
  208 |        data_wdata_dec = data_wdata_i;
      |        ^~~~~~~~~~~~~~
%Warning-MULTIDRIVEN: /home/thiru/ilam/ophw/core-v-verif/cv32e40p/tb/core/mm_ram.sv:294:9: Variable written to in always_comb also written by other process (IEEE 1800-2023 9.2.2.2): 'data_we_dec'
                                                                                         : ... note: In instance 'tb_top_verilator.cv32e40p_tb_wrapper_i.ram_i'
                      /home/thiru/ilam/ophw/core-v-verif/cv32e40p/tb/core/mm_ram.sv:294:9: 
  294 |         data_we_dec         = '0;
      |         ^~~~~~~~~~~
                      /home/thiru/ilam/ophw/core-v-verif/cv32e40p/tb/core/mm_ram.sv:209:8: ... Location of other write
  209 |        data_we_dec    = data_we_i;
      |        ^~~~~~~~~~~
%Warning-MULTIDRIVEN: /home/thiru/ilam/ophw/core-v-verif/cv32e40p/tb/core/mm_ram.sv:295:9: Variable written to in always_comb also written by other process (IEEE 1800-2023 9.2.2.2): 'data_be_dec'
                                                                                         : ... note: In instance 'tb_top_verilator.cv32e40p_tb_wrapper_i.ram_i'
                      /home/thiru/ilam/ophw/core-v-verif/cv32e40p/tb/core/mm_ram.sv:295:9: 
  295 |         data_be_dec         = '0;
      |         ^~~~~~~~~~~
                      /home/thiru/ilam/ophw/core-v-verif/cv32e40p/tb/core/mm_ram.sv:210:8: ... Location of other write
  210 |        data_be_dec    = data_be_i;
      |        ^~~~~~~~~~~
%Warning-MULTIDRIVEN: /home/thiru/ilam/ophw/core-v-verif/cv32e40p/tb/core/mm_ram.sv:312:9: Variable written to in always_comb also written by other process (IEEE 1800-2023 9.2.2.2): 'transaction'
                                                                                         : ... note: In instance 'tb_top_verilator.cv32e40p_tb_wrapper_i.ram_i'
                      /home/thiru/ilam/ophw/core-v-verif/cv32e40p/tb/core/mm_ram.sv:312:9: 
  312 |         transaction         = T_PER;
      |         ^~~~~~~~~~~
                      /home/thiru/ilam/ophw/core-v-verif/cv32e40p/tb/core/mm_ram.sv:211:8: ... Location of other write
  211 |        transaction    = T_RAM;
      |        ^~~~~~~~~~~
%Warning-COMBDLY: /home/thiru/ilam/ophw/core-v-verif/core-v-cores/cv32e40p/rtl/../bhv/cv32e40p_sim_clock_gate.sv:25:31: Non-blocking assignment '<=' in combinational logic process
                                                                                                                      : ... This will be executed as a blocking assignment '='!
   25 |     if (clk_i == 1'b0) clk_en <= en_i | scan_cg_en_i;
      |                               ^~
                  *** See https://verilator.org/warn/COMBDLY before disabling this,
                  else you may end up with different sim results.
%Error: Exiting due to 7 warning(s)
make[1]: *** [Makefile:491: testbench_verilator] Error 1
make[1]: Leaving directory '/home/thiru/ilam/ophw/core-v-verif/cv32e40p/sim/core'
make: *** [Makefile:485: sanity-veri-run] Error 2

ilamparithy, 2024-08-02:
I bypassed the warning in make file let me check

closed 2024-08-02 by ilamparithy

janschafrich, 2025-01-30:
I just cloned the core-v-verif repo and installed the latest Verilator version. 
Running make to perform the hello_world test I get the same error, see below.

How did you resolve the problem. Is there an issue with the testbench?
```
/home/jscha/projects/core-v-verif/mk/Common.mk:278: CV_SW_MARCH not defined in either the shell environment, test.yaml or cfg.yaml
/home/jscha/projects/core-v-verif/mk/Common.mk:291: CV_SW_CC not defined in either the shell environment, test.yaml or cfg.yaml
/home/jscha/projects/core-v-verif/mk/Common.mk:303: CV_SW_CFLAGS not defined in either the shell environment, test.yaml or cfg.yaml
/home/jscha/projects/core-v-verif/mk/Common.mk:318: RISCV set to /home/jscha/.config/mlonmcu/environments/default/deps/install/riscv_gcc
/home/jscha/projects/core-v-verif/mk/Common.mk:319: RISCV_PREFIX set to riscv32-unknown-elf-
/home/jscha/projects/core-v-verif/mk/Common.mk:320: RISCV_EXE_PREFIX set to /home/jscha/.config/mlonmcu/environments/default/deps/install/riscv_gcc/bin/riscv32-unknown-elf-
/home/jscha/projects/core-v-verif/mk/Common.mk:321: RISCV_MARCH set to rv32imc
/home/jscha/projects/core-v-verif/mk/Common.mk:322: RISCV_CC set to gcc
/home/jscha/projects/core-v-verif/mk/Common.mk:323: RISCV_CFLAGS set to 
make veri-test TEST=hello-world
make[1]: Entering directory '/home/jscha/projects/core-v-verif/cv32e40p/sim/core'
/home/jscha/projects/core-v-verif/mk/Common.mk:278: CV_SW_MARCH not defined in either the shell environment, test.yaml or cfg.yaml
/home/jscha/projects/core-v-verif/mk/Common.mk:291: CV_SW_CC not defined in either the shell environment, test.yaml or cfg.yaml
/home/jscha/projects/core-v-verif/mk/Common.mk:303: CV_SW_CFLAGS not defined in either the shell environment, test.yaml or cfg.yaml
/home/jscha/projects/core-v-verif/mk/Common.mk:318: RISCV set to /home/jscha/.config/mlonmcu/environments/default/deps/install/riscv_gcc
/home/jscha/projects/core-v-verif/mk/Common.mk:319: RISCV_PREFIX set to riscv32-unknown-elf-
/home/jscha/projects/core-v-verif/mk/Common.mk:320: RISCV_EXE_PREFIX set to /home/jscha/.config/mlonmcu/environments/default/deps/install/riscv_gcc/bin/riscv32-unknown-elf-
/home/jscha/projects/core-v-verif/mk/Common.mk:321: RISCV_MARCH set to rv32imc
/home/jscha/projects/core-v-verif/mk/Common.mk:322: RISCV_CC set to gcc
/home/jscha/projects/core-v-verif/mk/Common.mk:323: RISCV_CFLAGS set to 
*******************************************************************************************
* Cloning CV32E40P RTL model
*******************************************************************************************
git clone https://github.com/openhwgroup/cv32e40p /home/jscha/projects/core-v-verif/core-v-cores/cv32e40p; cd /home/jscha/projects/core-v-verif/core-v-cores/cv32e40p; git checkout fcd5968
fatal: destination path '/home/jscha/projects/core-v-verif/core-v-cores/cv32e40p' already exists and is not an empty directory.
HEAD is now at fcd5968 deleted redundant mul custom instructions (#661)
*******************************************************************************************
* Compiling CORE TB and CV32E40P with Verilator
*******************************************************************************************
verilator --cc --sv --exe \
         \
        --Wno-lint --Wno-UNOPTFLAT \
        --Wno-MODDUP --top-module \
        tb_top_verilator /home/jscha/projects/core-v-verif/cv32e40p/tb/core/tb_top_verilator.sv /home/jscha/projects/core-v-verif/cv32e40p/tb/core/cv32e40p_tb_wrapper.sv /home/jscha/projects/core-v-verif/cv32e40p/tb/core/tb_riscv/riscv_rvalid_stall.sv /home/jscha/projects/core-v-verif/cv32e40p/tb/core/tb_riscv/riscv_gnt_stall.sv /home/jscha/projects/core-v-verif/cv32e40p/tb/core/mm_ram.sv /home/jscha/projects/core-v-verif/cv32e40p/tb/core/dp_ram.sv \
        -f /home/jscha/projects/core-v-verif/core-v-cores/cv32e40p/cv32e40p_manifest.flist \
        /home/jscha/projects/core-v-verif/core-v-cores/cv32e40p/bhv/cv32e40p_core_log.sv \
        /home/jscha/projects/core-v-verif/cv32e40p/tb/core/tb_top_verilator.cpp --Mdir cobj_dir \
        -CFLAGS "-std=gnu++11 -O2" \
        -Wno-BLKANDNBLK  
%Warning-MULTIDRIVEN: /home/jscha/projects/core-v-verif/cv32e40p/tb/core/mm_ram.sv:291:9: Variable written to in always_comb also written by other process (IEEE 1800-2023 9.2.2.2): 'data_req_dec'
                                                                                        : ... note: In instance 'tb_top_verilator.cv32e40p_tb_wrapper_i.ram_i'
                      /home/jscha/projects/core-v-verif/cv32e40p/tb/core/mm_ram.sv:291:9: 
  291 |         data_req_dec        = '0;
      |         ^~~~~~~~~~~~
                      /home/jscha/projects/core-v-verif/cv32e40p/tb/core/mm_ram.sv:196:8: ... Location of other write
  196 |        data_req_dec = data_req_i;
      |        ^~~~~~~~~~~~
                      ... For warning description see https://verilator.org/warn/MULTIDRIVEN?v=5.033
                      ... Use "/* verilator lint_off MULTIDRIVEN */" and lint_on around source to disable this message.
```

MikeOpenHWGroup, 2025-01-30:
Hi @janschafrich, you are commenting on a closed issue - please open a new one.  Also, I would suggest running the CV32E40P from the `cv32e40p/dev` branch.