openhwfoundation/core-v-verif #2522: [cv_dv_utils reset_driver_c] reset_driver does not compile with uvm-1.2

khandelwaltanuj, 2024-08-08:
Hello,

In one of the class of the reset_driver_c, the constructor "new" was missing, it did not pose any issues with uvm-1.1d. 
Adding the constructor resolves the issue with uvm-1.2. 

Thanks and regards
Tanuj Khandelwal

MikeOpenHWGroup, 2024-08-08:
Hi @khandelwaltanuj, it is not clear to me what the issue is here and what action we should take.  If you are talking about [cv_dv_utils/uvm/reset_gen/reset_driver_c.svh](https://github.com/openhwgroup/core-v-verif/blob/master/lib/cv_dv_utils/uvm/reset_gen/reset_driver_c.svh), the constructor is implemented [here](https://github.com/openhwgroup/core-v-verif/blob/e06bd577f5db816015ec16a901e6e8f39cf7d640/lib/cv_dv_utils/uvm/reset_gen/reset_driver_c.svh#L121).  Did you mean [dummy_txn](https://github.com/openhwgroup/core-v-verif/blob/e06bd577f5db816015ec16a901e6e8f39cf7d640/lib/cv_dv_utils/uvm/reset_gen/reset_driver_c.svh#L85) or [reset_counter_c](https://github.com/openhwgroup/core-v-verif/blob/e06bd577f5db816015ec16a901e6e8f39cf7d640/lib/cv_dv_utils/uvm/reset_gen/reset_driver_c.svh#L89)?  If so, please create a pull-request to resolve this Issue.

Thanks!

khandelwaltanuj, 2024-08-08:
Hi @MikeOpenHWGroup 

My bad, it did not give the details. It is the reset_counter_c class that poses problem with UVM-1.2. 

Thanks!

MikeOpenHWGroup, 2024-08-08:
Is this Issue resolved in pull-request #2524?

MikeOpenHWGroup, 2024-08-29:
Hi @khandelwaltanuj, is there an update on this issue?

khandelwaltanuj, 2024-10-07:
Hi @MikeOpenHWGroup 

This issue has been fxed normally. 

Regards
Tanuj

MikeOpenHWGroup, 2024-10-07:
Hi @khandelwaltanuj, if that is the case, please add a comment to indicated which PR fixed the problemm and then close this issue.  Thanks.

khandelwaltanuj, 2024-10-09:
Hi @MikeOpenHWGroup

Normally the pull request https://github.com/openhwgroup/core-v-verif/pull/2524 has the fix of this issue.

Regards
Tanuj

closed 2024-10-09 by MikeOpenHWGroup