openhwfoundation/core-v-verif #2523: [uvms_axi5] issue while compile the axi_agent with questasim

khandelwaltanuj, 2024-08-08:
Hello
 
I did a fork to be able to use uvma_axi5 in my environnement. 
I tried to compile it with questasim (2023.3) et I have following errors.  Can you please take a look it ?
 
** Error (suppressible): (vlog-13276) /home/360.1.361-EPI/users/tk223379/core-v-verif/lib/uvm_agents/uvma_axi5/src/seq/[uvma_axi_slv_seq.sv](http://uvma_axi_slv_seq.sv/)(279): Could not find field/method name (rand_mode) in 'upper_byte_lane' of 'r_slv_rsp.upper_byte_lane.rand_mode'.
** Error (suppressible): (vlog-13276) /home/360.1.361-EPI/users/tk223379/core-v-verif/lib/uvm_agents/uvma_axi5/src/seq/[uvma_axi_slv_seq.sv](http://uvma_axi_slv_seq.sv/)(278): Could not find field/method name (rand_mode) in 'lower_byte_lane' of 'r_slv_rsp.lower_byte_lane.rand_mode'.
** Error (suppressible): (vlog-13276) /home/360.1.361-EPI/users/tk223379/core-v-verif/lib/uvm_agents/uvma_axi5/src/seq/[uvma_axi_slv_seq.sv](http://uvma_axi_slv_seq.sv/)(230): Could not find field/method name (rand_mode) in 'upper_byte_lane' of 'w_slv_rsp.upper_byte_lane.rand_mode'.
** Error (suppressible): (vlog-13276) /home/360.1.361-EPI/users/tk223379/core-v-verif/lib/uvm_agents/uvma_axi5/src/seq/[uvma_axi_slv_seq.sv](http://uvma_axi_slv_seq.sv/)(229): Could not find field/method name (rand_mode) in 'lower_byte_lane' of 'w_slv_rsp.lower_byte_lane.rand_mode'.
** Error (suppressible): (vlog-13276) /home/360.1.361-EPI/users/tk223379/core-v-verif/lib/uvm_agents/uvma_axi5/src/seq/[uvma_axi_slv_seq.sv](http://uvma_axi_slv_seq.sv/)(229): Could not find field/method name (rand_mode) in 'lower_byte_lane' of 'w_slv_rsp.lower_byte_lane.rand_mode.$0'.
** Error (suppressible): (vlog-13276) /home/360.1.361-EPI/users/tk223379/core-v-verif/lib/uvm_agents/uvma_axi5/src/seq/[uvma_axi_slv_seq.sv](http://uvma_axi_slv_seq.sv/)(230): Could not find field/method name (rand_mode) in 'upper_byte_lane' of 'w_slv_rsp.upper_byte_lane.rand_mode.$0'.
** Warning: /home/360.1.361-EPI/users/tk223379/core-v-verif/lib/uvm_agents/uvma_axi5/src/seq/[uvma_axi_slv_seq.sv](http://uvma_axi_slv_seq.sv/)(243): (vlog-2240) Treating stand-alone use of function 'randomize' as an implicit VOID cast.
** Error (suppressible): (vlog-13276) /home/360.1.361-EPI/users/tk223379/core-v-verif/lib/uvm_agents/uvma_axi5/src/seq/[uvma_axi_slv_seq.sv](http://uvma_axi_slv_seq.sv/)(278): Could not find field/method name (rand_mode) in 'lower_byte_lane' of 'r_slv_rsp.lower_byte_lane.rand_mode.$0'.
** Error (suppressible): (vlog-13276) /home/360.1.361-EPI/users/tk223379/core-v-verif/lib/uvm_agents/uvma_axi5/src/seq/[uvma_axi_slv_seq.sv](http://uvma_axi_slv_seq.sv/)(279): Could not find field/method name (rand_mode) in 'upper_byte_lane' of 'r_slv_rsp.upper_byte_lane.rand_mode.$0'.
 
 
I have following warnings also. 
 
** Warning: ** while parsing file included at /home/360.1.361-EPI/users/tk223379/core-v-verif/lib/uvm_agents/uvma_axi5/src/[uvma_axi_pkg.sv](http://uvma_axi_pkg.sv/)(70)
** while parsing file included at /home/360.1.361-EPI/users/tk223379/core-v-verif/lib/uvm_agents/uvma_axi5/src/seq/[uvma_axi_seq_lib.sv](http://uvma_axi_seq_lib.sv/)(17)
** at /home/360.1.361-EPI/users/tk223379/core-v-verif/lib/uvm_agents/uvma_axi5/src/seq/[uvma_axi_fw_preload_seq.sv](http://uvma_axi_fw_preload_seq.sv/)(18): (vlog-2181) Use of a parameterized class uvm_sequence as a type creates a default specialization.
** Warning: ** while parsing file included at /home/360.1.361-EPI/users/tk223379/core-v-verif/lib/uvm_agents/uvma_axi5/src/[uvma_axi_pkg.sv](http://uvma_axi_pkg.sv/)(70)
** while parsing file included at /home/360.1.361-EPI/users/tk223379/core-v-verif/lib/uvm_agents/uvma_axi5/src/seq/[uvma_axi_seq_lib.sv](http://uvma_axi_seq_lib.sv/)(18)
** at /home/360.1.361-EPI/users/tk223379/core-v-verif/lib/uvm_agents/uvma_axi5/src/comps/[uvma_axi_vseq.sv](http://uvma_axi_vseq.sv/)(17): (vlog-2181) Use of a parameterized class uvm_sequence as a type creates a default specialization.
 
Merci et cordialement
Tanuj Khandelwal

MikeOpenHWGroup, 2024-09-06:
Hi @khandelwaltanuj.  Can you provide us with the exact command lines you used to produce these errors?   Please start with `git clone...`.

khandelwaltanuj, 2024-09-09:
Hi @MikeOpenHWGroup 

Here are the commands that I greped from my build and simulation log files. Difficult to provide a concrete example for the moment. We have a testbench o check axi master and slave, but it is not part of the opensource yet. 

clone: 
 git clone git@github.com:openhwgroup/core-v-verif.git

setenv CORE_V_VERIF     <core-v-verif-clone>
setenv DV_UVMA_AXI_PATH $CORE_V_VERIF/lib/uvm_agents/uvma_axi5/
setenv DV_UVML_LOGS_PATH $CORE_V_VERIF/lib/uvm_libs/uvml_logs
setenv DV_UVML_TRN_PATH $CORE_V_VERIF/lib/uvm_libs/uvml_trn/
setenv DV_UVML_MEM_PATH $CORE_V_VERIF/lib/uvm_libs/uvml_mem/

VLOG: axi agent 
vlog -sv -F $CORE_V_VERIF/lib/uvm_libs/uvml_mem/uvml_mem_pkg.flist -F $CORE_V_VERIF/lib/uvm_libs/uvml_trn/uvml_trn_pkg.flist -F $CORE_V_VERIF/lib/uvm_libs/uvml_logs/uvml_logs_pkg.flist -F $CORE_V_VERIF/lib/uvm_agents/uvma_axi5/src/uvma_axi_pkg.flist -L $QUESTA_HOME/uvm-1.2 "+incdir+$QUESTA_HOME/verilog_src/uvm-1.2/src" -suppress 2583 -note 13314 $CORE_V_VERIF/lib/uvm_agents/uvma_axi5//src/uvma_axi_intf.sv -work tb_lib 

VLOG: ...some other environment pkgs 

VLOG: TOP
vlog -sv tb_lib -L $QUESTA_HOME/uvm-1.2 "+incdir+$QUESTA_HOME/verilog_src/uvm-1.2/src" -suppress 2583 -note 13314 top.sv -work tb_lib 

VOPT: 
vopt  -assertdebug  top -o opt -work tb_lib -L tb_lib  +acc -note 12003 -L /home/cao/mgc/questasim/2023.3/uvm-1.2/ -suppress 2583 -note 13314  

VSIM: 
vsim  -L tb_lib -note 8233 -note 12003 -c -classdebug "+UVM_TESTNAME=test" -do "run.do" -solvefailseverity=4 -solvefaildebug=2 -solvefailtestcase=constraint_failure.txt -sv_seed 1 -l test.log -wlf test.wlf "+UVM_VERBOSITY=UVM_DEBUG" -assertdebug -msgmode both -lib tb_lib opt 

Regards
Tanuj Khandelwal

khandelwaltanuj, 2024-09-12:
Hi @MikeOpenHWGroup 

As these errors are coming from the AXI thales, I believe it should be assigned to them ? 
For information, ludovic is on vacation until end of september. 

Thanks and Regards
Tanuj Khandelwal

MikeOpenHWGroup, 2024-09-12:
Hi @AyoubJalali , I am currently traveling for work and do not have access to Questa. Can you help out @khandelwaltanuj?

khandelwaltanuj, 2024-09-13:
Hi @AnouarZajni @MikeOpenHWGroup 

I believe the error is due to the fact that the variable, upper_byte_lane, lower_byte_lane are not rand.  

as per queta sim the function rand_mode exists only for the variable rand_mode. 
upper_byte_lane.rand_mode

Thanks and Regards
Tanuj