openhwfoundation/core-v-verif #2531: [BUG] : mip.MSIP and mie.MSIE are read-only zeros

AyoubJalali, 2024-09-12:
According to [CV32A65X Spec](https://docs.openhwgroup.org/projects/cva6-user-manual/04_cv32a65x/riscv/priv.html#_machine_interrupt_mip_and_mie_registers), As the system has only one hart then mip.MSIP and mie.MSIE are read-only zeros.

So there's a bug on Spike (solo and tandem), it allow modifying mip.MSIP and mie.MSIE

zchamski, 2024-09-27:
After locally updating the mask/override/init parameters in Spike Yaml config of CV32A65X the behavior is as per specification, duly exposing the CVA6 RTL issue (https://github.com/openhwgroup/cva6/issues/2500).

AyoubJalali, 2024-10-16:
Fixed by https://github.com/openhwgroup/cva6/pull/2523
Thanks to @zchamski