openhwfoundation/core-v-verif #2537: corev-dv rand test issue

vinomutty, 2024-10-08:
corev_rand_arithmetic_base_test & corev_rand_jump_stress_test  these test cases are passing except these two testcase remaining 
   test case facing with same issue
   ERROR: 
 /opt/lowrisc/lowrisc-toolchain-rv32imcb-20240206-1/bin/../lib/gcc/riscv32-unknown-elf/10.2.0/../../../../riscv32-unknown-elf/bin/ld: cannot find -lcv-verif
collect2: error: ld returned 1 exit status
make: *** [/Projects/marmik_project/hemashri.bhagavati/final/core-v-verif/mk/Common.mk:490: /Projects/marmik_project/hemashri.bhagavati/final/core-v-verif/cv32e40p/sim/uvmt/vsim_results/default/corev_rand_interrupt_exception/0/test_program/corev_rand_interrupt_exception_0.elf] Error 1

 we tried with this test corev_rand_interrupt_exception  this test case facing issue as cannot finding lcv_verify and shared the issue     
  Tried changing yaml config settings also  like two passing test but  facing same issue.
  whether the rand in test case name need to be updated for other test case also  ,because those test case name is already 
  mentioned with rand.

MikeOpenHWGroup, 2024-10-08:
Hi @vinomutty, thanks for your interest in CORE-V-VERIF!

Without more information it is difficult to say for certain what the trouble is.  In the future, when creating an Issue, please provide the command-lines used (starting with `git clone ....`) and attach the entire logfile to the Issue.

It seems that the [make bsp](https://github.com/openhwgroup/core-v-verif/blob/33917ee5db9ede3779cc658a5004d672320f164c/mk/Common.mk#L545) command in `mk/Common.mk` failed to compile the BSP to produce `libcv-verif.a`.  This is probably because you are using a very old toolchain.  When working with CORE-V-VERIF please follow the [TOOLCHAIN](https://github.com/openhwgroup/core-v-verif/blob/cv32e40p/dev/mk/TOOLCHAIN.md#toolchain-installation-instructions) instructions for installing the proper cross-compiler (toolchain).

Lastly, when working with the CV32E40P, please use the `cv32e40p/dev` branch of this repo.

MikeOpenHWGroup, 2024-11-15:
Hi @vinomutty, we haven't heard from you in a more than a month, so I will assume either my suggestions on #2539 resolved your issue or you are no longer in need of assistance and I can close this issue.

closed 2024-11-15 by MikeOpenHWGroup