openhwfoundation/core-v-verif #2539: corev-dv rand test issue

vinomutty, 2024-10-09:
Hi 
Please find below commands used for running test cases

command lines:
Available test programs
export CV_SW_MARCH=rv32imc 
export CV_SW_PREFIX=riscv32-unknown-elf- 
export CV_SW_TOOLCHAIN=/opt/lowrisc/lowrisc-toolchain-rv32imcb-20240206-1 
export RISCV=/opt/lowrisc/lowrisc-toolchain-rv32imcb-20240206-1
   
 For custom testcase :make test TEST= riscv_arithmetic_basic_test_0 CV_SIMULATOR=vsim USE_ISS=IMPERAS OVPSIM
 
COREV-DV Generated Tests:
make SIMULATOR=vsim corev-dv
make gen_corev-dv test TEST=corev_rand_arithmetic_base_test SIMULATOR=vsim USE_ISS=IMPERAS OVP

[error.txt](https://github.com/user-attachments/files/17303475/error.txt)

Thanks

MikeOpenHWGroup, 2024-10-09:
Hi @vinomutty, this Issue seems to be the same one as #2537.  If that is the case, please close that Issue and we will continue the discussion here.

A couple of comments:
- I need to see _all_ of your commands, starting from `git clone ...`  I am specifically looking to see which branch of the repo you have cloned and which directory you run the `make` commands from.
- The variable USE_ISS should be set to either "YES" or "NO"  ("1" or "0" should also work).  The CV32E40P environment does not support IMPERAS OVPSIM.
- As indicated in my comment in #2537, the problem seems to be with compiling the BSP:
```
*******************************************************************************************
* Compiling the BSP
*******************************************************************************************
mkdir -p /Projects/marmik_project/devipriya.rajendran/final/core-v-verif/cv32e40p/sim/uvmt/vsim_results/default/corev_rand_debug/0/test_program/bsp
cp /Projects/marmik_project/devipriya.rajendran/final/core-v-verif/cv32e40p/bsp/Makefile /Projects/marmik_project/devipriya.rajendran/final/core-v-verif/cv32e40p/sim/uvmt/vsim_results/default/corev_rand_debug/0/test_program/bsp
make -C /Projects/marmik_project/devipriya.rajendran/final/core-v-verif/cv32e40p/sim/uvmt/vsim_results/default/corev_rand_debug/0/test_program/bsp \
	VPATH=/Projects/marmik_project/devipriya.rajendran/final/core-v-verif/cv32e40p/bsp \
	RISCV=/opt/lowrisc/lowrisc-toolchain-rv32imcb-20240206-1 \
	RISCV_PREFIX=riscv32-unknown-elf- \
	RISCV_EXE_PREFIX=/opt/lowrisc/lowrisc-toolchain-rv32imcb-20240206-1/bin/riscv32-unknown-elf- \
	RISCV_MARCH=rv32imc \
	RISCV_CC=gcc \
	RISCV_CFLAGS="" \
	all
make[2]: Entering directory '/Projects/marmik_project/devipriya.rajendran/final/core-v-verif/cv32e40p/sim/uvmt/vsim_results/default/corev_rand_debug/0/test_program/bsp'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/Projects/marmik_project/devipriya.rajendran/final/core-v-verif/cv32e40p/sim/uvmt/vsim_results/default/corev_rand_debug/0/test_program/bsp'
make[1]: Leaving directory '/Projects/marmik_project/devipriya.rajendran/final/core-v-verif/cv32e40p/sim/uvmt'
```
`Make` is saying `make[2]: Nothing to be done for 'all'.` which means it is not compiling the BSP which means that the `cv-verif` library is not greated for the subsequent compile of the test-program:
```
*******************************************************************************************
* Compiling test-program /Projects/marmik_project/devipriya.rajendran/final/core-v-verif/cv32e40p/sim/uvmt/vsim_results/default/corev_rand_debug/0/test_program/corev_rand_debug_0.elf
*******************************************************************************************
/opt/lowrisc/lowrisc-toolchain-rv32imcb-20240206-1/bin/riscv32-unknown-elf-gcc \
	-DNO_PULP \
	-Os -g -static -mabi=ilp32 -march=rv32imc -Wall -pedantic  \
	 \
	-I ../../tests/asm \
	-I /Projects/marmik_project/devipriya.rajendran/final/core-v-verif/cv32e40p/bsp \
	-o /Projects/marmik_project/devipriya.rajendran/final/core-v-verif/cv32e40p/sim/uvmt/vsim_results/default/corev_rand_debug/0/test_program/corev_rand_debug_0.elf \
	-nostartfiles \
	/Projects/marmik_project/devipriya.rajendran/final/core-v-verif/cv32e40p/sim/uvmt/vsim_results/default/corev_rand_debug/0/test_program/corev_rand_debug_0.S \
	-T /Projects/marmik_project/devipriya.rajendran/final/core-v-verif/cv32e40p/sim/uvmt/vsim_results/default/corev_rand_debug/0/test_program/link.ld \
	-L /Projects/marmik_project/devipriya.rajendran/final/core-v-verif/cv32e40p/sim/uvmt/vsim_results/default/corev_rand_debug/0/test_program -L /Projects/marmik_project/devipriya.rajendran/final/core-v-verif/cv32e40p/sim/uvmt/vsim_results/default/corev_rand_debug/0/test_program/bsp \
	-lcv-verif
/opt/lowrisc/lowrisc-toolchain-rv32imcb-20240206-1/bin/../lib/gcc/riscv32-unknown-elf/10.2.0/../../../../riscv32-unknown-elf/bin/ld: cannot find -lcv-verif
collect2: error: ld returned 1 exit status
```
The error message `cannot find -lcv-verif` is the problem.

You can try a few things:
1. Make sure you are using the `cv32e40p/dev` branch of the core-v-verif repo.  This is _very_ important.
2. In the `cv32e40p/sim/uvmt` directory run `make clean_all SIMULATOR=vsim` and re-run the test.
3. Try the CORE-V prebuilt toolchain mentoned in the README (see my comment in #2537).

MikeOpenHWGroup, 2024-11-15:
Hi @vinomutty, we haven't heard from you in more than a month.  Have the above suggestions resolved your issue?

vinomutty, 2025-02-25:
Hi 
i am using synopsys tool (VCS simulator) and tried to run corev-dv testcases in cv32e40p.
other than USE ISS=0 all other options will run with reference model. AM i correct?

other options in the sense like Spike,Sail and Whisper

MikeOpenHWGroup, 2025-02-25:
> i am using synopsys tool (VCS simulator) and tried to run corev-dv testcases in cv32e40p.

Are you working on the `cv32e40p/dev` branch?  This is _very_ important.

> other than USE ISS=0 all other options will run with reference model. AM i correct?

`USE_ISS=1` will select the Synopsys reference model (known as ImperasDV). This is the default.  `USE_ISS=0` will run without the reference model.

> other options in the sense like Spike,Sail and Whisper

We have not used either Sail or Whisper at OpenHW.  Spike is used for the CVA6 and will be used CVE2.  There are currently no plans to use Spike for CV32E40P, although we would be happy to discuss a contribution if you would like to do this.

vinomutty, 2025-02-27:
Hi
i am using below clone
git clone https://github.com/openhwgroup/core-v-verif.git $CORE_V_VERIF

MikeOpenHWGroup, 2025-02-28:
Please see my comment above:

> Make sure you are using the cv32e40p/dev branch of the core-v-verif repo. This is very important.

Try `git clone -b cv32e40p/dev ....`